ev3dev/ev3dev #760: Ethernet over USB sporadic on ubuntu 16.04 - due to device name changes

kortschak, 2016-10-13:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: linux-image-4.4.19-15-ev3dev-ev3-kortschak-ev3+ (story)

* My host computer is running: Ubuntu 16.04

* I am connecting using: nil (story - no WiFi because of kernel above and USB is the issue).

### About my issue:

Attempting to connect to EV3 with ethernet over USB is sporadic (increasing less frequently working). The device brings up an interface as shown by Brickmon (IP is 169.254.10.110), but laptop shows the ethernet over USB device to not be connected to an interface. The relevant ifconfig stanza:

```
enp0s20f0u4c2 Link encap:Ethernet  HWaddr 12:16:53:xx:xx:xx  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:57 errors:0 dropped:0 overruns:0 frame:0
          TX packets:141 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:14750 (14.7 KB)  TX bytes:23927 (23.9 KB)
```

This sometime works, though now rarely. I suspect that is has something to do with negotiation between the network devices and not due to the kernel changes which are solely in the lego driver `sysfs_notify` code for motor state. I think that this may be a 16.04 issue since I have never seen this behaviour with 14.04, though that laptop is now dead, so I cannot test this.

**I have just figured out why this is the case. The ethernet over USB device name changes between sessions. I'm not sure that there is anything that we can do about this, but I am posting the issue anyway so that if others find this problem they can try the relevant solutions - change the device identity in the network manager connections editor.**

dlech, 2016-10-13:
So what are the exact steps to repeat the problem? Are you rebooting your EV3, or just unplugging the USB cable? How is the device configured in NetworkManager?

kortschak, 2016-10-13:
I need to play with this some more, but I suspect (with close to zero evidence) that it is due to plugging the USB cable into different ports and the device name changing dependent on that because of difference USB hubs being involved. The device is configured as described [in the getting started guide](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/).

kortschak, 2016-10-14:
Confirmed. The difference is due to the USB port that is used. One gives the device with a suffix u4c2 another with a suffix u3c2 and the last with u1c2. When the name does not match the name specified in the network manager connection editor the network manager fails to recognise it.

dlech, 2016-10-14:
Interesting. Sounds like a bug with NetworkManager rather than ev3dev.

kortschak, 2016-10-14:
Yeah, it's horrible. The devices all look like nasty bsd things. This issue is really here to let others know in case they get bitten. I don't think there is anything we can do.

kortschak, 2016-10-14:
It appears to be a [systemd thing](https://github.com/systemd/systemd/blob/master/src/udev/udev-builtin-net_id.c#L43). I will have a look later to see if just specifying the device number allows the network manager to set up the connection. Otherwise it's probably an issue against network-manager to properly handle systemd device names.

dlech, 2016-10-14:
Might be something useful here: https://lists.ubuntu.com/archives/snappy-devel/2016-April/001697.html

kortschak, 2016-10-14:
The easy solution for us here is a documentation change to the [connecting via USB page](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/), particularly the graphic [here](http://www.ev3dev.org/images/ubuntu/15.10/editing-ev3dev-ethernet.png) to show that only the MAC address should be entered - in that state the connection is port-agnostic.

dlech, 2016-10-14:
Great suggestion! That is much betting than setting up custom udev rules.

closed 2016-10-15 by dlech