ev3dev/ev3dev #762: rpi3 - reboot after upgrade to kernel 16 keeps kernel 15, not kernel 16

BertLindeman, 2016-10-16:
### System information:

* I am using: *Rapsberry Pi Model 3B*

* I installed ev3dev using this image file: *rpi2-ev3dev-jessie-2015-09-15* 
But I am not really sure about this!

* My kernel version is: *4.4.19-15-ev3dev-rpi2*

* My host computer is running: *WIN 10*

* I am connecting using: *Bluetooth, Wi-Fi*

### About my issue:

Today I ran ```apt update && apt -V upgrade``` so `flash-kernel` was upgraded and 
`linux-image-4.4.23-16-ev3dev-rpi2` was installed as expected.

After a reboot `uname -a` still shows:
```
Linux rpi3 4.4.19-15-ev3dev-rpi2 #1 SMP Sun Sep 4 17:15:21 CDT 2016 armv7l GNU/Linux
```
`journalctl -b` starts with
```
Oct 16 09:25:26 rpi3 kernel: Linux version 4.4.19-15-ev3dev-rpi2 (david@freyr) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1 SMP Sun Sep 4 17:15:21 CDT 201
```
Verified that I really *did* reboot.

* The same occurs on my rpi-b+

Tried:
```
apt-get -V --reinstall install flash-kernel linux-image-4.4.23-16-ev3dev-rpi2
```
And answered *no* on the question to stop the install.

Output from the reinstall:

```
Examining /etc/kernel/preinst.d/
Done.
Unpacking linux-image-4.4.23-16-ev3dev-rpi2 (1) over (1) ...
Examining /etc/kernel/postrm.d .
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 4.4.23-16-ev3dev-rpi2 /boot/vmlinuz-4.4.23-16-ev3dev-rpi2
run-parts: executing /etc/kernel/postrm.d/zz-flash-kernel 4.4.23-16-ev3dev-rpi2 /boot/vmlinuz-4.4.23-16-ev3dev-rpi2
Preparing to unpack .../flash-kernel_3.35ev3dev13_armhf.deb ...
Unpacking flash-kernel (3.35ev3dev13) over (3.35ev3dev13) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up linux-image-4.4.23-16-ev3dev-rpi2 (1) ...

 Hmm. There is a symbolic link /lib/modules/4.4.23-16-ev3dev-rpi2/build
 However, I can not read it: No such file or directory
 Therefore, I am deleting /lib/modules/4.4.23-16-ev3dev-rpi2/build

 Hmm. The package shipped with a symbolic link /lib/modules/4.4.23-16-ev3dev-rpi2/source
 However, I can not read the target: No such file or directory
 Therefore, I am deleting /lib/modules/4.4.23-16-ev3dev-rpi2/source

Running depmod.
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.4.23-16-ev3dev-rpi2 /boot/vmlinuz-4.4.23-16-ev3dev-rpi2
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.4.23-16-ev3dev-rpi2 /boot/vmlinuz-4.4.23-16-ev3dev-rpi2
update-initramfs: Generating /boot/initrd.img-4.4.23-16-ev3dev-rpi2
Installing bcm2710-rpi-3-b.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi2
Taking backup of dtb-4.4.23-16-ev3dev-rpi2.
Installing new dtb-4.4.23-16-ev3dev-rpi2.
Installing bcm2710-rpi-3-b.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi2
Taking backup of dtb-4.4.23-16-ev3dev-rpi2.
Installing new dtb-4.4.23-16-ev3dev-rpi2.
flash-kernel: deferring update (trigger activated)
run-parts: executing /etc/kernel/postinst.d/zz-flash-kernel 4.4.23-16-ev3dev-rpi2 /boot/vmlinuz-4.4.23-16-ev3dev-rpi2
Installing bcm2710-rpi-3-b.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi2
Taking backup of dtb-4.4.23-16-ev3dev-rpi2.
Installing new dtb-4.4.23-16-ev3dev-rpi2.
Installing bcm2710-rpi-3-b.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi2
Taking backup of dtb-4.4.23-16-ev3dev-rpi2.
Installing new dtb-4.4.23-16-ev3dev-rpi2.
flash-kernel: deferring update (trigger activated)
Setting up flash-kernel (3.35ev3dev13) ...
````

The uncommented lines of  `/boot/flash/config.txt`
```
dtoverlay=pi3-miniuart-bt
core_freq=250
```
Reboot . . . . 

No success. Did I miss any configuration action?

dlech, 2016-10-16:
It looks like `flash-kernel` did not run automatically like it is supposed to. This should happen when upgrading the kernel. Please try running `sudo flash-kernel` manually.

Also, what is `cat /proc/device-tree/model; echo`?

BertLindeman, 2016-10-16:
`cat /proc/device-tree/model; echo`
```
Raspberry Pi 3 Model B Rev 1.2
```
and on my rpi-b+:
```
Raspberry Pi Model B Plus Rev 1.2
```

Running `sudo flash-kernel` on the rpi-3:
```
Installing bcm2710-rpi-3-b.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi2
Taking backup of dtb-4.4.23-16-ev3dev-rpi2.
Installing new dtb-4.4.23-16-ev3dev-rpi2.
flash-kernel: installing version 4.4.23-16-ev3dev-rpi2
Will use /dev/mmcblk0p1 as boot device.
Taking backup of kernel7.img.
Installing new kernel7.img.
Taking backup of bcm2710-rpi-3-b.dtb.
Installing new bcm2710-rpi-3-b.dtb.
```
reboot....
`uname -a`
```
Linux rpi3 4.4.23-16-ev3dev-rpi2 #1 SMP Sat Oct 15 13:07:22 CDT 2016 armv7l GNU/Linux
```
So fine.  Thanks @dlech 

Keep this one open until the auto-run of flash-kernel is fixed?

dlech, 2016-10-16:
`flash-kernel` ran automatically for me on the `2016-10-15` snapshot build, so I'm not sure why it isn't working for you automatically. Everything seems to be in order since you were able to run it manually.

There should be a script at `/etc/initramfs/post-update.d/flash-kernel`. This is what causes it to run automatically when updating a kernel. Running `sudo update-initramfs -u` manually should also trigger `flash-kernel`.

BertLindeman, 2016-10-16:
Might this be caused by using `apt  upgrade` in stead of `apt-get upgrade` ?
That installed the updated flash-kernel *and* the kernel in one go.

On my rpi-b+ I now dist-upgrade so only the 16 kernel was installed.
```
Running depmod.
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.4.23-16-ev3dev-rpi /boot/vmlinuz-4.4.23-16-ev3dev-rpi
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.4.23-16-ev3dev-rpi /boot/vmlinuz-4.4.23-16-ev3dev-rpi
update-initramfs: Generating /boot/initrd.img-4.4.23-16-ev3dev-rpi
Installing bcm2708-rpi-b-plus.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi
Installing new dtb-4.4.23-16-ev3dev-rpi.
Installing bcm2708-rpi-b-plus.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi
Taking backup of dtb-4.4.23-16-ev3dev-rpi.
Installing new dtb-4.4.23-16-ev3dev-rpi.
flash-kernel: deferring update (trigger activated)
run-parts: executing /etc/kernel/postinst.d/update-notifier 4.4.23-16-ev3dev-rpi /boot/vmlinuz-4.4.23-16-ev3dev-rpi
run-parts: executing /etc/kernel/postinst.d/zz-flash-kernel 4.4.23-16-ev3dev-rpi /boot/vmlinuz-4.4.23-16-ev3dev-rpi
Installing bcm2708-rpi-b-plus.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi
Taking backup of dtb-4.4.23-16-ev3dev-rpi.
Installing new dtb-4.4.23-16-ev3dev-rpi.
Installing bcm2708-rpi-b-plus.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi
Taking backup of dtb-4.4.23-16-ev3dev-rpi.
Installing new dtb-4.4.23-16-ev3dev-rpi.
flash-kernel: deferring update (trigger activated)
Setting up linux-image-ev3dev-rpi (13) ...
Processing triggers for flash-kernel (3.35ev3dev13) ...
Installing bcm2708-rpi-b-plus.dtb into /boot/dtb-4.4.23-16-ev3dev-rpi
Taking backup of dtb-4.4.23-16-ev3dev-rpi.
Installing new dtb-4.4.23-16-ev3dev-rpi.
flash-kernel: installing version 4.4.23-16-ev3dev-rpi
Will use /dev/mmcblk0p1 as boot device.
Taking backup of kernel.img.
Installing new kernel.img.
Taking backup of bcm2708-rpi-b-plus.dtb.
Installing new bcm2708-rpi-b-plus.dtb.
```
This worked OK after reboot `uname -a` shows
Linux ev3dev-rpi.local 4.4.23-16-ev3dev-rpi #1 Sat Oct 15 12:49:28 CDT 2016 armv6l GNU/Linux

dlech, 2016-10-16:
Sounds plausible. I always use `apt-get`. It could be that `apt-get` also invokes triggers that `apt` does not. When installing multiple packages, the `flash-kernel` run is deferred, so if `apt` doesn't know how to handle that, it could explain the situation.

BertLindeman, 2016-10-16:
Tried the new apt and it shows a nice progress bar.
I am known to be inconsistent just for fun ;-)  
The doc in ev3dev **always** tells us to use `apt-get` so let's blame apt for now and close this issue.
If someone else runs into the same problem, we might re-open.
Thanks @dlech David

closed 2016-10-16 by BertLindeman