ev3dev/ev3dev #763: LEGO EV3 - servo-motor class not found (/sys/class/servo-motor)

EsdrasChaves, 2016-10-20:
# System information:
- I am using: LEGO MINDSTORMS EV3
- My kernel version is: 4.4.24-16-ev3dev-ev3
- My host computer is running: Linux
- I am connecting using: USB

# About my issue:
I'm facing a problem while trying to run some servo-motor C programs. 
It's look like this directory (/sys/class/servo-motor) doesn't exist. I don't know what is the content of this directory, but I imagine it's just to save some "status" of the program that is running, so I imagine I can create it by myself, but it looks like I can create a directory there.
How can I solve this problem of a missing directory?

WasabiFan, 2016-10-20:
What programs are you trying to run?

That directory isn't exactly what you'd imagine to be a normal one, where you can add and remove files. Instead, everything under `/sys/` is created by the operating system (ev3dev) and allows you to talk to the underlying drivers that ev3dev exposes. Essentially, if you plug in a servo motor, that directory will appear and there will be one inside it corresponding to the new motor which allows you to control it. So, if you are trying to run a servo motor script but don't have a servo plugged in, that won't do what you want.

In general, you don't need to use these interfaces yourself. People have [written libraries](http://www.ev3dev.org/docs/programming-languages/) that expose functions in your language of choice for operating on ev3dev's interfaces.

EsdrasChaves, 2016-10-20:
I'm trying to run this example: [https://github.com/in4lio/ev3dev-c/tree/master/eg/servo](url) ,so I'm using one of the c-library you mentioned. I'm using a Large-Servo-Motor-45502. 
I've already tested the sensor, light, tacho-motor examples and this tutorial for tacho-motor using the command-line [http://www.ev3dev.org/docs/tutorials/tacho-motors/](url) and all of them worked without problem.

WasabiFan, 2016-10-20:
Ah, I see. We'll have to get input from @dlech then, because I don't see anything obviously wrong (although I haven't checked the specifics of that script). I can't remember the specific part numbers of the types of motors I am aware of, but is the motor you are using a standard EV3 large motor (the kind that comes in standard kits)? If so, that is actually considered a "tacho" motor; servos are specific separate devices, and as far as I know tacho motors don't also function with the `/sys/class/servo-motor` interface. It's also possible that the device isn't properly auto-detected.

EsdrasChaves, 2016-10-20:
Yes, it's a standard EV3 large motor.
When I run the tacho-motor example it iss detected like: lego-ev3-l-motor. Is there any difference between the functions I use with large-motor and medium-motor?

WasabiFan, 2016-10-20:
Ah, yes; in that case you want to use it as a tacho motor. The medium motor (the other kind of motor that comes with the EV3) is mostly just different in its form factor. Its maximum speed is a bit different and the gearing/torque is different, but from a software perspective they are both tacho motors.

EsdrasChaves, 2016-10-20:
Thanks @WasabiFan , After you told me that the motor I'm using is a tacho one, I read the tacho-motor example again and I saw a variable called LEGO_EV3_M_MOTOR, I just changed it to LEGO_EV3_L_MOTOR and it worked.

Problem solved xD, thanks again

WasabiFan, 2016-10-20:
No problem! If this has been resolved, please close the issue to mark it as such.

closed 2016-10-20 by EsdrasChaves