ev3dev/ev3dev #764: Brickman does not provide proper permissions for controlling the tty when running a user program

dlech, 2016-10-23:
If you try to run a program from the Brickman file browser that manipulates the virtual console (e.g. `/dev/tty`) it will fail with error. For example, trying to change the font will fail.

```
#!/bin/bash

setfont Lat15-Terminus-14 2>set_font.log
```

Will give the `font_test.log`

```
putfont: KDFONTOP: Operation not permitted
```

Or trying to use ev3devKit via python will give an error like...

```
Traceback (most recent call last):
  File "./ui_demo.py", line 44, in <module>
    main()
  File "./ui_demo.py", line 35, in main
    Ev3devKit.console_app_init()
GLib.Error: ev3dev_kit_console_app_console_app_error-quark: Error setting driver (1)
```

cc: @cavenel

dlech, 2016-10-23:
We already have [some code](https://github.com/ev3dev/brickman/blob/893e545456e611e7af125bd99cb9c46ab1015c3b/src/controller/FileBrowserController.vala#L75) that changes the permission of `/dev/tty<num>`, but we probably need to also change permissions of `/dev/tty` as well.

Or figure out a better way to do this in general.

dlech, 2016-10-23:
For example `systemd-run` might be a possibility.

Or we could create a systemd service that uses the file name as a parameter.

dlech, 2016-10-23:
Or use `agetty --autologin robot --noissue` along with `FAKE_SHELL` environment variable.

dlech, 2016-10-23:
Or set `CAP_SYS_TTY_CONFIG` on `openvt` and `chvt`.

This could be useful for running from a remote terminal as well.

cavenel, 2016-10-31:
For now, and for my particular issue, using `sudo openvt -s -w /home/robot/my_program.py` from ssh works perfectly (it opens a new tty and starts my ev3devKit program on the brick without the need of conspy and waits until the program finishes to go back to brickman).
This makes me think that the last option (`CAP_SYS_TTY_CONFIG` on `openvt` and `chvt`) would make sense. But I'm no expert here.

dlech, 2016-10-31:
> This makes me think that the last option (CAP_SYS_TTY_CONFIG on openvt and chvt) would make sense.

Yes, it makes sense. However, I have tried it and it works on `chvt`, but it is not enough for `openvt` because `openvt` has to read/write `/dev/ttyN`, which is owned by `root:root`.

You could try `sudo chmod +s $(which openvt)` and then see if you can run `openvt` without `sudo`.

dlech, 2016-10-31:
> sudo openvt -s -w /home/robot/my_program.py

The problem with this is that you are running `my_program.py` as root, so you could easily mess up something important.

cavenel, 2016-10-31:
Before `sudo chmod +s $(which openvt)`:
```
robot@ev3dev:~$ openvt -s -w -- /home/robot/writer_ev3devKit.py 
openvt: Unable to open /dev/tty5: Permission denied
openvt: ioctl(VT_ACTIVATE): Operation not permitted
```
after:
```
robot@ev3dev:~$ openvt -s -w -- /home/robot/writer_ev3devKit.py 
openvt: Cannot open /dev/tty5 read/write: Permission denied
```

I know the issue with running my program as root, it is not a final solution but it gives me ideas of what works and what doesn't :)

cavenel, 2016-10-31:
If I try this: 
```
sudo openvt -s -w -- sudo -u robot /home/robot/writer_ev3devKit.py
```
then the program runs but ev3devKit fails with the same error as when started from brickman:
```
GLib.Error: ev3dev_kit_console_app_console_app_error-quark: Error setting driver (1)
```

dlech, 2016-10-31:
If you add the `-v` option to `openvt`, it should tell you which vt it is running on. The problem is that root owns `/dev/ttyN` (where N is whatever the `-v` option tells you).

This is essentially the workaround we are currently using in brickman.

cavenel, 2016-11-01:
As seen on https://gitter.im/ev3dev/chat, using https://github.com/ev3dev/ev3dev-tools/issues/1 works to launch a program through `openvp` with user (`robot`) permission.
The complete command used is
```
sudo /bin/openvt --switch --wait -- /bin/bash -c "/usr/bin/tty >/tmp/tmp.bNsW3hkZMI;/bin/chown robot: $(/bin/cat /tmp/tmp.bNsW3hkZMI);/usr/bin/sudo --login --set-home --non-interactive --user=robot -- \"/home/robot/my_program.py\" || sleep 5s"
```
Could this workaround be used from brickman as well somehow?

dlech, 2017-01-14:
Fix is released in `brickman_0.8.1`

closed 2017-01-14 by dlech

cho934, 2019-04-05:
Same issue with ev3dev stretch

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-11-13
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           227.32
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534073F8
BOARD_INFO_TYPE=main
```

Here is the error:
```
robot@ev3dev:~$ setfont /usr/share/consolefonts/Lat7-Terminus12x6.psf.gz 
putfont: KDFONTOP: Operation not permitted
robot@ev3dev:~$ 
```