ev3dev/ev3dev #767: Network Connection

crossroads-robot, 2016-10-29:
Hello friends,

I started using ev3dev, got the image flashed and installed on EV3. Now I can't get past the network stage. I am using mac (10.11.6) and there is no CDC composite device showing up. How do i move forward? I read the issue #471 but they talk about connecting using bluetooth. I powered the bluetooth on EV3dev  but it does not show up on the mac bluetooth scan. Can you help me move past this stage? I know linux and python but not too much about network. It will be great if you guys can give me more details on this stage.

Thanks.

dwalton76, 2016-10-29:
ThePiHut wifi dongle works well. They are about $10 on Amazon

bmegli, 2016-10-29:
@crossroads-robot 

> Is there no way to connect to the ev3 using wifi?

Yes it's possible, most of us use wifi for that.

See:
- [ev3dev networking](http://wasabifan.github.io/ev3dev.github.io/docs/networking/)
- [ev3dev basic networking tutorials](http://www.ev3dev.org/docs/tutorials/)

dlech, 2016-10-29:
The key commit is [this one](https://github.com/ev3dev/ev3dev/issues/471#issuecomment-180946320). You need to edit `ev3dev.txt` to get USB network connection on OS X 10.11 and 10.12. And also, file a bug with Apple or they may never fix it.