ev3dev/ev3dev #768: interface B

modulo7, 2016-10-29:
I am trying to add support for the Lego Dacta Interface B. The Interface B connects to the host serially and provides a large number of sensor and motor ports.

I have managed to write user mode code to talk to it,

I have noticed that the protocol for talking to the interface B is very similar to how the EV3 talks to it's UART sensors.

I tried to make a copy of the Line Discipline driver but couldn't figure out how to make the build system build it. 

I also tried modifying the existing UART driver to try and add interface B support but couldn't figure out how to debug it so couldn't figure out if my changes were making any difference.

I have attached a copy of the user mode code I have written for talking to the interface B.
[ibt1.c.zip](https://github.com/ev3dev/ev3dev/files/560065/ibt1.c.zip)

dlech, 2016-10-29:
Just to make sure I know what you are talking about, you mean [this thing](http://www.lgauge.com/technic/LEGOInterfaceB/9751.htm). Do you have documentation on the communication protocol?

Since this device has many input and output ports, it should really be a `lego-port` class device. Since it is a serial device, a line discipline would be the way to go to make a proper kernel driver. However, we have also thrown around the idea of user-mode sensors and motors. We actually have a driver in place for user-mode sensors. We could do something similar for user-mode ports.

modulo7, 2016-11-04:
>Just to make sure I know what you are talking about, you mean this thing.

Yup

> Do you have documentation on the communication protocol?

There is some documentation at http://www.lgauge.com/technic/LEGOInterfaceB/9751.htm 

That documentation omits the serial port settings, we have found the following settings work.

9600 baud
no flow control
no parity
no modem control lines
one stop bit
8 bit data

burf2000, 2016-11-25:
Now that would be awesome, I done a little work with this in VB6

JorgePe, 2016-11-26:
Hello. I'm doing some "archeology research" and I got myself one of these. Never thought would see someone here suggesting a driver.
So, I also have one Interface B.
Python (on Ubuntu) works fine, through a USB to RS232 converter and a bunch of gender changers and a null modem cable. Just using pyserial I can already control all motors, didn't test it in ev3dev yest but I see no reason to fail.
You can count with me to help (at least testing it for you because coding a driver is still something from a different multiverse).
I have also some old sensors somewhere, I think most will work with the Interface B, have to check.