ev3dev/ev3dev #769: Gitter Topics

dlech, 2016-10-31:
Something new to play around with...

gitter.im is developing a new forum-like feature. Try it out at <https://gitter.im/ev3dev/topics>.

WasabiFan, 2016-10-31:
Ooh, that looks cool!

dwalton76, 2016-11-01:
I made a testing123 topic...does gitter link back to the ev3dev github repo somehow?

WasabiFan, 2016-11-01:
What do you mean? I see your topic and it lets me reply to it, but there doesn't seem to be much integration. It's pretty much just a separate forum.

dwalton76, 2016-11-01:
It made me grant access from my github account and I'm trying to figure out why

dwalton76, 2016-11-01:
On a related note has slack been considered for this sort of thing?

WasabiFan, 2016-11-01:
Why would we use Slack? It's functionally the same as Gitter, except that Gitter is free for public use while Slack is the inverse.

dwalton76, 2016-11-01:
slack is free unless you want to archive more that 10k messages, I imagine it would take us a long time to hit that limit (and would we care if we did).  I mention slack because it is pretty easy to use and it does integrate with github.

WasabiFan, 2016-11-01:
We have a Gitter account (with a general room) [here](https://gitter.im/ev3dev/chat/). Gitter is built specifically for GitHub, so it has integration at a deeper level than Slack. Gitter is also completely free and unlimited for public use, meaning that anyone can follow a link to ask us questions in the chat room. Slack, on the other hand, is only free as you mentioned for _private_ use; that means that maintainers could be added but it wouldn't work as a place to chat with users. It also requires a separate GitHub plugin, meaning you don't get automatic GitHub links or snapped log.

I work with both services on an hourly basis, so I know how they work. We use Gitter because it serves the specific purpose that we needed.

The topic of this issue is a new feature called _Topics_; this feature adds a public forum to Gitter. This is separate from standard chat and something that Slack does not have.

dwalton76, 2016-11-01:
From the ev3dev repo here how do I see the testing123 topic that I created in gitter...or is it only integrated from github->gitter but not gitter->github?

dwalton76, 2016-11-01:
I'm not saying I'm against Gitter, I'm just trying to understand exactly how it integrates with github.

dlech, 2016-11-01:
You can't see anything from gitter on the github site. You can see stuff from github on the gitter site though.

WasabiFan, 2016-11-01:
It's also important to understand that the "testing123" topic is a _Topic_, which is a beta feature only recently introduced. The primary functionality of Gitter is standard chat.

dlech, 2017-10-03:
It doesn't seem that this ever really took off.

closed 2017-10-03 by dlech