ev3dev/ev3dev #770: How to write hardware controlling libraries?

crossroads-robot, 2016-10-31:
One question. I am not a software engineer but a biomedical engineer. I know linux and Python.  I want to know how were these libraries written to control the hardware like motors, sensors.  What kind of knowledge do i need to write such hardware controlling library. I was planning to see if i can get any ultrasonic sensor (not lego) and interface it with ev3. is that possible?

WasabiFan, 2016-10-31:
For the majority of users, we recommend using a pre-made library from [our "programming languages" page](http://www.ev3dev.org/docs/programming-languages/). That lets you just follow some quick install instructions and get started right away.

If you are intent on doing it yourself (the hard way), you'll need some perseverance, because this isn't the common path that we make sure is easy. This requires some knowledge of file I/O APIs, but with the way that ev3dev is designed it isn't too complex. We have a [landing page](http://www.ev3dev.org/docs/driver-overview/) for this, however it doesn't explain everything. You'll probably need to look at how the existing libraries do some things.

dlech, 2016-10-31:
> I was planning to see if i can get any ultrasonic sensor (not lego) and interface it with ev3. is that possible?

Sure it is most likely possible. In fact, we have special modes for the input ports for just such an application.

> What kind of knowledge do i need to write such hardware controlling library.

First, you need to know how the sensor communicates. Does it use I2C, UART or have an analog output? If not, you will need a way to convert the signal to one of these.

Then you need to know the electrical specifications. The EV3 uses 3.3V I/O, but it is 5V tolerant. If the ultrasonic sensors uses something > 5V, then you will need to design a circuit to fix this. For I2C, you will also need 82k pullup resistors (I think).

Once you know this info, you can design an adapter cable for your sensor and connect it to the EV3.

The code depends on the type of communication. For UART, you will read and write `/dev/tty_inX` directly, for I2C `python3-smubs` is useful and for analog, you can use the generic analog driver via `python3-ev3dev`.

crossroads-robot, 2016-10-31:
Thanks for your feedback. Looks like a big project. For the lego hardware, how did you guys write the libraries. Is there like a file that the hardware writes to that the python library reads or is it more involved than that. Just trying to understand things.  Suppose i want to write a library for lego touch sensor, how would i go about doing that?

WasabiFan, 2016-10-31:
When you say "library", what do you mean exactly? The word "library" has a specific programming meaning, and I think what you are actually trying to do is just write your own code to use EV3 devices.

For that, it's as I said in the first paragraph of my comment. Visit [this page](http://www.ev3dev.org/docs/driver-overview/) and choose the programming language that you would like to use -- it sounds like you are looking at Python. The link will take you to the appropriate place with instructions on getting started.

P.S. For Python beginners, I recommend checking out [ev3python.com](ev3python.com) .

crossroads-robot, 2016-10-31:
Thank you. I will check that out.

dlech, 2016-10-31:
> For the lego hardware, how did you guys write the libraries

First, we wrote [kernel drivers](https://github.com/ev3dev/lego-linux-drivers) that expose the hardware as attribute files via [sysfs](https://www.kernel.org/doc/Documentation/filesystems/sysfs.txt).

The libraries then use these sysfs attributes to monitor and control the hardware. The kernel drivers are documented [here](http://www.ev3dev.org/docs/driver-overview/). The libraries are written by applying the information in the docs to the language of your choice.

> Suppose i want to write a library for lego touch sensor, how would i go about doing that?

For a touch sensor, you will need to know about the [lego-sensor class](http://www.ev3dev.org/docs/drivers/user-lego-sensor/) and the [touch sensor](http://www.ev3dev.org/docs/sensors/lego-ev3-touch-sensor/) itself. You need to write code that opens the sysfs attribute files and read/write to them as  described in the driver docs. Each attribute serves a specific purpose. The main one you would be interested in for a touch sensor would be the `value0` attribute of the `lego-sensor` class. So, you might read the file `/sys/class/lego-sensor/sensor0/value0` and find that it has the value `0\n` (`\n` is a newline character).

There are several python libraries for ev3dev already. You can have a look to see how they did it.
* [ev3dev-lang-python](https://github.com/rhempel/ev3dev-lang-python)
* [python-ev3](https://github.com/topikachu/python-ev3/)

crossroads-robot, 2016-11-01:
Thank you guys.

bmegli, 2016-11-01:
> I was planning to see if i can get any ultrasonic sensor (not lego) and interface it with ev3. is that possible?

If you have some specific device on your mind it would help if you shared the link.

Not knowing what the device is it can be anything from trivial to rather involved.

You can get hardware specs of EV3 ports in [EV3 Hardware Developer Kit](https://education.lego.com/en-au/learn/middle-school/mindstorms-ev3/support/ev3-developer-kits) (the last one)

You can see example of interfacing device not designed for Lego in [Using-xv11-lidar] (http://www.ev3dev.org/docs/tutorials/using-xv11-lidar/) tutorial.

In this case:
- the communication interface was compatible electrically (e.g. UART, 8N1 at 3.3V)
- there wasn't enough power for the laser in the 5V port pin so the power was drawn from USB (you can get pin power information in  [EV3 Hardware Developer Kit](https://education.lego.com/en-au/learn/middle-school/mindstorms-ev3/support/ev3-developer-kits))
- the motor was compatible electrically and could be used as ev3dev `dc-motor`
- on the software level ev3dev has `other-uart` mode for ports which lets you take care of UART communication from your programs (this means "just" some reads/writes)

But this is just an example, depending on hardware you are interested in the picture may be totally different.

crossroads-robot, 2016-11-01:
Looks like this stuff is not for people with my background. I should stick with python programs and the drivers you have already built. Thanks for the info.

bmegli, 2016-11-01:
And finally don't forget that the hypothetic ultrasonic sensor (not lego) may already have the linux drivers written which may also work with ev3dev.

closed 2019-03-25 by dlech