ev3dev/ev3dev #772: Brickman cannot launch programs in directories with spaces in the name

rhempel, 2016-11-06:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3dev-jessie-ev3-generic-2016-10-17.zip

### About my issue:

I am able to run a script in a directory under `/home/bobot/...` but when I run the same script under Brickman the script never even starts.

I can't even use the `stdout` and `stderr` redirects to log the output because the script never even starts.

As a last resort I ended up using my phone camera to take a shot of the screen as the error message passed by, and then it became obvious:

```
~bash: line 0: cd: /home/robot/PYTHON: No such file or directory
```

The actual path was `/home/robot/PYTHON WEBSERVER` ;-)

Changing the space to an underscore made everything work perfectly.

So Brickman needs to quote the path that the script is being run in

rhempel, 2016-11-06:
Here's the line in Brickman's `FileBrowserController.vala` that needs to change:

```
                                "/usr/bin/tty >%s;tty=$(/bin/cat %s);/bin/chown %s: \$tty;/sbin/runuser --login --shell /bin/bash --command '(cd %s && exec %s)' %s"
```

The `%s` for the script directory and name should be quoted

dlech, 2016-11-06:
Related: #764

dlech, 2017-01-14:
Fix is released in `brickman_0.8.1`

closed 2017-01-14 by dlech