ev3dev/ev3dev #773: apt-get update freezing

daniilS, 2016-11-06:
### System information:

* I am using: EV3

* I installed ev3dev using this image file: *IMAGE NAME GOES HERE*
ev3dev-jessie-ev3-generic-2016-10-17

* My host computer is running: *OS GOES HERE*
Windows 10, but trying to ssh from Android or running ```apt-get update``` from a file through brickman gives the same problem

### About my issue:

Halfway into running ```apt-get update```, the process gets stuck, oftentimes on ```[Waiting for headers]```. Sometimes the connection will be lost a bit later. Example console output:

```
robot@ev3dev:~$ sudo apt-get -m update
Hit http://archive.ev3dev.org jessie InRelease
Hit http://security.debian.org jessie/updates InRelease
Get:1 http://archive.ev3dev.org jessie/main armel Packages [41.1 kB]
Get:2 http://security.debian.org jessie/updates/main armel Packages [392 kB]
Ign http://httpredir.debian.org jessie InRelease
Get:3 http://security.debian.org jessie/updates/contrib armel Packages [994 B]
Get:4 http://security.debian.org jessie/updates/non-free armel Packages [20 B]
Hit http://httpredir.debian.org jessie Release.gpg
Hit http://httpredir.debian.org jessie Release
Get:5 http://httpredir.debian.org jessie/main armel Packages [8866 kB]
15% [5 Packages 950 kB/8866 kB 11%] [Waiting for headers]      116 kB/s 1min 8s
```

```
robot@ev3dev:~$ sudo apt-get -m update
Hit http://security.debian.org jessie/updates InRelease
Hit http://archive.ev3dev.org jessie InRelease
Ign http://httpredir.debian.org jessie InRelease
Hit http://httpredir.debian.org jessie Release.gpg
Hit http://httpredir.debian.org jessie Release
Get:1 http://security.debian.org jessie/updates/main armel Packages [392 kB]
45% [InRelease gpgv 8415 B] [1 Packages 176 kB/392 kB 45%]
```

daniilS, 2016-11-06:
Update: I tried connecting through USB instead of the Wi-Fi dongle I used before, and it seems to work now. Seems like something was wrong with it then.

closed 2016-11-06 by daniilS

dlech, 2016-11-06:
Sounds like #758.