ev3dev/ev3dev #775: pybluez missing gattlib (required for BLE)

JorgePe, 2016-11-08:
Hi.
I'm using yesterday snapshot, 07Nov2016, ev3dev 4.4.24-16-ev3dev-ev3, no updates and nothing else installed.
I'm trying pybluez since it is now included with ev3dev. I'm interested in BT 4.0 BLE communication.

pybluez README has a few examples for BLE but they only work if we have also gattlib (pygattlib)

I tried to install gattlib according to the README instructions but failed:
`Directory '.' is not installable. File 'setup.py' not found.`

In the past I've managed to install gattlib compiling from source but it takes a lot of time and requires a huge swapfile (process is described here: http://www.ev3dev.org/docs/tutorials/controlling-wedo2-motor/)

Is it possible to include gattlib with pybluez in future snapshots?

JorgePe, 2016-11-09:
Just a quick note: the reason for installation faillure seems the same, out of memory.
So I added a ~900 MB swap again, but installation keeps failling. But direct installation of gattlib works:
`sudo pip3 install gattlib`
(pip also works, for python 2.x version)

900 MB seems overkill but just one of the components, "bindings", makes cc1plus to allocate more than 580 MB!

After successful installation, I can now use bluetooth.ble:

`from bluetooth.ble import DiscoveryService`

(and of course also import gattlib directly).

Since I'm planning a few other tutorials on BLE devices (LEGO WeDo 2.0, TI SensorTag, Vengit SBrick/SBrick+ and probably BBC micro:bit) I'm considering to split the WeDo 2.0 tutorial I already made in two, so the mess of pyhton + BLE can be in a separate tutorial that all others can link to.
(sorry for my english).

dlech, 2017-01-17:
I have uploaded a new `python3-gattlib` package to the ev3dev repositories. Please test.

JorgePe, 2017-01-17:
@dlech Great work!!!

Snapshot image from 17 Jan 2017, no updates whatsoever
Just enabled Bluetooth in the Brickman and checked with hciconfig -a that my USB BT4-BLE dongle is hci0
First example from the README.md works at first attempt (just added shebang):
```
#!/usr/bin/env python3
from gattlib import DiscoveryService
service = DiscoveryService("hci0")
devices = service.discover(2)
for address, name in devices.items():
    print("name: {}, address: {}".format(name, address))
```

chmod +x test01.py
sudo ./test01.py

`name: LPF2 Smart Hub 2 I/O, address: A0:E6:F8:1E:37:B5`

Now I just need some time to rewrite and retest my scripts and the update the tutorials.

JorgePe, 2017-01-17:
@dlech sorry, spoke to soon

python-gattlib version seems to be to old - it can discover BLE devices but cannot connect to them (at least I cannot).

I tried to use my basic WeDo 2.0 motor script but it complains about arguments - compiled version accepts choosing the hci device but ev3dev packaged doesn't (it was merged to the code 6 months later).
If I ignore the hci device the script works (as long as my BLE dongle stays hci0) but I never get a connection and the WeDo 2.0 SmartHub powers off.
Tried low level example from source (from ev3dev source)  and it fails with:

`RuntimeError: Channel or attrib not ready`

Maybe I can create a .deb package from present source code but I don't know how to integrate it on ev3dev ecosystem so it will probably break next time bluez or any other library is updated.

dlech, 2017-01-17:
I was wondering about that. It looks like the PyPi version is the same old version, so I hoped that it would be good enough.

I created the package from https://github.com/ev3dev/gattlib, so if you want to update that to a newer version of gattlib, be my guest. You can read more about packaging [here](http://www.ev3dev.org/docs/devtools/packaging-for-ev3dev/). If there is a tarball of a newer version of pygattlib, I would suggest using `gbp import-orig`.

JorgePe, 2017-01-17:
Thanks, have to find some time alone to read it.
Pypi seems 0.20150805 while seems 0.20150131
Funny how just half an year can make quite a difference.

dlech, 2017-01-17:
It was easy for me, so I've gone ahead and started building a package based on the `0.20150805` version.

dlech, 2017-01-17:
`python3-gattlib_0.20150805-1ev3dev1` is now pushed. Please test.

closed 2017-01-17 by dlech

JorgePe, 2017-01-17:
@dlech Very nice, motor running!

satish1dalvi, 2021-11-02:
Hi

I am not able to connect to ble device using gattlib0.20201113; error appearing like "gattlib.BTIOException:Channel or attrib not ready".
Gone through few support community however, could not find solution. 
Can somebody help please?

Regards,
Satish