ev3dev/ev3dev #776: Error compiling on docker c++ cross compiler

volansky, 2016-11-10:
Hey guys, I'm converting an arduino program to the ev3 using a docker c++ cross-comipler.
After debugging for a while I came a cross an error from one of the ev3dev files (modp_numtoa.c), as I know the origin of this error most be from my code (or maybe from running the c++ compiler?) I can't really find it or don't want to change the modp file without asking about it first.. 
Anyone knows what could flag this or how to find and fix?

Error message:

> make: Warning: File '../../3d_party/modp_numtoa.c' has modification time 18 s in the future
arm-linux-gnueabi-g++ -I. -I../../source/ev3 -I../../3d_party -O2 -std=gnu99 -W -Wall -Wno-comment -c ../../3d_party/modp_numtoa.c -o ../../object/modp_numtoa.o
cc1plus: warning: command line option '-std=gnu99' is valid for C/ObjC but not for C++
../../3d_party/modp_numtoa.c:26:27: error: 'const double pow10 []' redeclared as different kind of symbol
 static const double pow10[] = {1, 10, 100, 1000, 10000, 100000, 1000000,
                           ^
In file included from /usr/include/features.h:374:0,
                 from /usr/include/stdint.h:25,
                 from /usr/lib/gcc/arm-linux-gnueabi/4.9/include/stdint.h:9,
                 from ../../3d_party/modp_numtoa.h:37,
                 from ../../3d_party/modp_numtoa.c:4:
/usr/include/arm-linux-gnueabi/bits/mathcalls.h:122:1: note: previous declaration 'double pow10(double)'
 __MATHCALL (pow10,, (_Mdouble_ __x));
 ^
../../3d_party/modp_numtoa.c: In function 'char* modp_dtoa(double, char*, int)':
../../3d_party/modp_numtoa.c:121:46: warning: pointer to a function used in arithmetic [-Wpointer-arith]
     double tmp = (value - whole) * pow10[prec];
                                              ^
../../3d_party/modp_numtoa.c:121:46: error: invalid operands of types 'double' and 'double(double) throw ()' to binary 'operator*'
../../3d_party/modp_numtoa.c:128:31: warning: pointer to a function used in arithmetic [-Wpointer-arith]
         if (frac >= pow10[prec]) {
                               ^
../../3d_party/modp_numtoa.c:128:31: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]
../../3d_party/modp_numtoa.c: In function 'char* modp_dtoa2(double, char*, int)':
../../3d_party/modp_numtoa.c:223:46: warning: pointer to a function used in arithmetic [-Wpointer-arith]
     double tmp = (value - whole) * pow10[prec];
                                              ^
../../3d_party/modp_numtoa.c:223:46: error: invalid operands of types 'double' and 'double(double) throw ()' to binary 'operator*'
../../3d_party/modp_numtoa.c:230:31: warning: pointer to a function used in arithmetic [-Wpointer-arith]
         if (frac >= pow10[prec]) {
                               ^
../../3d_party/modp_numtoa.c:230:31: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]
makefile:242: recipe for target '../../object/modp_numtoa.o' failed
make: *** [../../object/modp_numtoa.o] Error 1

cavenel, 2016-11-10:
It complains that `pow10` (in `modp_numtoa.c`) is already defined in `/usr/include/arm-linux-gnueabi/bits/mathcalls.h`. I believe this is because you included `math.h` somewhere in your code.
In other versions of `modp_numtoa.c` (see https://github.com/bro/bro/blob/master/src/modp_numtoa.c), `pow10` (in `modp_numtoa.c`) is replaced by `_pow10` (to avoid this I guess).

The simplest fix is thus to replace all `pow10` in `modp_numtoa.c` with `_pow10`.

volansky, 2016-11-10:
Yea that worked! thanks for the quick respond..

Hope you don't mind I got another one coming from ev3.c this time

> ../../source/ev3/ev3.c:112:4: error: invalid conversion from 'void*' to 'char*' [-fpermissive]
  p = buf;
    ^
../../source/ev3/ev3.c:123:24: error: invalid use of 'void'
  return (( void *) p - buf );
                        ^
../../source/ev3/ev3.c: In function 'size_t ev3_write_bool(const char*, bool)':
../../source/ev3/ev3.c:267:56: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
  return ev3_write_binary( fn, ( value ) ? "1" : "0", 1 );
                                                        ^
../../source/ev3/ev3.c:43:8: note: initializing argument 2 of 'size_t ev3_write_binary(const char*, char*, size_t)'
 size_t ev3_write_binary( const char *fn, char *data, size_t sz )
        ^
../../source/ev3/ev3.c: In function 'size_t ev3_multi_write_bool(uint8_t*, uint16_t, const char*, bool)':
../../source/ev3/ev3.c:272:71: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
  return ev3_multi_write_binary( sn, pos, fn, ( value ) ? "1" : "0", 1 );
                                                                       ^
../../source/ev3/ev3.c:56:8: note: initializing argument 4 of 'size_t ev3_multi_write_binary(uint8_t*, uint16_t, const char*, char*, size_t)'
 size_t ev3_multi_write_binary( uint8_t *sn, uint16_t pos, const char *fn, char *data, size_t sz )
        ^
../../source/ev3/ev3.c: In function 'size_t __ev3_listdir(char*, void*, size_t)':
../../source/ev3/ev3.c:124:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^

cavenel, 2016-11-10:
That's because you are compiling C code with C++ compiler. It is never a good idea... See http://stackoverflow.com/questions/18879344/c-error-invalid-conversion-from-void-to-char.
Some solutions are provided in the link above.
I think all these errors are more related to https://github.com/in4lio/ev3dev-c so maybe you should open issue there.

volansky, 2016-11-10:
I was trying to compile it with the c compiler but it raised a lot of errors as there are parts of c++ in the arduino coding so I'm not quite sure now which one is better for usage.

and I'll look into it, thanks again for the help!

dlech, 2016-11-10:
Tip: code fences (```` ``` ````) are better than quotes (` > `) for code and terminal output. I edited your comments to make them easier to read.

closed 2019-03-25 by dlech