ev3dev/ev3dev #777: [HELP]SSH login :Permission denied, please try again.

LEXUGE, 2016-11-13:
`
[lexuge@lexuge-pc ~]$ ssh robot@10.42.0.31  
robot@10.42.0.31's password:   
Permission denied, please try again.  
robot@10.42.0.31's password:   
Permission denied, please try again.  
robot@10.42.0.31's password:   
Permission denied (publickey,password).  
`
It is the log ,I input the password "maker".It is corect
What should I do now?

dlech, 2016-11-13:
What image file did you use to flash your SD card?

Have you been able to successfully connect in the past?

LEXUGE, 2016-11-13:
@dlech I have already fix it.My EV3DEV is 0.8.0. now.
I want to connect the Internet with the Bluetooth.But,when I connected EV3 by Linux,few seconds ago,It will disconnect.I don't know why that.When I connect PC by EV3.It said "GDBUS:No such file or dictionary".And I also can't use the "apply" button  when I edit the option in PC.

dlech, 2016-11-13:
What did you do to fix it?

0.8.0 sounds like the ev3dev-lang-python version, not the version of the image file. (Image files use the date as a version).

Please open a new issue if you have another problem and explain the exact steps that you took that lead up to the problem.

WasabiFan, 2016-11-13:
> 0.8.0 sounds like the ev3dev-lang-python version

I'd guess it is the Brickman version as shown in the "about" pane on-brick. @LEXUGE, the information about ev3dev itself is lower down on that screen; you can scroll with the up/down buttons.

closed 2016-11-21 by dlech