ev3dev/ev3dev #778: Code design in Brickman

WasabiFan, 2016-11-13:
I recently decided to look into parts of #737 with the intent of looking at #456 in the future. However, while working on the "executable bit" part of the former, I realized that almost all of the code in the "fake" version of the controller is copied from the real one. From a diff of the two files, I can see that they've only diverged in two ways: alterations for the complexities of running GUI apps on the brick and various inconsequential changes that result from independently editing two copies of a file (e.g. moving all the code from the window creation method into the constructor, adding a comment, and adding some extra error handling).

Digging deeper, the "file browser window" is really just a window with a `Menu` control in it: there's nothing other than naming that makes it a non-generic control.

At this point, the simple change I wanted to make would require an ugly design hack, and the whole thing smells like broken code. There's a "preferred" design in my mind, but it would require adding the equivalent of view-models, which @dlech has said he doesn't want in Brickman. So, how should I approach this?

My preferred design would be as follows:
- A new `FileMenuItem` class would be created which inherits from `MenuItem` but takes a `File` object. It would handle creating the string representation of the file for display. It would watch the file it was given for changes and update itself accordingly.
- A new `FileMenu` class would be created which inherits from `Menu` and contains `FileMenuItem`s. It would have a "current directory" and would watch that directory for file additions and deletions to add and remove files.
- The `FileBrowserWindow` would be modified to contain a `FileMenu`. It would take an initial directory and publicize controls for its current one. It would also watch for a selection signal to be emitted and handle updating the current directory and/or emitting a "file selected" signal accordingly.
- The controllers would be responsible for watching for the "file selected" signal and handling the execution of those files as is appropriate.
- For handling the feature additions mentioned above, new dialog classes would be created for things like asking the user about executable bits and file types, and those would be used in each controller.

It's easily possible that I am missing something, but you get the gist. The idea would be to separate out the logic into the most largely-shared section of code that would be reasonable. I think that such a design would make maintenance easier _and_ help new contributors, which are both major pluses in my mind.

However, this goes in opposition to the previously-stated principles from @dlech, so I don't want to make any drastic changes without discussion just to have more changes sitting unmerged. This does smudge the lines a bit between views and controllers, but splitting that out more would require going against the "no-models" principle even further, and I don't think that the change would be particularly beneficial when considering the circumstances. What can we do about this? Is there a better design that I am not considering? Or should I just hack it together and be done?

dlech, 2016-11-13:
First, there are some other things we need to discuss before getting into the code design of brickman.

The first thing is #522. If we decide to go ahead with ev3dev-stretch, I am going to be more open to making drastic changes to brickman.

The second thing is that before getting into the coding details, I would like to see a mock up (hand-drawn or otherwise) of what this new feature is going to look like and a detailed, high-level description of how it is supposed to work.

I'm not really sold on the utility of being able to set the executable bit from brickman at this point. If a person does not know about executable bits, then that person probably also does not know about Windows line endings, so their program will still not run, and they will still not know why.

I think it is much more useful to show how to set the executable bit and line endings using the tools on your desktop computer before files ever get to the EV3.

dlech, 2016-11-13:
Regarding the duplicate code. The fake file controller code is copied, but that is wrong because it is not really a fake controller. All of this code needs to be replaced with code that can supply an arbitrary file/directory tree via the brickman test control panel. The whole purpose of the fake controllers is to be able to test every aspect of the views. It is not intended at all to test the actual behavior of brickman. The only brickman behavior that is intentionally replicated is navigation, merely for convince.

WasabiFan, 2016-11-13:
> First, there are some other things we need to discuss before getting into the code design of brickman.

Agreed. I have not actually implemented anything and do not intend to do so until we design these features: I got my environment set up and started messing around with the code to make sure that I wouldn't get stuck after spending time designing the thing.

> The first thing is #522. If we decide to go ahead with ev3dev-stretch, I am going to be more open to making drastic changes to brickman.

I'm not sure I follow. If we _don't_ go ahead with Stretch, what would we be doing? And how does that affect whether we add more features to Brickman now? Is the idea that we would be destabilizing it, and you want to attempt to keep it in a known-good state?

> I'm not really sold on the utility of being able to set the executable bit from brickman at this point. If a person does not know about executable bits, then that person probably also does not know about Windows line endings, so their program will still not run, and they will still not know why.

I am operating with the goal of making ev3dev usable by beginners. Shebangs, line endings and executable bits are all things which can mess with a newbie, so my intent is to have Brickman or other ev3dev infrastructure alleviate those issues. I could see this happening in multiple ways; attempting to execute a file suffering from one of the aforementioned problems could show a dialog which allowed you to fix the issue or included a URL for the site where we would explain how to fix it manually, for example. The solution might also vary depending on which of those aspects we're looking at, as the executable bit is really just a one-time thing that you do on the brick; I wouldn't expect someone's editor to do it, but an SCP app might. This is more of a discussion for the original issue, I think; my main focus here is on the design of Brickman code in general.

----

Given that Brickman hasn't been modified in half a year, I am assuming that we won't be creating a "real" fake controller for a while. Doing so is a lot of work (one must create a usable desktop interface and link it to the internal structure properly) yet doesn't provide much benefit over just creating real files on your hard drive; I think our time is much more valuably spent implementing new features and fixing bugs. Operating under that assumption, we can still design around the expectation that it might happen at some point. I'm open to any design that doesn't duplicate significant chunks of code; this means that the file stringification logic, file sorting, error/warning boxes and whatever new features we add should all be common between the two. I'd be happy with a design which had an `IFileBrowserFile` (or similar) interface and implementations backed by varying sources, likely even more so than I would be with my original proposal. However, this is clearly a view-model.

My goal is to make this code maintainable in the state that it is currently, because I honestly doubt that it will change significantly in function in the near future.

dlech, 2016-11-13:
> I'm not sure I follow. If we don't go ahead with Stretch, what would we be doing?

Continue making major changes to ev3dev-jessie.

> And how does that affect whether we add more features to Brickman now?

It determines if the changes get applied to the jessie branch or stretch branch.

> Is the idea that we would be destabilizing it, and you want to attempt to keep it in a known-good state?

Pretty much. So far, ev3dev has been a quickly moving target and no one has been able to build anything on it without a significant amount of work to keep up with the changes. It's not so much about a known *good* state, but just a known state period.

> This is more of a discussion for the original issue, I think; my main focus here is on the design of Brickman code in general.

Like I said, open some new issues, draw some pictures and show us your grand plan for the high level changes, then we will come back to the issue of how to implement them. (And hopefully by that time, I will have got back into the brickman code and will be able to make better decisions here).

dlech, 2016-11-13:
> Given that Brickman hasn't been modified in half a year, I am assuming that we won't be creating a "real" fake controller for a while.

Bad assumption. There have just been many, many other more pressing issues. Now that most of those have been addressed, I am hoping to be able to spend some more time on the brickman/ev3devKit/GRX stack over the next 3-6 months.

> Doing so is a lot of work (one must create a usable desktop interface and link it to the internal structure properly) yet doesn't provide much benefit over just creating real files on your hard drive;

It's probably not as bad as you think. :wink:

> I'm open to any design that doesn't duplicate significant chunks of code; this means that the file stringification logic, file sorting, error/warning boxes and whatever new features we add should all be common between the two.

Like I said, the fake controller should not be handling behavior, so should not have any of these things at all. Any common code is most likely wrong and should be removed rather than refactored.

> I'd be happy with a design which had an IFileBrowserFile (or similar) interface and implementations backed by varying sources, likely even more so than I would be with my original proposal. However, this is clearly a view-model.

If you can come up with a roadmap on how to convert all of brickman to using the same pattern, then I will have another look at it. But, having never done any significant amount of MVVM programming myself, "clearly view-model" is not very clear to me.

WasabiFan, 2016-11-14:
OK, I'll draw up some diagrams and see how it looks.

> Bad assumption. There have just been many, many other more pressing issues. Now that most of those have been addressed, I am hoping to be able to spend some more time on the brickman/ev3devKit/GRX stack over the next 3-6 months.

I understand that, however working on Brickman likely doesn't (or at least _shouldn't_ in my mind) mean adding a fake file tree to the desktop test. It seems like such a feature should be prioritized far below most other things on the list, because it doesn't affect a user and only makes it marginally easier to get file browser screenshots. I get the idea, though; whatever is done to this code should at least accommodate the possibility of switching backing data sources.

> Like I said, the fake controller should not be handling behavior, so should not have any of these things at all. Any common code is most likely wrong and should be removed rather than refactored.

In my mind, the test UI control panel is there to replace hardware access and system configuration: it emulates motors and sensors, networking, and presumably the file system in the future. But the core logic on top of that data source should still be common across both platforms, right? If I declare that there is a motor connected to port B and it has rotated 100 degrees, I would expect the on-screen UI to render exactly what would be seen on the brick in the same circumstances, using the same internal logic. Based on your comment, however, it doesn't seem that you think of it in the same way. What do you intend the fake infrastructure to be used for?

> If you can come up with a roadmap on how to convert all of brickman to using the same pattern, then I will have another look at it. But, having never done any significant amount of MVVM programming myself, "clearly view-model" is not very clear to me.

While I think it's helpful to leverage these named patterns for reference, there really is no strict definition of how any of them operate. As I implement systems like this, I choose the architecture that fits best for each particular aspect such that interactions are logical and code feels "clean." In the case of Brickman, there are some places which are well-suited to the "View-Controller" approach that is currently used, and there are others where such a model produces ugly or difficult-to-maintain code. Uniformity is, so some extent, important -- I believe that every component should have views and a controller -- yet I think that the minutiae of every module don't need to be the same just for the sake of sticking to a named pattern.

dlech, 2016-11-14:
> In my mind, the test UI control panel is there to replace hardware access and system configuration: it emulates motors and sensors, networking, and presumably the file system in the future. But the core logic on top of that data source should still be common across both platforms, right?

Wrong. The test UI is only meant to test the UI widgets (everything in the view folder in the brickman source and nothing else). Think of it like preview of a visual designer in an IDE. It looks like what you should see when you really run the program, but it has none of the actual behaviors of the program.

> What do you intend the fake infrastructure to be used for?

The fake infrastructure is merely to feed values into view properties so we can test things like... What happens if I have a really long string, does it wrap correctly? Does the scroll bar work correctly when we have a long list? Does the custom graphics drawing code for a status bar item look right? etc.

dlech, 2016-11-14:
> there are some places which are well-suited to the "View-Controller" approach that is currently used, and there are others where such a model produces ugly or difficult-to-maintain code

Could you point out some specific examples, please?

WasabiFan, 2016-11-14:
> Wrong. The test UI is only meant to test the UI widgets (everything in the view folder in the brickman source and nothing else) ... It looks like what you should see when you really run the program, but it has none of the actual behaviors of the program.

OK, at least we both understand the existing system. Why is it designed this way? I can't think of a case where you would want the UI _not_ coupled to the presentation logic; if a condition affecting the display of data could potentially appear in production with the real data sources, it must by extension be reproducible with fake ones. If it _can't_ appear in production, you aren't gaining anything by testing it. On the other hand, testing things like sort logic and navigation/UI control is easiest on the desktop where you can simulate edge-cases and see how the full stack, barring I/O, handles it.

> Could you point out some specific examples, please?

Sure.

The [BatteryController](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/0.8.0/src/controller/BatteryController.vala) works well with the existing pattern. There is a single state of the system and there aren't any individual components that the user can select to "drill into," so the controller can simply dump state into the view.

A simple example of this structure gone wrong is the [FileBrowserController](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/0.8.0/src/controller/FileBrowserController.vala). You can see that there's a bunch of logic _in the controller_ which takes a lower-level `File` object and converts it to a string. This presentation logic is intermingled with file I/O, permissions checks and TTY handling. Sections like this worry me:

```vala
var mode = file_info.get_attribute_uint32 (FileAttribute.UNIX_MODE);
if (file_info.get_file_type ()  == FileType.DIRECTORY) {
    // add '/' to the end of directories
    file_name += "/";
 } else if ((mode & Posix.S_IXUSR) == Posix.S_IXUSR) {
    // add '*' to the end of executable files
    file_name += "*";
}

```

The file has a lot of mixed code like this, and following the chain of handlers and registration of notifications is difficult.

This is more extremely demonstrated by the [DeviceBrowserController](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/0.8.0/src/controller/DeviceBrowserController.vala). The file is full of `printf`s and other display logic, which makes it very difficult to read. At some points in the file, there are sections (display logic no less!) which are nested six levels deep. Again, the chains of logic for `connect`s and `disconnect`s are very difficult to reason through. I have a hard time imagining a new contributor being able to modify this chain without running into cases where they forget to change event registration and then can't figure out how to fix it.

Looking toward potential _solutions_, It seems like there are a few design issues that are compounding. Firstly, there is no distinction between the internal logic and the user interface. The general idea behind patterns such as MVVM and MVC is that they are the _presentation layer_ built on top of _business logic_, whereas here the controllers have I/O built in. Second, there is no concept of a data container; despite the use of an object-oriented language, it reads much like unstructured C. Lists of items aren't stored as lists, and selected items are passed around through closures and deeply-nested lambdas instead. There are also some miscellaneous things, such as loops calling event handlers to populate lists, which make the code difficult to interpret.

dlech, 2016-11-14:
> Why is it designed this way?

Because I found that testing "does it look good if I try this" on the desktop to be much more useful than testing "does it work just like it should on the EV3". You can quickly check iterations of graphical things like does this look better with a margin of 2 or 3 on the desktop without having to setup all sorts of stuff to emulate what the EV3 is doing.

> On the other hand, testing things like sort logic and navigation/UI control is easiest on the desktop where you can simulate edge-cases and see how the full stack, barring I/O, handles it.

I have never felt that there is a need for something like this up to this point.