ev3dev/ev3dev #779: [question] RT-Preempt patch

fgr, 2016-11-15:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3dev-jessie-ev3-generic-2016-10-17

* My kernel version is: 4.4.24

* My host computer is running: OS X

* I am connecting using: Wi-Fi

### About my issue:

I'm interested in running ev3dev (for Lego Mindstorms EV3) with the RT-Preempt patch (https://rt.wiki.kernel.org).

Has anyone experimented with this already? If so, which kernel did you use? Unfortunately there does not seem to exist a RT-Preempt patch for the current stable EV3dev kernel (4.4.24). I'm going to try it anyway using the RT-Preempt patch for 4.4.23 or .25. But if anyone has been down this road, I'd be interested in what you experienced.

Thanks in advance,
Frank

dlech, 2016-11-15:
The BeagleBone version of ev3dev does use an RT kernel (because that is what the upstream BeagleBone kernel uses). I haven't heard of anyone taking advantage of any of the RT features on it though. And we do have one issue (#710) that I have a hunch may be due to the fact that we are using an RT kernel on BeagleBone.

fgr, 2016-11-16:
I've managed to patch the 4.4.24 kernel (with the RT-Preempt patch 4.4.23-rt33) and successfully build it (with CONFIG_PREEMPT_RT_FULL=y). However, when trying to install the respective deb file, the flash-kernel trigger fails:

> $ sudo dpkg --install linux-image-4.4.24-16-ev3dev-ev3-rt33+_1_armel.deb
> [...]
> Processing triggers for flash-kernel (3.35ev3dev13) ...
> Kernel /boot/vmlinuz-4.4.24-16-ev3dev-ev3-rt33+ does not match any of the expected flavors (ev3), therefore not writing it to flash.

Could that failure result from kernel-flash not recognising the image as an EV3 one because of the extra "-rt33+" at the end of the image's filename (vmlinuz-4.4.24-16-ev3dev-ev3-rt33+)?

dlech, 2016-11-16:
Yes, `flash-kernel` requires `-ev3` at the end of the kernel version. I think it ignores the `+`. So, you need to add another `-ev3` to the end (`4.4.24-16-ev3dev-ev3-rt33-ev3`) or, even better, move the `rt` part before the ev3dev version (`4.4.24-rt33-16-ev3dev-ev3`)

fgr, 2016-11-17:
Thanks, for the hint. I managed to install the 4.4.24 kernel patched with the RT-Preempt patch 4.4.23-rt33. Unfortunately, Bluetooth and Wifi do not work properly. Bluetooth can be turned on via Brickman, but does not work. The Wifi dongle is not recognised at all (but worked fine with ev3dev's original 4.4.24 kernel).

On the bright side: The Lego (EV3) color and IR  sensors as well as the motors seem to work fine.

Next thing I'm going to try is USB tethering to get remote access to the EV3...

dwalton76, 2016-11-17:
@fgr any chance you could see if #710 is resolved in your setup?

fgr, 2016-11-17:
@dwalton76 Sure, will try it as soon as I have remote access.

dlech, 2016-11-18:
If the WiFi dongle uses the rtl8188eu driver, then it doesn't work because it is compiled as [a separate module](https://github.com/ev3dev/rtl8188eu).

If USB doesn't work out, there is also the possibility of connecting with a USB/serial adapter on input port 1.

fgr, 2016-11-18:
I managed to connect the EV3 via USB tethering and log into it via SSH. Unfortunately, EV3 freezes after an undefined amount of time -- seconds (SSH did not even mange to proceed to the CLI prompt) or a few minutes.

@dwalton76  I was therefore not yet able to test #710.

@dlech Is it possible to get shell access using the serial interface?

dlech, 2016-11-18:
Yes. Boot from a known good kernel and run `sudo ev3dev-config`. This is a menu driven program that has an option for enabling console on the serial port. Once enabled, reboot to make sure it is working, then try your RT kernel again. Hopefully it will show you some kernel messages on the serial console when it crashes. For more info on how to build your own adapter cable, see [this blog post](http://botbench.com/blog/2013/08/15/ev3-creating-console-cable/).

dlech, 2017-01-25:
> @fgr any chance you could see if #710 is resolved in your setup?

It turns out that with regards to #710/#795 having PREEMT_RT enabled actually makes things significantly worse. I don't think many people realize this, but the realtime kernel patch makes *everything* slower. The only timing that is really improved is the worst case senario.

closed 2019-03-25 by dlech