ev3dev/ev3dev #780: BeagleBone will not boot ev3dev-jessie-bone-generic-2016-10-17.zip

DanAragona, 2016-11-15:
### System information:

* I am using: *BeagleBone [Black, White, Green]*

* I installed ev3dev using this image file: *ev3dev-jessie-bone-generic-2016-10-17.zip*

* My kernel version is: *KERNEL VERSION GOES HERE*

* My host computer is running: *Linux*

* I am connecting using: *Ethernet*

### About my issue:

I got a new 16 gb sd card:
I used etcher - flashed ev3dev-jessie-bone-generic-2016-10-17.zip (downloaded from web) 
- etcher reported complete
- inserted in Bone - powered up - wait 5 - 10 minutes - did not boot. 
- Powered off and powered back on waited and still no boot.
- retried entire process 3 times - just to make sure!

I placed the SD card into my ubuntu 16.04 box , it mounted successfully boot and rootfs partitions and I could see the listing of all files with no problem so the flashing works

- I went back to my 2016-08-18.img which works fine. (I have this image on a separate SD CARD)

.

dlech, 2016-11-15:
You need to edit `uEnv.txt`on the SD card. Sorry, we haven't updated the website to say that yet.

dlech, 2016-11-15:
You need to uncomment the line `cape=evb` (delete the `#`)

dlech, 2016-11-20:
Website has been updated.

closed 2016-11-20 by dlech