ev3dev/ev3dev #781: Connecting ev3 to raspberry pi 3 jessie

HockeyKS, 2016-11-17:
Hi,

I am pretty new to raspberry pi and to ev3 so I dont have that much knowledge about Linux but I am getting used to it.
We bought an ev3 brick and a Raspberry Pi 3 to start programming with Python. I have set up a connection between the two devices via usb and it works perfectly.
After a few tests I tried to connect the two devices via Bluetooth and it is just frustrating. I followed these instructions http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/ and installed Bluetooth manager etc.
I made it to the point where I set up an NAP and try to connect the two devices via bluetooth. After confirming the pairing request on both devices I get the following error message:
"GDBus.Error:org.bluez.Error.NotAvailable:Operation currently not available. Try to connect manually"
Trying to connect manually it´s a similar problem:
"Failed to pair: org.bluez.Error.AlreadyExists"
Even when I try to connect my smartphone to the raspberry pi it fails.

- I have a raspberry pi 3 running jessie and connected to the wifi
- The ev3 runs on brickman v0.8.0 and Kernel is 4.4.19-15.ev3dev-ev3

Beside the problem I habe another question: Do I have to connect the Raspberry Pi 3 via Ethernet to the internet before setting up an NAP? That doesn´t make any sense to me as long as I just want to use the Bluetooth connection.

I would really appreciate your help.
Thanks a lot,
Konstantin

dlech, 2016-11-17:
Are you running ev3dev-jessie on the RPi or are you using the official RPi Foundation Raspbian?

HockeyKS, 2016-11-21:
@dlech I am running an official RPi Foundation Raspbian. I tried a few different ways to connect and now it is running pretty well. Just ignored the error message on my ev3 and then scrolled down to devices, clicked on raspberry and set the network connection!
But since then my usb connection crashes every time I try to connect.

`pi@raspberrypi:~ $ sudo ip ad add 10.42.0.1/24 dev usb0
pi@raspberrypi:~ $ ssh robot@ev3dev.local
ssh: connect to host ev3dev.local port 22: No route to host
` 

Do you know if blueman made some kind of changes in the system? 
Another question: Is it possible to write a program with python3 on Raspberry and then execute the program on the ev3 from my computer?

Thank you very much for your help. I really appreciate it!

dlech, 2016-11-21:
I have only used Blueman with NetworkManager. If the RPi is not running NetworkManger, maybe you could try installing it instead of manually configuring the network on the RPi. Another option would be to install ConnMan on the RPi instead of NetworkManager and use ConnMan on the RPi to enable tethering on the Bluetooth instead of using Blueman.

I don't think Blueman makes any changes to your system that persist after Blueman is closed. While it is running, it does affect how Bluetooth works.

Generally it is best to ask one question per GitHub issue, but since the python question has been asked many, many times, I don't see a need to open yet another issue about it. Yes, you can write a python program on the RPi and run it on the EV3. You just have to copy the file from one to the other. You can do this from the command line using `scp` or if you are using a desktop environment on the RPi, you can use Filezilla copy the file.

closed 2019-03-25 by dlech