ev3dev/ev3dev #783: Look into options for promoting ev3dev as an educational tool

WasabiFan, 2016-11-20:
At this point, I think we have a reasonably stable platform with good support and plans for a stable release. Along with that milestone, I think we should start considering what we can do to make ev3dev known more widely and promote it for use by users that aren't the "core" developers we have so far. One of the big user groups that I imagine this applying to is educators: people who are teaching textual programming and need an engaging way to do it. This effort would require some preparation and planning, so I wanted to get some discussion started now to see what comes out of it.

The most obvious part of this push would be to update the website with language explaining the potential here. With some tweaks to the jumbotron and three-column group at the top of the homepage, we could spin the platform as the "versatile software environment" that we hope it can be. Additionally, it would make sense to add a featurette section lower down specifically targeted at this "education" demographic; such a section could pitch ev3dev with a link to a more fully-developed page.

Past that, it would be nice to look into something more direct. I know that we have community members who are already working on using ev3dev like this (@ndward I believe is one great example), so we have a good start on this. Getting in contact with early users and hearing from them on how they've leveraged ev3dev (or, more likely, how they plan to) would be a great way to improve the platform and gain credibility quickly in site content.

Finally, we should look into ways of gaining broader user feedback. Engagement is something that we currently don't promote explicitly, yet getting information on what we are doing well and what we aren't is an easy and self-maintaining way of improving.

My hope is that I personally can devote some of my time directly to this; I think that I am most valuable in planning and producing user-focused experiences, and a bit of work in this regard could be extremely valuable.

What do others think? What can we do to further these goals, and is there something else that needs to be considered? Tangibly, when and how should we begin implementing these ideas?

JorgePe, 2016-11-22:
I would love to see ev3dev being adopted as an educational tool but honestly I think it is still difficult to anyone in the "educational" field to adopt it if it has no previous experience with linux and also a geeky background.
Perhaps having a few "education killer apps" to showcase would help. I've never used Open Roberta but I think it can be a good choice, as perhaps something like Scratch/Snap!/App Inventor. I believe the Raspberry Pi Foundation decision to include Scratch and Minecraft in Raspbian helped a lot on bringing young kids and teachers to the Pi world.
These "killer apps" need to be solid stable but they need also to be easy enough to use without having to dig to deeply in the shell environment.
And having many good examples available also helps... I remember first time I heard about ev3dev was because of a video of (i think) Ralph Hempel using a Logitech webcam with his EV3 when I was looking for alternatives to LEGO official environment.

rhempel, 2016-11-22:
No it was not me with the webcam video. I agree that it's not that easy to get into ev3dev without some help, but that's not the target audience. LEGO has a good, easy to use programming environment for education - and if you want to program using a text based language then you will need to have some hacking background. Programming robots is hard - easy to use GUI tools will still let you write programs that don't work as expected and are hard to debug 😀

JorgePe, 2016-11-22:
Yes, i know, i still remember college and mathlab + Turbo C times without documentation  (and also forgot almost everything I learn).
I don't want ev3dev to be a dumb easy thing... it's just that some classrooms could benefit from some kind of tools like a "companion app" or a "browser environment" , something that allows quicker demos.

WasabiFan, 2016-11-22:
> Perhaps having a few "education killer apps" to showcase would help. I've never used Open Roberta but I think it can be a good choice, as perhaps something like Scratch/Snap!/App Inventor. I believe [...]

The way I see it, ev3dev's goal is to support _textual_ programming with real tools. Although I think that there are certain cases where a hybrid drag-and-drop platform could be useful in furthering that goal, I don't think that it is a primary mission for this platform.

> These "killer apps" need to be solid stable but they need also to be easy enough to use without having to dig to deeply in the shell environment.

Agreed. I think that we should have at least one or two language frameworks which are well-supported. The Python lib fits that standard in my mind, so it's a great start. We do need to do some extra work on the overall platform to make it more usable, but overall I think it is in a good spot.

> I don't want ev3dev to be a dumb easy thing... it's just that some classrooms could benefit from some kind of tools like a "companion app" or a "browser environment" , something that allows quicker demos.

I think it would be nice to have some variety of integration with external tools to make it easy to get started. I could imagine this being a set of tutorials for common IDEs or a well-supported browser-based environment. This, however, seems like a farther-out endeavor; I think that some basic tutorials would be fine for now to test the waters.

> it is still difficult to anyone in the "educational" field to adopt it if he/she has no previous experience with linux and also a geeky background.

How about this: what if we created a documentation page on "ev3dev fundamentals"? Such a page would explain the relationship between images, kernels and "ev3dev versions"; essentials of packages; basic concepts of using sensors/motors; and the capabilities that the EV3 provides, among other things. I think that would be useful as a step in the "getting started" guide, and would likely help any ev3dev user, beginner or advanced.

ddemidov, 2016-11-22:
> The way I see it, ev3dev's goal is to support textual programming with real tools.

I'd like to second that. The stock Lego software is already very good at visual programming, so I don't think we want to compete with them here.

JorgePe, 2016-11-23:
>> The way I see it, ev3dev's goal is to support textual programming with real tools.
>I'd like to second that. The stock Lego software is already very good at visual programming, so I don't think we want to compete with them here.

I agree. I used Scratch and other visual tools just as an example but have to admit that when I read the first post I paid more attention to the "educational" than to the "educators: people who are teaching textual programming", so sorry, my fault.

A "problem" with ev3dev is that it's really a full operating system so it's much more than a simple tool, it's an huge toolbox or at least a swiss knife so it's diifcult to explain how to make full use of it.
Back to the Pi example, you can promote it as a generalist education tool but sooner or later you have to give real life examples, so the Raspberry Pi Foundation commitment to python was (I think) a good choice, not because python is better than C++ or Java but because adopting one particular language gives a common ground to everybody (like adoption of English in the net)  . So I agree with @WasabiFan:

>I think that we should have at least one or two language frameworks which are well-supported. The Python lib fits that standard

WasabiFan, 2016-11-23:
I like that thinking. What do you think about some sort of an "ev3dev fundamentals" page (and maybe a more introductory "what is ev3dev?" as well) like I was contemplating? It seems to me that such an addition could alleviate some of these issues. I wouldn't be surprised if part of our problems stem from the fact that people aren't really sure how the system is _supposed_ to work, let alone how it actually does. It might even make sense to create an introductory video on this stuff (to make it more personal), but that would need to come later, after we got other things up-and-running.

JorgePe, 2016-11-23:
I like the idea of an "ev3dev fundamentals" page as also a "what is ev3dev". I know that there are already many pages trying to achieve that same purpose but there isn't a sort of "ev3dev for beginners" guide offering a good, complete, from-start-to-the-end tutorial.

I can give a personal example:
These last 2 years I've been posting most of the small things I've been doing with ev3dev on my LUG forum. We (PLUG, Portuguese LEGO User Group) are a small LUG in a small country, ~150 members nation wide,  we meet just 2 or 3 times each year and just a few like to use Technic and even few make use of Mindstorms product.
From time to time someone asks me about ev3dev or linux but only once someone really tried to use it - someone with a good background in Engineering, Electronics, Mindstorms family and even RobotC. He was trying to promote some kind of team-building contest at his job with other engineers and felt that EV3-G was not appropriate for the task so decided to give ev3dev a try. He got lost in the site python documentation/examples and kept asking me for some examples on how to use python for some basic tasks with motors and sensors. I could not teach him because I have no method, I just have enough time to "learn" by pure curiosity, trial-and-error and googling around and he had not the time (and the patience) for that so he gave up and went to RobotC.

dwalton76, 2016-11-24:
One thing that I found very confusing when I started researching Python on ev3 was figuring out which ev3 Python library was the correct one to use. At the time there were at least three or four to choose from and it took me a while to figure out that this one was the official one. So maybe a page on what all of choices are for Python on ev3 and why one should use this one, would be handy.

dwalton76, 2016-11-24:
Do FLL teams tend to use ev3dev or do they use the Lego graphical language? Maybe we could do some docs and code to show how to do last years FLL robot in Python?

dwalton76, 2016-11-24:
+1 for "fundamentals" and "what is ev3dev" pages

dwalton76, 2016-11-24:
Just throwing this out there...what if we had a series of lessons that educators could use in there classroom? Hmmm as soon as I typed that it hit me how much this would overlap with ev3lessons.com so maybe we find some way to leverage the great work there.

JorgePe, 2016-11-24:
Yes, ev3lessons.com also came to my mind.
Perhaps organizing information in lessons, like a course, could work.
Regarding python, we already have some good information in the '[demo](https://github.com/rhempel/ev3dev-lang-python/tree/develop/demo)' folder of github. We could design some lessons or guides or "instruction cards" based on this demos, using a common physical framework  i.e. a "starter robot" easy enough to be assembled by most educators but powerful/modular enough to cover most of the examples.

massimiliano-mantione, 2016-11-24:
Hi, 1st of all some background so you see where this is coming from.

I am using ev3dev to program a sumo robot with my son (13 years old), so he can learn something about programming while also having some fun participating to sumo robot competitions.
We picked ev3dev because programming complex logic with the Lego visual tools can get really hard, having a textual language is just more practical at that point.
I am a skilled software engineer using Linux every day so this has been easy.

That said, what I think would benefit ev3dev is a "zero config" setup to write and run programs, without even installing anything on a PC.

With a bit of work it would be possible to install on the standard image a web server serving an IDE, with simple "save and run" buttons.
At the high extreme of the "complexity spectrum" something like Cloud 9 (https://c9.io/), which I would consider overkill (and too complex).
I remember that something simpler, just to allow remote editing from the browser, already exists.

The idea is that the ev3 (or pi) would only need a network setup, and then the kid would just need to point a browser to it and everything would just work.
No need for separated services on the cloud or anything.

From my experience such a web server would not be too heavy, and the IDE would actually run in the browser so I guess even an EV3 could support this kind of setup.

I don't think I can commit time to help on this :-(
But if you like the idea I could define it a bit more, and *maybe* even find some time to set it up.
Actually it is possible that there are already remote browser based editors that would do most of what would be needed.
There is even a terminal emulator that runs on the browser that could be used as an interface for the robot process running on the ev3!

Anyway, what do you think of the idea?

dlech, 2016-11-24:
> So maybe a page on what all of choices are for Python on ev3 and why one should use this one, would be handy.

We kind of have that already at http://www.ev3dev.org/docs/programming-languages/

> Do FLL teams tend to use ev3dev or do they use the Lego graphical language? Maybe we could do some docs and code to show how to do last years FLL robot in Python?

Most competitions I have seen, including FLL, restrict the use of programming languages to EV3-G only. The code itself is judged part of these competitions, and they can't expect judges to be familiar with every language out there.

> From my experience such a web server would not be too heavy, and the IDE would actually run in the browser so I guess even an EV3 could support this kind of setup.

We have tried something like this with iPython/Jupyter on the EV3, but the general consensus was that it is too slow. So, any web based IDE would need to be lighter weight than that.

massimiliano-mantione, 2016-11-24:
Small followup: the project I was thinking about is nodemirror: "https://github.com/soliton4/nodeMirror".
It is 3 years old and apparently unmaintained, which *might* be ok because this means it should run on the ancient 0.10 nodejs we have on the ev3 :-D

In the next days I'll try to make it run, likely eliminating all the unneeded bells and whistles (mostly X11 and media files forwarding).

If it works it could be exactly the kind of development environment I was describing in the previous comment.
It would not *shield* users from the unix terminal experience, but it would provide them a zero configuration solution to write code and run it.

WasabiFan, 2016-11-24:
> At the time there were at least three or four to choose from and it took me a while to figure out that this one was the official one. So maybe a page on what all of choices are for Python on ev3 and why one should use this one, would be handy.

We now have some explanation on that issue on our "programming languages" page at ev3dev.org. Does that seem noticeable and understandable enough?

> what if we had a series of lessons that educators could use in there classroom?

Although I doubt we could handle a full course worth of lessons, supplying some basic pre-made material would also be nice. There's also @ndward's ev3python.com, which has some of this stuff for Python.

> That said, what I think would benefit ev3dev is a "zero config" setup to write and run programs, without even installing anything on a PC.

Agreed. We've had a fair amount of discussion on this in #477; I'd highly recommend that you skim through some of that and see if there's anything interesting there. If we can't find an in-browser IDE that works well, providing support for sync-on-save plugins in existing cross-platform editors might also get us what we want.

> We have tried something like this with iPython/Jupyter on the EV3, but the general consensus was that it is too slow. So, any web based IDE would need to be lighter weight than that.

Jupyter does some fancy stuff involving websovkets and live updates, which I think was the main problem. I haven't tried it personally though so I wouldn't know.

ndward, 2016-11-24:
I'm excited to see this discussion since promoting ev3dev Python as an educational tool is exactly what I have been working on for the last couple of months. Yes, I did say Python, so you may feel this message belongs in the Python issues section, but I'll continue anyway. I feel that the special status of Python as the EV3Dev-compatible language most likely to be of interest in an educational context should be recognised and that all the effort to make EV3Dev education-friendly should be focused on EV3Dev Python. Python programs are relatively concise and easy to read compared to programs written in other textual programming languages, making Python perfect for education. I'm guessing that the EV3 is the most popular robot on the planet and that Python is the most-taught textual programming language, therefore using Python to program the EV3 in an educational context should be a no-brainer.  I've done what I can to help beginners get started with EV3 Python by making the site [ev3python.com](url) which now has about 50 pages and which is very much oriented towards education (I'm an ICT teacher myself). What is missing from the site at the moment is a teaching curriculum for EV3 Python – I'm starting to work on that… 

In the sphere of university education, there may be no better university than Harvard. I note that the University of Harvard can now be added to the long list of educational institutions to recognise the pedagogical value of Python for they have designed their new 'Root' robot to be compatible with Python – see [www.codewithroot.com](url). 

@JorgePe links to github.com/rhempel/ev3dev-lang-python/tree/develop/demo where there is a section 'ev3dev demo programs'. Am I right in thinking this section currently contains no ev3dev programs at all??

@JorgePe suggests that we could use

> a common physical framework i.e. a "starter robot" easy enough to be assembled by most educators but powerful/modular enough to cover most of the examples.

I strongly agree – the emphasis should be on coding and not on building and then disassembling multiple models. I suggest that the best choice of standard robot model would be the official Lego EV3 Driving Base model (aka 'Educator Vehicle'). For the time being EV3dev.org is promoting Laurens Valk's Explor3r but I think that's slightly less good for it's not designed to simultaneously support  the touch senor, IR/US sensor, color sensor and gripper attachment as the Driving Base is, and because there are many official Lego tutorials based on the Driving Base model, so that model is probably already popular in many schools. Build instructions: [robotsquare.com/wp-content/uploads/2013/10/45544_educator.pdf](url) 

@massimiliano-mantione  suggests that

> what would benefit ev3dev is a "zero config" setup to write and run programs, without even installing anything on a PC. With a bit of work it would be possible to install on the standard image a web server serving an IDE, with simple "save and run" buttons.

I-Python/Jupyter is too resource-hungry and thus too slow, but I understand that RPYC could fit the bill (though I've never tried it). If so, then a push to make EV3dev (Python!) more education-friendly and beginner-friendly could include an effort to document the RPYC option.

@rhempel says 

> if you want to program using a text based language then you will need to have some hacking background

I like to think this is not so and that stepping up from EV3-G to EV3 Python is in fact not so hard. 

@JorgePe says

>  I would love to see ev3dev being adopted as an educational tool but honestly I think it is still difficult to anyone in the "educational" field to adopt it if he/she has no previous experience with linux and also a geeky background.

but I feel that by using an SSH client like MobaXTerm Free Edition very little knowledge of Linux is necessary. Using MobaXTerm one can (once the SSH connection has been established) create a Python file, edit it and save it to the brick without typing a single Linux command. Then you can either run it from Brickman, still without typing any Linux commands, or type a single Linux command to run the program via SSH.

dwalton76, 2016-11-24:
"We now have some explanation on that issue on our "programming languages" page at ev3dev.org. Does that seem noticeable and understandable enough?"

yep that looks great

WasabiFan, 2016-11-24:
> Am I right in thinking this section currently contains no ev3dev programs at all??

I'm not sure what you mean. Looking at that page, there is a list of demo programs and folders containing their code.

> but I feel that by using an SSH client like MobaXTerm Free Edition very little knowledge of Linux is necessary.

I agree, however I think it can get easier than that. MobaXTerm has a text editor, but it _isn't_ an IDE -- i.e. you can edit your code but you aren't given autocomplete suggestions, intelligent refactoring help or live error checking. The editor that I use, for example, does this when you start attempting to call a function or instantiate a class:

[image]

That means that you rarely have to open documentation, which is extremely nice. When I'm working on code to run on the EV3, I enable the option in my IDE to automatically copy my code to my EV3 every time I save it. That way I just hit <kbd>Ctrl</kbd>+<kbd>S</kbd> and then press the button on the EV3 to run it. Supporting an option similar to this -- potentially completely in-browser so you don't even have to install software on your PC -- would make this workflow easily accessible to anyone.

dwalton76, 2016-11-24:
"@JorgePe links to github.com/rhempel/ev3dev-lang-python/tree/develop/demo where there is a section 'ev3dev demo programs'. Am I right in thinking this section currently contains no ev3dev programs at all??"

I'm not following you on this one, the demo directory has several ev3dev programs in it (in sub directories by robot)

JorgePe, 2016-11-24:
@ndward suggests
> that the best choice of standard robot model would be the official Lego EV3 Driving Base model (aka 'Educator Vehicle'). 

It makes sense, it would be easier for newcomers to switch back and forth EV3-G and ev3dev.
I once tried Microsoft Robotics Developer Studio with the EV3, I'm not sure if Microsoft still supports it but at that time they have something similar, a Reference Design, but I can not find it at this time.

@ndward said
> but I feel that by using an SSH client like MobaXTerm Free Edition very little knowledge of Linux is necessary. Using MobaXTerm one can (once the SSH connection has been established) create a Python [...]

True. Even plain ssh through putty isn't much difficult. I was thinking on the deep relation between ev3dev tools and linux environment, most noticeable when something doesn't work exactly how it's supposed to.
Don't get me wrong - for me it's a wonderful thing to pick up almost any device and several software tools and see that they work almost out of the box but most people just want something that "just works", like the "default" EV3 - yes, it's linux, but no one know and no one really care.
That's why I said "solid stable": a "killer app" should be resilient even when network breaks or a device misbehaves.

ndward, 2016-11-27:
My comments about the page [](https://github.com/rhempel/ev3dev-lang-python/tree/develop/demo) were not clear. What I am saying is that the bottom (README.md) section is entitled 'ev3dev demo programs' but contains no ev3dev demo programs, only links to the models themselves and EV3-G solutions. The EV3 Python solutions are in the folders at the top of the page. Thus I still think the title of the README section is misleading and it could perhaps be made clearer that that section only introduces the models and that the actual Python solutions are to be found in the folders above.

To make EV3dev / EV3 Python as easy-to-use as possible for education purposes perhaps students should be using **tethering** because that must be easier and because students would not need internet access from the brick. But there is still no documentation for USB tethering on [](http://www.ev3dev.org/docs/tutorials/using-usb-tethering/). There _is_ documentation on Bluetooth tethering - I must give that a try.

And I'm still hoping that someone will do an **RPYC** tutorial, especially if that means that programming could be done on a browser rather than through SSH.

In an earlier post I mentioned **Harvard's Root robot** ([](http://www.codewithroot.com/))which can be programmed in Python, has more than 50 sensors and actuators, can be programmed from a tablet or phone and which is much cheaper than EV3. Is it likely that schools wanting to do Python robotics would massively adopt that platform rather than EV3?

WasabiFan, 2016-11-27:
> the bottom (README.md) section is entitled 'ev3dev demo programs' but contains no ev3dev demo programs, only links to the models themselves and EV3-G solutions. The EV3 Python solutions are in the folders at the top of the page.

A README file's purpose is to describe the files in a given directory, so generally when you see a "README" file of any variety you are expected to look at the other files around it if you want to find the code. To make it extra clear, I added a note at the top of that README file. Either way, I'd like to keep this thread on-topic with more general discussion.

> To make EV3dev / EV3 Python as easy-to-use as possible for education purposes perhaps students should be using tethering because that must be easier and because students would not need internet access from the brick.

Keep in mind that less functionality doesn't necessarily mean "easier." There are no USB tethering instructions because there isn't any real benefit to doing it instead of the Ethernet-over-USB option. I'm not even sure that Windows has any tethering mechanism built-in, so it would almost definitely be more work than using the existing USB options. Keep in mind that Windows' USB internet stack is a bit buggy, so it can drop the connection without notice (as explained on the tutorial page for that method). That's why I prefer using a Wi-Fi dongle (there isn't a middle-man that needs to be configured).

> Is it likely that schools wanting to do Python robotics would massively adopt that platform rather than EV3?

Would this product be adopted at any significant rate? I highly doubt it in the short-term, as with any new device. However, the EV3 is very different from the device you liked to in that it is a platform designed to be built around with the LEGO Technic pieces and your own motors while this other robot is just a sensor/motor platform that can run code.

> I had problems with the 'add a link' button while making this post. Just me?

I don't use the  formatting buttons, so I'm not sure how they work or if they function properly.

ddemidov, 2016-11-28:
> And I'm still hoping that someone will do an RPYC tutorial, especially if that means that programming could be done on a browser rather than through SSH.

@ndward, have you seen this page: http://python-ev3dev.readthedocs.io/en/latest/rpyc.html? Does it qualify as RPYC tutorial?

massimiliano-mantione, 2016-11-29:
A few more thoughts on the issue.

I totally agree that Python is a good choice as a language for an "introduction to programming through robotics" curriculum.
And ev3dev is a *very* good platform for this curriculum because it provides a full Linux system, so that the pupils will eventually have plenty of options to explore and learn new things about programming.

Using Lego pieces to build the actual robot is also an excellent choice.
It solves a lot of silly mechanical challenges that one would face when building a robot from plastic, wood or metal pieces, it does not require a 3d printer, and it also gives lots of options for creativity about the design of new robots.
And it is infinitely more flexible than a pre-built, unmodifiable robot (like the one from Harward, which *is* indeed a good robot but cannot be modified).
This "robot design flexibility" is probably the biggest advantage of a Lego solution.

At this point I am not so sure that the EV3 brick itself is the best possible hardware platform for classrooms, but this is a different issue.

The EV3 brick is underpowered and overpriced by today's standards.
It costs about 200 Euros, plus almost 100 more for its lithium battery, and its CPU is really slow.
A PI-based solution (with a Lego-compatible case like the ones sold by Mindsensors or the BrickPi) is way cheaper and has so much more computing power that the comparison is embarrassing.

This could be important because with a more advanced CPU the "web-based IDE" option would be easier to implement (the PiStorms environment from Mindsensors looks amazing).
If I had the budget to set up a classroom I would likely buy PIs instead of EV3s.

On the other hand, on an EV3 the easiest thing to do would be to forget about web-based IDEs and just pick sensible choices for tools to install on a PC (Linux-Windows-Mac).

IMHO the setup that @WasabiFan was describing is the most promising.
VS Code has very good autocompletion (intellisense) support, and starts up faster than Atom on my machine (I regularly use both).
A "foolproof" tutorial page on how to set up:

* VS code
* its Python extensions
* its remote file sync extension connected to the EV3
* how to start programs using BrickMan
* filezilla, connected to the EV3
* a terminal to ssh into the EV3

IMHO would provide a usable environment for a classroom.
Some of the "tool choices" above are a bit arbitrary, but in this case IMHO it's important to pick *one* tool  for each function and show how to use it, so that the pupils don't have to do the choice themselves (choosing these tools requires knowledge that they don't have).

The nice thing about a Linux system is that you are always free to pick different tools later, when you have learned more.

Now, again about web-based IDEs.

The Adafruit one is overkill (too complex to install on an EV3).
It solves a *lot* of issues (like linking the development environment to a remote, private, free git repository), but it is a bit too much.

About nodemirror, I tried it on my notebook and it works just fine.

I then tried to install it on the EV3 and failed in several ways.

`npm` is really slow and needs too much memory if I attempt a full install of the module.

I then tried installing the needed modules one by one, starting from the "problematic" ones (the ones with C bindings that needed to be built, like `pty.js` and `ws`).
I could not get them installed because of API issues with the nodejs bindings (node-gyp builds had errors).
Having a newer npm and-or nodejs is also nontrivial on the EV3 (this should go to a different issue).

However, I am confident that with a proper nodejs installation a "simple" web-based IDE could work on the EV3.
The trouble is just the installation.

Are there easy instructions to set up a virtual environment (preferably docker-based) where I could cross compile software for the EV3?
Just for fun I tried to start a nodejs build on the EV3 but, obviously, it will take hours to complete (my estimate is actually days).

ddemidov, 2016-11-29:
> Are there easy instructions to set up a virtual environment (preferably docker-based) where I could cross compile software for the EV3?

http://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/

ndward, 2016-11-29:
I agree with @massimiliano-mantione that the EV3 now seems overpriced and underpowered, though its modular nature, its wide adoption, its Linux brain and its rechargeable battery are all very valuable.  Almost all other electronic devices rapidly drop in price but not the EV3. I'm sure the profit margin on the EV3 is very high – nearly all the parts are just cheap-to-produce plastic, after all, and I think Lego is being greedy by not dropping the price in the face of increasing competition and in order to keep market share. Alternatively, as an intermediate solution prior to the introduction of a completely new robot, they could bring out an EV3+ model with similar hardware except that the brick would have a faster processor, more RAM, built-in reliable wifi, backlit screen, a couple of extra sensor ports, an audio out socket, support for programming from a browser on any PC, tablet or smartphone, and of course official support for programming in Python, like the Harvard Root has! Maybe the EV3+ model could combine the home and education sets into a single set including  the US sensor, the castor wheel and the IR beacon so that people would no longer have to buy additional parts just to be able to make all the most standard models (I had to buy the IR beacon, for example). I didn't mention the rechargeable battery because that's a more expensive item. Perhaps the two kinds of EV3 software could be combined into one also, and offered free like the home software already is.

dlech, 2016-11-29:
> Having a newer npm and-or nodejs is also nontrivial on the EV3 (this should go to a different issue).

https://github.com/ev3dev/ev3dev/issues/448

ndward, 2016-11-29:
One thing that would make EV3dev / EV3 Python much more attractive to educators and everyone else would be to have some introductory information in other key languages such as Spanish, French and Russian. I found people to translate the EV3 Basic manual into all these languages, as well as German, and I can probably use the same people for basic info on EV3dev / EV3 Python if we can agree on one short introductory text. Yes, I know automatic translation exists but it doesn't work very well yet.

dlech, 2016-11-29:
> is there something else that needs to be considered?

Not exactly related to ev3dev, but here is some philosophical reading on consideration of using robotics education in general: https://www.facebook.com/groups/legomindstorms/permalink/741727152642180/

laurensvalk, 2016-11-30:
>  I suggest that the best choice of standard robot model would be the official Lego EV3 Driving Base model (aka 'Educator Vehicle'). For the time being EV3dev.org is promoting Laurens Valk's Explor3r but I think that's slightly less good (...)

Unfortunately, the Educator Vehicle can't be built with the 31313 set. This was the main reason for designing EXPLOR3R. I consider the TRACK3R model to be not that suited for teaching, mainly because it uses tank threads (inaccurate positioning and steering) and because it doesn't fit the rechargeable battery.

I'd say that ev3dev could promote the use of the Educator Vehicle for those who have set 45544, and the EXPLOR3R for those with 31313.

laurensvalk, 2016-11-30:
> At this point, I think we have a reasonably stable platform with good support and plans for a stable release. Along with that milestone, I think we should start considering what we can do to make [...]

> What do others think? What can we do to further these goals, and is there something else that needs to be considered? Tangibly, when and how should we begin implementing these ideas?

There are a lot of good ideas in this discussion, but for most educators (at least up to high school), I think ev3dev is simply a step too far. From what I understand, RobotC is hard enough for most teachers, even though it comes with an all-in-one IDE and tech support.

> One of the big user groups that I imagine this applying to is ....

I think these are home users / hobbyists who want to go beyond what they can do with the EV3 software. These users would typically be new to most or all of the following:
- Flashing an SD Card
- Setting up internet connections
- Linux / Command Line
- Python, or any text-based language
- Debugging/Technical trouble shooting (in the broadest sense)

RobotC automates/avoids most of these, letting the user focus only on learning a new language (C instead of Python above), which can be hard enough.

I'm not saying ev3dev should be doing all of the above (though obviously, such all-in-one IDE would be fantastic), but I think this is almost a necessity if the long term goal is to make ev3dev easily accessible.

Back in the RCX/NXT days, there was RobotC for schools as well, but there was [Brixcc](http://bricxcc.sourceforge.net/) for hobbyists. It had the tools to interact with the brick and run programs with a few clicks.

In the absence of such all-in-one IDE, I think [LMS Compat](https://github.com/ev3dev/lms2012-compat/) could be a good intermediate step between EV3-G and full ev3dev, since the user would need to do only the first two steps on the list above.

JorgePe, 2016-11-30:
@laurensvalk said
>RobotC automates/avoids most of these, letting the user focus only on learning a new language

I remember reading somewhere about an effort to "mix" ev3dev and RobotC. Something like that could be a good solution for many of the points that arose here.

laurensvalk, 2016-11-30:
> I remember reading somewhere about an effort to "mix" ev3dev and RobotC.

I think that was where RobotC developers would use ev3dev. Here it's the other way around, where ev3dev would ideally have an IDE. But RobotC is non-free, so that wouldn't really work.

I meant to say that I think that some form of IDE is necessary to make ev3dev practically usable for educators and many hobbyists.

JorgePe, 2016-11-30:
So we need an "OpenRobotC-like" killer app :)
Just kidding but nevertheless I like the idea.

dwalton76, 2016-11-30:
So from reading this thread it sounds like the #1 thing we could do is have an IDE?  Some of the ones mentioned in this thread:
- jupyter - too heavy in terms of CPU
- Adafruit - overkill in terms of features, not sure about CPU
- nodemirror - install issues so TBD if EV3 has enough CPU for this to be an option

I'm a "ssh to the box and use vim" kinda guy so I don't have any others to add to the list.  What are some others that we should explore?

laurensvalk, 2016-11-30:
An IDE just one solution. There are other good ideas in this thread too.

I think it's more than just selecting an existing IDE. I'll open a separate issue about it.

ndward, 2016-11-30:
@laurensvalk , what is it that stops the educator vehicle being made with the home kit? If it's just the caster wheel then that's a useful accessory that can be easily bought for about $6
http://www.ebay.com/itm/Lego-BALL-CASTER-mindstorm-s-robot-omnidirectional-technic-pivot-ev3-nxt-rcx-/272442632495 or an alternate wheel can be easily improvised.

Otherwise there is Damien Kee's Riley Rover which is a 'light' version of the Educator vehicle. Build instructions here: http://www.damienkee.com/home/2013/8/2/rileyrover-ev3-classroom-robot-design.html  Instructions are also provided for the Retail Rover which is modified version that can be built with the home (retail) kit. I used the Riley Rover with my students in Brussels with great success except that they felt that the brick was the 'wrong way round' relative to their expectations - this was easily fixed. https://sites.google.com/site/gask3t/lego-ev3/ev3eeb3

Flashing an SD card with Etcher should be VERY easy. Yes, setting up the connection from PC to EV3 is awkward, but as I said previously, once the SSH connection is made one can create, edit and run Python programs without using any Linux commands at all. 

I consider the need to use a textual language such as Python as a big PLUS - for me a main reason to use EV3 Python in an educational context would be to let students who already have some knowledge of Python reinforce and extend that knowledge in a highly motivational and novel way.

laurensvalk, 2016-11-30:
> @laurensvalk , what is it that stops the educator vehicle being made with the home kit? If it's just the caster wheel [...] for $6

The regular wheels are missing as well. (31313 wheels are too small). Ordering new elements (even if cheap) is another hurdle as it takes a few days, and modifying the design also takes time. It's simply not necessary.

ndward, 2016-12-01:
With lots of help from @ddemidov and http://python-ev3dev.readthedocs.io/en/latest/rpyc.html I've managed to set up RPyC on my Windows 10 PC and I can now use the PyCharm Edu IDE (which is free) to write Python scripts which I can then run directly from PyCharm without using SSH. Maybe this is in fact the easiest way for students and others to program? There's a bit more to it than that - I'll put a page on ev3python.com, maybe with a video.

ndward, 2016-12-09:
I have made this page https://sites.google.com/site/gask3t/lego-ev3/building-plans/educator-vehicle-retail-kit-version to show how the building instructions for the official Lego Educator Vehicle can be modified to suit the home (retail) version of the EV3 kit without buying additional parts. Of course an educator vehicle based on the home kit will not have a gyro sensor and will have an IR sensor in place of the US sensor. Most of the official exercises will work with this modified version, though some scripts may also need modification - see https://sites.google.com/site/ev3python/learn_ev3_python/basics .

loopspace, 2017-01-04:
I'd like to chip in here.  I'm a UK maths/IT teacher with about 20yrs experience of using Linux.  So in terms of education, I'm probably your "first wave" target in that any resources or whatever wouldn't have to be 100% perfect for me to work with them.  I've many more years of experience with Lego, and have the ev3 home kit, and my school has a couple of the education ev3 kits.  We don't currently do anything particular with them, but are always on the look-out for ways to increase IT stuff.

I came across this project just a couple of days ago and have managed to get a working robot that I'm quite pleased with.  As an experienced Linux user, it was all quite easy.  There were a couple of things that I needed to look up, or configure, which might not have been so easy had I not been used to Linux and programming.  The ev3python site is a fantastic resource as it has lots of examples of how to do stuff (which shed light on the API documentation).

Anyway, here's my thoughts:

1. I saw some discussion in the above about Mindstorms vs other robots.  I took a look at the Harvard root and my reaction was "Meh".  It's not significantly cheaper than the ev3, and not customisable.  Lego is easy to build and so makes it easy to get started with building a robot.  And there may be more powerful things one can do with an RPi and some bits and pieces, but once you've bought the Mindstorms kit then you have it all there.  So my first thought is: stick with Lego as the primary target and focus on being easy to get started with.  Ultimately, there's a lot of room in the market for entry-level kits and so I don't think you should be concerned about "competition" from other systems, just figure out the strengths of what you're offering and play to them.

2. Python appears to be the "in" language in schools in the UK.  So even in for hobbyists, making python easy to use makes it easier to connect to what kids are seeing in school.  I would take a leaf out of the RPi's book here and focus on making it super easy to work with python.  In particular, I had to install pip3 so that I could install the pyserial module.  Both really ought to be there by default (is there room for more python modules in the initial build?).  Can `python` be `python3` or is there system stuff that depends on `python` being `python2`?

3. When making resources, you should think about things from two viewpoints: the teacher's and the student's.  A teacher might well be happy putting in a bit of preparatory work, or might have an IT department that could do some, but will want it to go smoothly in the lesson.  Having explanations and what-to-do-if sections will be useful for them.  Students will more want lots of examples to start with and then modify.

4. It's unlikely that a school would have a Mindstorms kit for every student in a class.  So coding directly on the device, or via a web browser, is not going to be the way that it would work.  A more likely scenario would be to have the students working on a PC and transferring their program once happy with it.  What would help with this would be to have a simulator python module that could be on the PC which could be used for testing.  In this scenario, all that is needed is an easy way to get files onto the device, the fancy IDE stuff is on the PC and assuming that they've already done some python then they'll already have a preferred IDE for that.  So I wouldn't put too much energy into "live programming" systems.

My robot, by the way, is the Grabb3r robot from the home kit.  What I used ev3dev for is that I've connected a micro:bit to the USB (hence the need for pySerial) and I have another micro:bit which is hand-held.  The two micro:bits communicate via radio, and these get passed on to the ev3.  I then have a python script that listens to the USB and converts the information from the micro:bit into commands.  I'm thus able to control the robot by tilting the micro:bit in my hand.  Total amount of code: 120 lines.

In conclusion, I think ev3dev is fantastic, and I think you're right to look at providing education resources for it.  I've just gotten started with it, and I would definitely be interested in doing something with it in my school.  So if there's any way I can help ...

fisherds, 2017-01-05:
Just saw my name mentioned so I thought I'd share a bit @loopspace.  I'm sure everybody does their own thing, but I thought I'd share what I'm doing just for fun.  I'm in the prep stages of using EV3 dev in the classroom for a college intro to programming class that uses Python.  Summary of my plan just as an fyi for what people are doing.
 - 26 students only this quarter. At Rose-Hulman (college in Terre Haute, IN USA btw) we use quarters not semesters.
 - 10 robots built as Snatch3r (https://goo.gl/photos/kT5EaGc4W75HMBSa9)
 - Students work in teams of 3 or 4
 - I'm having everyone install PyCharm Professional ([my instructions to the students](https://docs.google.com/document/d/1MXThtuT79o6_SmovrWtPNulhi423kdGGDxMsu7LZXt4/edit?usp=sharing)) it makes the transfer of files about as slick as I could find.
 - I'm telling them about the robot ([my slides for that day](https://docs.google.com/presentation/d/17xGhRQt3VTX3Ltg_2FE2_aC7nSCNqHZzLzyaS6N8Yb4/edit?usp=sharing))
 - I'm getting them started using PyCharm Professional tools with the robots ([my slides for that day](https://docs.google.com/presentation/d/15cYiil-p30to6tcTKZdxGlgHFTFg8EivHJV6Z-Om64s/edit?usp=sharing))
 - That day they run some demo programs on Snatch3r: 
[ev3-demo-programs.zip](https://github.com/ev3dev/ev3dev/files/686385/ev3-demo-programs.zip)
 - Then my main project now is getting together their learning material.  I like to give them starting code and leave a bunch of TODOs as they learn, then they'll make their own projects from scratch.  I have a bunch of good stuff but it's not ready yet.

Just sharing a bit about my progress.  I like to make everything sharable with others so just let me know what I can do to help get more educators on board.  

Here is [a bit about my setup process](https://docs.google.com/document/d/1jNlT9JKff3p9TyrDQwF3XjakaLEp_ilD1wNaHr7c67o/edit?usp=sharing) for each robot too.

That's the direction I've been going.

loopspace, 2017-01-05:
Lots to respond to ...

> We prefer Debian packages over using PyPi

I did search for a package matching `serial`, but without success which is why I went down the `pip3` route.  I used `apt-cache` to search and initially it turned up nothing (in retrospect, I should have simply installed `aptitude` with which I'm more familiar).  Although it's not really `ev3dev` specific, "installing new modules" would be a good FAQ.

On the other things @dlech mentioned, again with the benefit of hindsight, I think it's more about a clear pathway to add stuff.  Although I've years of Linux experience I only started learning python last summer so don't have the same understanding of all the different ways of installing modules.  I use `pip3` on my Mac so that's what I tried on the ev3.

@WasabiFan The `pip3` and `pyserial` were the main things that I missed.  (Well, I also installed `zsh` but that's just me ...)

> how did you discover ev3dev?

Googling!  The short story is that I wanted to connect a BBC micro:bit to the ev3 and so I googled "microbit ev3".  That led me to a youtube video with someone controlling the leds on an ev3 from a micro:bit over bluetooth.  The blurb said that it used ev3dev so I googled that and found myself on your site (after looking at it, and related links, I realised I'd come across mind-storms.com and ev3python.com before but had filed them in my "to look at some day" bookmarks).

> Was it easy to understand what ev3dev was ...

Yes, but my background probably helped (specifically Linux and having played with nqc in the NXT2 days).

> ... find your way to our related guides?

Yes and no.  Getting it up and running was really easy (incidentally, the "you can revert by ejecting the SD disk" will probably be one of the major selling points!).  Once I had gotten to the point of wanting to program then finding the information I needed to write the program was less easy.  I found the API documentation hard to parse, but once I'd started with the tutorials on ev3python.com then life got much, much easier!  It is linked in the right place, so I'm not sure why it took me time to find it - probably because I started with the github link at the top of the python section (on http://www.ev3dev.org/docs/programming-languages/) and on the github README page then ev3python.com is not clickable so I looked at the other stuff first.

With the rest, I think you're right: probably the most useful thing I can do is to try things out and see where I get stuck (or where I think I would get stuck if I didn't have a bit of experience to fall back on).  I shall definitely have a look at what @fisherds is doing and see what I can steal^H^H^H^H^H be inspired by.

dlech, 2017-01-05:
> I used apt-cache to search and initially it turned up nothing

Did you run `apt-get update` first? The SD card image ships with an empty cache to save on download size.

> he pip3 and pyserial were the main things that I missed. (Well, I also installed zsh but that's just me ...)

Like I said, if you think these are useful enough to include in the default install, please open a new issue. I won't find these comments buried this deep in an unrelated issue when I get around to doing another release. Same goes for any other suggestions really. If a suggestion is not just talk and you would like to see someone actually take action on it, open an issue specifically for that idea. Otherwise, these ideas tend to get lost and forgotten.

JorgePe, 2017-01-05:
> That led me to a youtube video with someone controlling the leds on an ev3 from a micro:bit over bluetooth

@loopspace Great to here that my micro:bit video brought you to the project :)

But I feel I can do better in my videos and my blog. Do ev3dev have something like a "media kit" for supporters and media sites?

loopspace, 2017-01-05:
@dlech I get that.  After this discussion, my feeling is that what is really needed is what you just wrote about the cache should be somewhere prominent.  That's the sort of thing that even someone familiar with Linux might miss.  I can put that as an issue.  But if I'd known that, the fact that `pip3` and `pyserial` aren't there by default wouldn't have been something I'd've made a fuss about.  (I think I ran `apt-cache` *before* thinking to do `apt-get update` and then didn't re-run `apt-cache`).  I'll suggest `pyserial` but I don't feel as strongly as I did when I first wrote!

@JorgePe Ah, that was you!  I didn't find your blog, though.  I just saw the blurb below the video.

rdtheriault, 2017-01-13:
Couple thoughts as I actually implemented this in my classroom this year. Lots of schools have the EV3/NXT systems but their “brains” are going bad (my NXT don’t even work anymore as well as some of my EV3s). Not only is using Python via EV3dev great as Python is a real language (sure I am going to get flak for that, sorry RobotC users) but then as the rest of my EV3’s die off I can replace them with BrickPi’s and continue to use the Lego parts (saving tons of money which schools love). 

Since this is all in the context of a Computer Science course, I teach them a bit of Linux anyway so SSHing works out fine (which is not even needed if you connect a screen to the BrickPi). Putty doesn't need to be installed so I can get around IT needing to install anything.

I think the key for this is a well-developed curriculum that is plug and play. I.E. there is a great setup and basic use section then teachers can just take pieces out of there that fits their need. If all the work is done and teachers can show or refer to videos and easy worksheets, I believe this can a feasible teaching tool for the average teacher. Maybe there be an intro and setup section and then a bunch of projects that can be chosen from. 

I see the biggest benefit (beside teaching Python) is the ability to use what they already have and extend its life span all while using pretty much the same code. I was able to take a relatively small grant and take a classroom full of parts and a hand full of “brains” and turn it into a full classroom set again for a fraction of the cost (I can use EV3 and BrickPi at the same time since the code is the same). 

I have a couple projects up at [https://github.com/rdtheriault/EV3Dev](https://github.com/rdtheriault/EV3Dev). I started with the Codecademy Python Course, moved on to trinket [https://trinket.io/sense-hat](https://trinket.io/sense-hat) with a few projects and then EV3dev robots. I do believe if there was a course of action for teachers they would follow it, especially if it would give their current equipment more life. 

Just my thoughts.

laurensvalk, 2017-01-13:
> Lots of schools have the EV3/NXT systems but their “brains” are going bad (my NXT don’t even work anymore as well as some of my EV3s). 

Did you find out what caused them to go bad? I have several NXT bricks, and some have been used extensively (literally the motors grinding down the plastic on the rotating hubs after running for days at demos), but the NXT bricks ("brains") all still work.

Some NXTs have a known issue where the screen fails due to a badly soldered connection, but continue to operate normally otherwise, which isn't too bad. (And the screen can be fixed if you want to, although I've not tried it.)

And sometimes the firmware disappears, leaving the NXT with the clicking syndrome, or the EV3 in eternal "starting" mode. In both cases, resetting and upgrading the firmware usually restores them to normal operation.

rdtheriault, 2017-01-13:
I inherited the devices, some didn't turn on, for all I know it was the a battery that is dead (some are just missing the battery pack), some are just so slow it is painful, some where already discarded by the previous teachers. I didn't have a bunch of time to try to salvage any at this point (thanks for the potential solutions though!). I took what was working augmented it with BrickPi's and use the EV3dev to make them all act the same. Plus the NXT doesn't work with EV3dev so I don't plan on spending too much time on them. I was going to completely scrap the Lego's until I found EV3dev, the ability to teach/use Python was very important to the class.

jhconning, 2017-01-15:
I'm just a Dad having fun trying to learn about robots and coding with my 9 year old and I haven't played any role in the development of ev3dev, but I found the discussion interesting and would like to see the ev3dev platform improve and prosper.   I have kids aged 5 and 9 and have also played with Dash and GoPiGo and BrickPi robots.  Based on my experiences trying to get things running  on ev3dev (see also rhempel/ev3dev-lang-python/issues/282 on suggestions to improve Bricpi documentation) a few questions/suggestions:

- first of all, this is a great project, the most exciting project out there.  Thank you!  I bought the EV3 retail kit several years ago but hated the bundled Mindstorms/labview software and have tried to steer my own kids toward better/cleaner visual programming options (Blockly, Roberta) and now (and hopefully years to come) python.  

**cloud platform interface?**

 - The [google/coding-with-chrome](https://github.com/google/coding-with-chrome) IDE establishes a bluetooth connection to the EV3 *very* easily with just a few clicks and currently supports blockly and python.  However, the ev3dev-python  library is much richer and object-oriented than the stripped down python they currently offer.  *Would it be possible to make ev3dev-python work on this cloud platform (and pair with ev3dev on the brick)?*  You'd have users up and running in seconds with no need to install software on their local computer (just a chrome extension).

**ev3dev on BrickPi**

- the BrickPI is already popular with educators, if you could make ev3dev just a little bit easier to setup, you'd stimulate this market further. It would seem Dexter Industries ought to welcome this rather see it as competition (since it is likely to only increase BrickPi sales) 
- A key barrier now is that ev3dev BrickPiev3dev users don't have brickman interface which complicates initial configuration of wifi/bluetooth.  Plugging the pi into HDMI with keyboard/mouse does not lead to a login screen (just what looks like an incomplete startup), so users have to SSH in via an ethernet connection but this adds an extra layer of complication which will scare people away.  
- The Dexter industries BrickPi image allows users to boot directly into a linux jessie desktop environment where it is easy to configure wifi and other settings and launch a python IDE. But IMHO their robot control python library is not as nice and object oriented as ev3dev-python.   One can of course also just establish an SSH terminal connection (my preferred setup, using pycharm) , but for new hobbyist/education users, and for easy first-use configuration, it would be very useful.  
- *Would it be possible to modify the ev3dev image for the pi to allow booting into a desktop environment if an HDMI connection is made for easier configurations?*

rdtheriault, 2017-01-15:
I have figured out how to get the BrickPi into terminal, you just hit ALT F2 when it hangs up and it will take you to the log on. I am annotating all of the differences I find here [https://github.com/rdtheriault/EV3Dev/blob/master/BrickPiNotes.md](https://github.com/rdtheriault/EV3Dev/blob/master/BrickPiNotes.md) for the BrickPi VS the EV3. If you have anything else to add please let me know.

dlech, 2017-01-15:
> cloud platform interface?

There is a discussion about IDEs in #788. This would be a good place to bring this up.

> ev3dev users don't have brickman interface which complicates initial configuration of wifi/bluetooth

[It doesn't have to be that way](http://lechnology.com/2016/05/adding-a-display-to-brickpi/) :smiley:

> Plugging the pi into HDMI with keyboard/mouse does not lead to a login screen (just what looks like an incomplete startup), so users have to SSH in via an ethernet connection but this adds an extra layer of complication which will scare people away.

The other side of this is that having to purchase an HDMI screen for your Pi is an extra cost with little use for a robot. In fact, this is why the login screen is currently missing there. I don't have an HDMI display and no one else has volunteered to make it work.

> Would it be possible to modify the ev3dev image for the pi to allow booting into a desktop environment if an HDMI connection is made for easier configurations?

Sure, it is possible. But, ev3dev is primarily about the EV3. To keep things simple, we want the platform to be the same (as much as possible) no matter what device your are running on. The EV3 doesn't have the CPU and RAM to run a desktop environment, so we don't have that on any of the other platforms currently.

The way we have the ev3ev build system set up, it would be quite easy if someone wanted to maintain an "ev3dev + desktop" version.

We also have hopes that [Etcher](https://etcher.io/) will [make configuration possible before you even boot the first time](https://github.com/resin-io/etcher/issues/718) in the near(ish) future.

> I have figured out how to get the BrickPi into terminal, you just hit ALT F2 when it hangs up and it will take you to the log on

It doesn't actually "hang". Brickman tries to run on the first virtual console (`tty1`) instead of a login prompt (`getty`), but Brickman must be failing to start. I suspect that there is not an fbdev device for the HDMI output. In <https://github.com/rhempel/ev3dev-lang-python/issues/279>, we figured out that you can run `sudo systemctl stop brickman.service` and `sudo systemctl disable brickman.service` to get the login back on `tty1`.

> If you have anything else to add please let me know.

Why not put this stuff on the ev3dev website? http://www.ev3dev.org/docs/platform-comparison/

rdtheriault, 2017-01-15:
That is great info, I will try it out with my class next time we meet. This is what I am talking about one quick and easy guide for everything an educator would need to get it running (step by step) not search and get. How would one go about adding to that page?

dlech, 2017-01-15:
There is a button at the top of the page that says "Edit on GitHub". Click it.

jhconning, 2017-01-20:
Thank you for your responses, and @rdtheriault  thank you in particular for the ALT F2 tip.  With that knowledge one can boot directly into terminal mode (on an HDMI connected screen). Setting up wifi from there is then easy using connman (although I had to go searching for that detail too).  

I think many first-time pi/BrickPi users will prefer this route to get networking started.  I found this route easier than the ethernet route (which requires tinkering with settings on the Windows/OSx/Linux side) and sometimes failed.

I've just submitted two pull-requests  (my first ever so I hope I did these right) that I hope make things a bit clearer for first-time users:
-  ev3dev/ev3dev.github.io#332  proposes adding a sentence explaining the ALT-F2 option to the existing 'note' to raspberry pi users at  [Step 4: Boot ev3dev](http://www.ev3dev.org/docs/getting-started/#step-4-boot-ev3dev) of the getting started guide.  
- ev3dev/ev3dev.github.io/#333 modifies the existing note to pi users on the [Networking](http://www.ev3dev.org/docs/networking/) to add an external link to the [connman wiki](https://wiki.archlinux.org/index.php/Connman#Usage) for detail on how to establish a wifi/bluetooth connection.  It also mentions the ALT-F2 tip again for users connecting with HDMI screen.

fisherds, 2017-02-17:
Just sharing an update on my course.  It's a college freshman level Introduction to Python course.  I have 22 students finishing up the ev3dev curriculum that we made.  The curriculum is a series of 24 ev3-python programming challenges that finishes in a course project.  During the 24 programming challenges students learn all about the ev3-python API and make various applications using their robot.  I tried to make everything publicly available if anyone wants to see what we did or leverage it for their own students.  I decided the best way to share it was as a Github repo with links in the README to all the slides and other documents used in the course.  Here is the link to the course material if anyone wants to collaborate on the project:

https://github.com/Rosebotics/ev3dev-curriculum

Fire away if you have any questions.  Took a ton of time to put it all together, but the course was a real success.  I have a bunch of videos that I'll have to post.

WasabiFan, 2017-02-17:
@fisherds That's really great -- I'm glad to hear that it was successful! I browsed through some of your PowerPoints and really like how you walk through things progressively. Just as I would start to think "there's a better way!" the next slide would explain why the original was inefficient and how to make it more elegant. I didn't see much mention of the "hold" stop mode; that isn't really useful for the driving examples, but for attachments that may be more beneficial. If you have any feedback on the platform, please open issues for those items to tell us what we should improve! That goes for the OS, the Python library, the documentation, and anything else you have.

If we ever decide to build official ev3dev curriculum, yours would be an excellent starting point (with your permission, of course): you have developed assignment programs that are interesting yet simple and progressively build on concepts.

For @fisherds and anyone else in a similar situation (although I know that @fisherds has already done this), I recommend that you add yourself to the watchers list of the GitHub repo corresponding to your language's ev3dev library; the Python one at least is being enhanced regularly and many changes can simplify your demos. For example, we recently added constants for the color sensor's integer color values; that type of thing can greatly minimize required explanation. There's also talk of adding better infrastructure for the `IR-SEEK` mode of the IR sensor.

P.S. in rhempel/ev3dev-lang-python#305 we removed the warnings you see on the driving PPT's "Intentionally block your code" slide; next time we do a release those large orange boxes will disappear. I figured I'd mention that given that it makes the page look completely different.

fisherds, 2017-02-17:
Thanks for your feedback @WasabiFan we'll keep making updates as the tools get better and better.  BTW you are totally welcome to use anything I've done if you make an official ev3dev curriculum.  No worries there, the goal is to help.

[8 more replies]