ev3dev/ev3dev #784: Large, Medium tacho motors not working after update to 2016-10-17/2016-11-16 from 2015-12-30

rodrigobaquero, 2016-11-20:
Hi,

### System information:

* I am using: LEGO MINDSTORMS EV3
* I installed ev3dev using this image file: snapshot-ev3dev-jessie-ev3-generic-2016-11-16.img.xz
* I also tried with image: ev3dev-jessie-ev3-generic-2016-10-17.zip but got same issue.
* My kernel version is: 4.4.32-17-ev3dev-ev3 #1 PREEMPT Wed Nov 16 11:12:06 CST 2016
* My host computer is running: OS X
* I am connecting using: Bluetooth

### About my issue:

Large,Medium Tacho motors does not rotate any more after updating ev3dev 2016-10-17/2016-11-16 from 2015-12-30. I'm using python 3 in the new releases as python2 throws an exception when importing ev3dev libraries. The snippet below does not generate any error, but the motors do not react to the command neither.

See below python snippet used:

'''

robot@ev3dev:~$ python3
Python 3.4.2 (default, Oct  8 2014, 14:47:30) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ev3dev.ev3 as ev3
>>> motor = ev3.LargeMotor(ev3.OUTPUT_A)
>>> motor.run_timed(time_sp=3000, duty_cycle_sp=75)

'''

The motor seems to be recognized correctly:

'''

robot@ev3dev:~$ dmesg | tail
[  222.407615] lego-port port4: Added new device 'outA:lego-ev3-l-motor'
[  222.680470] tacho-motor motor0: Registered 'lego-ev3-l-motor' on 'outA'.

'''

The code worked correctly in older ev3 release:

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img.zip
* My kernel version is: 3.16.7-ckt21-9-ev3dev-ev3 #1 PREEMPT Tue Dec 15 15:16:17 CST 2015

'''

robot@ev3dev:~$ python
Python 2.7.9 (default, Mar  1 2015, 13:52:09)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import ev3dev.ev3 as ev3
>>> motor = ev3.LargeMotor(ev3.OUTPUT_A)
>>> motor.run_timed(time_sp=3000, duty_cycle_sp=75)

'''

Any help is greatly appreciated,

ddemidov, 2016-11-20:
You should replace `duty_cycle_sp=75` with something like `speed_sp=600`. See http://www.ev3dev.org/news/2016/04/11/Kernel-Release-Cycle-10/ for more details.

rodrigobaquero, 2016-11-20:
Thank you so much, it is working fine now.

closed 2016-11-20 by rodrigobaquero