ev3dev/ev3dev #787: Proposal for the addition of rpyc in ev3dev

DJuego, 2016-11-28:
Ok. I am definitely a C/C++ boy. I will always be interested in the [C++ binding ](https://github.com/ddemidov/ev3dev-lang-cpp)(!) for multiple reasons.  My field is researching and i am interested in game development. 

However now i am exploring the fantastic world of ev3dev-python. Your strengths are undeniable (flexible, interactive,  fast prototypes,...),  it seems the most active binding and its learning curve is sexy. :-) In fact,  I want to use it for education the next year.

I think than [rpyc](https://rpyc.readthedocs.io/en/latest/) (or a similar alternative) would be a awesome  addition. The reasons is marvelously described [here](http://ev3dev-lang-python.readthedocs.io/en/latest/rpyc.html).

However, i suspect there are a reason why nobody (rhempel, ddemidov, WasabiFan...) have done this request before 🎱 ...

DJuego

WasabiFan, 2016-11-28:
I'm not sure what you're asking for... Can you elaborate a bit? You linked to our existing tutorial on using RPyC, so you know that it exists; what else are you looking for?

DJuego, 2016-11-28:
Sorry. :-( I am requesting that it be considered the addition of RPyC in the ev3dev image.

DJuego

dlech, 2016-11-28:
I don't see a debian package for this (e.g. `sudo apt-get install python3-rpyc` does not work). Am I missing something?

I'm not sure how I feel about installing python packages by default that don't have a debian package.

dlech, 2016-11-28:
Would [python3-protorpc-standalone](https://packages.debian.org/stretch/python3-protorpc-standalone) be a good substitute?

ddemidov, 2016-11-29:
It does not look like it. Google RPC is web oriented protocol: http://www.grpc.io/docs/quickstart/python.html.

It should be easy to create a debian package for RPYC using python-ev3dev as a template. I could do that if we decide it should be done.

ddemidov, 2016-11-29:
I've successfully built the debian package for rpyc-3.3. The configuration is here: https://github.com/ddemidov/rpyc/tree/debian/sid. I have yet to actually test the package on the EV3 though.

dlech, 2016-11-29:
@ddemidov, thanks for doing that. It will be much better to have it as a debian package.

ddemidov, 2016-11-29:
@dlech, I've updated the repo after your comments. I also tested the package on the EV3, it works.
It has python3-plumbum as a dependency which should also be included into the image (around 45 KB).
I can upload the packages when you say its OK.

dlech, 2016-11-29:
The only potential issue I see is that you are using `native` source format. If there is a problem with the packaging and you have to release a new version, you have to increase the version number. But since you don't control the upstream version number, you may find that a new upstream version, when released, is smaller than your local version. If you change the source format to `quilt` (or possibly `git`) and add a `-1` to the version number, then you can release new packages without worrying about the upstream version.

ddemidov, 2016-11-29:
Ok, I switched to `quilt` for the source format, moved the debian staff to `ev3dev-jessie` branch, and uploaded the packages for debian and raspbian.

ddemidov, 2016-11-29:
Next step would probably be to add a brickman menu entry for starting `rpyc_classic.py`, but I would not know how to do that.

dlech, 2016-11-29:
I would suggest making a systemd service. Then it will be easier to integrate into brickman and you can start it from the command line too. 

The [openroberta service](https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/debian/openrobertalab.service) is a good example to start with. Except you don't need a tty since rpyc is not using graphics, so we can leave that part out.

dlech, 2016-12-08:
> Next step would probably be to add a brickman menu entry

See #129 and #486

`python3-rpyc` package has been published to ev3dev package repository, so closing this issue.

DJuego, 2016-12-08:
Thanks to all!  I expect to make good use of this. ;-)

closed 2016-12-11 by dlech