ev3dev/ev3dev #789: Invoke Sound using  Python and PHP

rsamudrala, 2016-12-01:
Hello
I am using
ev3dev-jessie-ev3-generic-2016-10-17
python-ev3
ngnix and php.

We are trying to invoke a sound from a browser connected to the nginx running on the brick. Our goal is to run a python program that invokes a sound, detects information from an IR sensor and a touch sensor and then run an EV3 large motor.

When I run the python program using SSH it works fine. However, when we run it from a browser page I see the program run (because of entry in a file in /tmp), but the sound is not emitted. I have not tested if sensor and large motor related logic in the python program will work or not yet. The first step is to make sure a sound is emitted.

I have added www-data to audio and ev3dev groups.

The python program is as below - hello.py

```
import ev3dev.ev3 as ev3

ev3.Sound.speak('hello there!')
```
The PHP file is as below - action.php

```
<?php
    system('/usr/bin/python hello.py');
    system ('echo "Python program was invoked" > /tmp/hello.txt')
?>
```

What am I missing here?

thanks in advance.

ddemidov, 2016-12-01:
Try to do 
```py
ev3.Sound.speak('hello there!').wait()
```
It's possible the snippet exits before any sound is produced when you do it from php.

rsamudrala, 2016-12-01:
So, to test further, I changed the PHP as below:
'''
<?php
   system('/usr/bin/python high-pitch.py >> /tmp/high-pitch-output.txt');
   system('/usr/bin/python just-print.py >> /tmp/high-pitch-output.txt');
   system ('echo "Python program was invoked" >> /tmp/high-pitch-output.txt')
?>
'''

high-pitch.py is as below:
'''
import ev3dev.ev3 as ev3
import time

print (time.now().strftime('We are the %d, %b %Y-%m-%d %H:%M'))

ev3.Sound.speak('Hello there').wait()

print ('python called successfully!!')
print (time.now().strftime('We are the %d, %b %Y-%m-%d %H:%M'))
'''

just-print.py is as below:
'''
import time

print ('just print current time: ' + time.strftime("%Y-%m-%d %H:%M"))
'''

and I only see the below, meaning the file which is invoking Sound is throwing error that is getting swallowed somewhere.

'''
just print current time: 2016-12-01 16:34
Python program was invoked
'''

rsamudrala, 2016-12-01:
Invoking "aplay" works

```
import time
import os

os.system('/usr/bin/aplay guinea-pig-daniel_simon-mono.wav')

print ('just print current time: ' + time.strftime("%Y-%m-%d %H:%M"))
```

ddemidov, 2016-12-01:
It should be /usr/bin/python3 in your system call.

Cheers,
Denis

rsamudrala, 2016-12-01:
Is this required even if /usr/bin/python is linked to /usr/bin/python3 ?

robot@ev3dev:~$ ls -la `which python`
lrwxrwxrwx 1 root root 9 Nov 28 22:10 /usr/bin/python -> python3.4
robot@ev3dev:~$

ddemidov, 2016-12-01:
> Is this required even if /usr/bin/python is linked to /usr/bin/python3 ?

Probably not, although that could be a bad idea for other reasons. Can you get the stdout/stderr of the script somehow?

dlech, 2016-12-01:
> I have added www-data to audio and ev3dev groups.

Off topic, but to allow `www-data` to use all ev3dev features you need [more groups](https://github.com/ev3dev/ev3dev-adduser-config/blob/ev3dev-jessie/adduser.conf.ev3dev#L77).

rsamudrala, 2016-12-01:
Redirecting STDERR and STDOUT showed the problem. time.now() does not exist. This was masking the problem of not using "wait()" while invoking Sound.

@dlech - thanks. What other groups shall I add. As mentioned before, I have added audio and ev3dev. Should I add video and plugdev as well? If so, why would I need them?

thanks

dlech, 2016-12-01:
This is a bit of an over-generalization, but... `video` lets you use the frambuffer device to draw on the screen, `tty` lets you do console switching, `bluetooth` lets you use Bluetooth, `plugdev` lets you use USB devices, `i2c` lets you directly communicate with I2C sensors, `input` lets you monitor raw keyboard input (including the EV3 buttons), and I don't remember exactly what `dialout` does, but there may be one or two things that depend on it too.

rsamudrala, 2016-12-01:
Thanks. I may need them later on and this is good to know. 

I appreciate the help. I would like to close this comment. Is that alright?

dlech, 2016-12-01:
Yes, if you feel like you question is answered, please close the issue.

closed 2016-12-01 by rsamudrala