ev3dev/ev3dev #790: Need an ev3dev application launch programs for consistency and ease of use

dlech, 2016-12-01:
One of the trickiest things about ev3dev is running graphical programs. This is mainly due to the fact that we are using Linux the "right way". In order to take control of the display on the EV3, a program has to be running on a virtual console. Programs that are run from a remote terminal can only use the display if they are run as root.

Another common problem is that (non-graphical) programs behave differently when launched from a remote terminal vs. when run from the Brickman file browser. Again, this has to do with virtual consoles and other Linuxy things.

Furthermore, a common problem with launching programs from Brickman is seeing the error messages when a program fails. They are shown on the screen in a very small font for only a brief moment, so it is nearly impossible to tell why the program failed.

For example, here are some of those issues: #388, #456, #575, #772, #764, https://github.com/ev3dev/ev3dev/issues/788#issuecomment-264103841. 

Yet another issue with running programs from a remote terminal is stopping in the middle of a program. Currently each program has to have special handling for this or else, motors could keep running when the program ends. And if a program crashes with a SEGFAULT, there is nothing to stop the motors at all.

So, to solve all of these issues, I propose that we make an ev3dev program runner DBus service. This service can be used by Brickman to replace the existing `sudo chvt runas ...` craziness. There will also be a command line tool that can be used to start programs from a remote terminal exactly as if they had been started using the Brickman file browser.

The way this will work is the DBus service will be managed by systemd so that it will be assigned a tty, similar to the way Brickman and OpenRoberta work now. This service will log in on the tty as the `robot` user. There will be a DBus method to pass a command and arguments to the service that runs the command on the tty.

The sequence will be something like this:
* Switch to the ev3dev runner tty
* Run the command with stderr redirected to systemd journald
* Listen for long button press to kill the program
* When the command ends (whether normally or because of crash or it was killed) stop all motors, restore LEDs, etc.

WasabiFan, 2017-03-16:
I'd like to propose a name for this utility: how about `brickrun` or alternately `bricklaunch`? It fits with the "brickman" name and `brickrun` is pretty fast to type.

moriarty, 2017-03-18:
I just thought it was worth mentioning that ROS has two utilities for this, both a ```roslaunch``` and ```rosrun```. One is used with a launch file, which is an xml file which describes several actions, and parameters, and the other is use to directly start an executable.

http://wiki.ros.org/rosbash#rosrun

http://wiki.ros.org/roslaunch

dlech, 2017-07-25:
I have started `console-runner` as a generic solution for virtual console issue. We can write another ev3dev-specific client (say `brickrun`) that uses the `conrun-server` but also takes care of things like stopping motors.

https://github.com/ev3dev/console-runner/blob/ev3dev-stretch/doc/console-runner.rst

jabrena, 2017-07-26:
Hi @dlech,

The idea to stop motors once the program stop could be useful for everyone.

In a practical way, what is the idea to use console-runner?

Juan Antonio

dlech, 2017-07-26:
The main purpose of the console-runner program is to reserve a virtual terminal for running programs remotely. If you want to run a program from a remote SSH terminal and use the display on the EV3, you either have to run as root or use `conspy` to do so. Otherwise, you just end up drawing over Brickman and text based programs that display on the LCD are not possible.

dlech, 2017-08-31:
Done. https://github.com/ev3dev/brickrun

closed 2017-08-31 by dlech

jabrena, 2017-09-01:
I read the documentation but I have some doubts. Maybe it is necessary to add some examples in the .srt file:
https://github.com/ev3dev/brickrun/blob/ev3dev-stretch/doc/brickrun.rst

Example:

If I have a `jar` file in the path: `/home/robot/` with the name: `demo.jar` the normal execution should be: 

`java -jar demo.jar`

how to use the new command to call java with the parameters?

Juan Antonio

dlech, 2017-09-01:
brickrun -- java -jar demo.jar

You need the `--` because you have a parameter that starts with a single `-`.

jabrena, 2017-09-03:
Hi @dlech,

The new command is available for Jessie?
It is possible to install as a Debian Package manually?

Juan Antonio

dlech, 2017-09-03:
ev3dev-jessie is stable and we will not be adding new features. Backporting packages usually requires rebuilding packages because they were built against newer libraries.

So, if you want a stable platform, use ev3dev-jessie. If you want the newest features, use ev3dev-stretch.