ev3dev/ev3dev #791: Commands not working (as expected)

florensie, 2016-12-07:
I have a couple of questions:

Why can't I use these commands: man & ifconfig?

Why does sudo apt-get install sambaclient not work? I get this: 

```
robot@ev3dev:~$ sudo apt-get install smbclient
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 smbclient : Depends: libarchive13 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
I need it to connect to my NAS that uses smb so I can easily write programs, save them there and run them on my ev3.

mob41, 2016-12-07:
Wait. You are expected to have this issue? So why are you asking...

mob41, 2016-12-07:
>Commands not working (as expected)
Expected? It would be weird why cannot use ```man``` and ```ifconfig```. How can't you use it?

>I'm also struggling to find how to label things on github.
Issue labels can be only labelled by the owners/members

florensie, 2016-12-07:
Sorry for the confusion I meant it like this: Commands not working or not working as they're supposed to (apt). man just gives command not found and so does ifconfig but for some reason ifconfig works when I use sudo. apt-get has the above mentioned problem. Haven't tried with anything else than sambaclient though.

mob41, 2016-12-07:
Have you done ```sudo apt-get update```?

florensie, 2016-12-07:
robot@ev3dev:~$ sudo apt-get update
[sudo] password for robot:
Get:1 http://security.debian.org jessie/updates InRelease [63.1 kB]
Get:2 http://archive.ev3dev.org jessie InRelease [8415 B]
Ign http://httpredir.debian.org jessie InRelease
Get:3 http://httpredir.debian.org jessie Release.gpg [2373 B]
Get:4 http://httpredir.debian.org jessie Release [148 kB]
Get:5 http://security.debian.org jessie/updates/main armel Packages [410 kB]
Get:6 http://archive.ev3dev.org jessie/main armel Packages [41.8 kB]
Get:7 http://security.debian.org jessie/updates/contrib armel Packages [994 B]
Get:8 http://security.debian.org jessie/updates/non-free armel Packages [20 B]
Get:9 http://httpredir.debian.org jessie/main armel Packages [8866 kB]
Get:10 http://httpredir.debian.org jessie/contrib armel Packages [43.3 kB]
Get:11 http://httpredir.debian.org jessie/non-free armel Packages [71.1 kB]
Fetched 9656 kB in 2min 2s (78.6 kB/s)
Reading package lists... Done
robot@ev3dev:~$ man
-bash: man: command not found

ddemidov, 2016-12-07:
I think man files were stripped from the image to save space.

dlech, 2016-12-07:
> Why can't I use these commands: man & ifconfig?

`man` requires that the `man-db` package is installed, which it is not by default. `ifconfig` is in `/sbin`, so is not in the `$PATH` of a normal user. `sudo ifconfig` will work.

> Why does sudo apt-get install sambaclient not work

Now that you have run `sudo apt-get update`, it should work. Always run `sudo apt-get update` before `sudo apt-get install`.

> I think man files were stripped from the image to save space.

This was the case in the very, very early days of ev3dev when we were using Emdebian packages instead of regular Debian packages, but this has not been the case for nearly 3 years.