ev3dev/ev3dev #792: lego-ev3-color

buess, 2016-12-08:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3dev-jessie-ev3-generic-2016-10-17.zip

* My kernel version is: 4.4.32-17-ev3dev-ev3

* My host computer is running: Windows

* I am connecting using: Wi-Fi

### About my issue:

Under "higher" cpu load (e.g. running apt update) the lego-ev3-color sensor starts switching the mode, the value files sometimes disappear

root@ev3dev:/sys/class/lego-sensor/sensor0# cat value0
cat: value0: No such device or address

while [ 1 ]; do  cat /sys/class/lego-sensor/sensor0/mode; done
RGB-RAW Thu Dec  8 23:05:56 UTC 2016
RGB-RAW Thu Dec  8 23:05:57 UTC 2016
RGB-RAW Thu Dec  8 23:05:57 UTC 2016
RGB-RAW Thu Dec  8 23:05:58 UTC 2016
                  Thu Dec  8 23:05:58 UTC 2016
COL-REFLECT Thu Dec  8 23:05:58 UTC 2016
RGB-RAW Thu Dec  8 23:05:59 UTC 2016
RGB-RAW Thu Dec  8 23:05:59 UTC 2016
RGB-RAW Thu Dec  8 23:06:00 UTC 2016
COL-CAL Thu Dec  8 23:06:00 UTC 2016
REF-RAW Thu Dec  8 23:06:01 UTC 2016
COL-AMBIENT Thu Dec  8 23:06:01 UTC 2016
RGB-RAW Thu Dec  8 23:06:02 UTC 2016
RGB-RAW Thu Dec  8 23:06:02 UTC 2016
RGB-RAW Thu Dec  8 23:06:02 UTC 2016

dmesg show the following messages:
[Thu Dec  8 23:05:54 2016] ttyS1: buffer overrun
[Thu Dec  8 23:05:54 2016] ttyS1: buffer overrun
[Thu Dec  8 23:05:57 2016] tty ttyS1: Reconnected due to: Received duplicate percent scaling INFO.
[Thu Dec  8 23:05:57 2016] ev3_uart_receive_buf: 13 callbacks suppressed
[Thu Dec  8 23:05:57 2016] ttyS1: buffer overrun
[Thu Dec  8 23:05:57 2016] ttyS1: buffer overrun
[Thu Dec  8 23:05:57 2016] ttyS1: buffer overrun
[Thu Dec  8 23:06:00 2016] tty ttyS1: Reconnected due to: Invalid name INFO.
[Thu Dec  8 23:06:00 2016] ttyS1: buffer overrun
[Thu Dec  8 23:06:00 2016] ttyS1: buffer overrun
[Thu Dec  8 23:06:00 2016] ttyS1: buffer overrun
[Thu Dec  8 23:06:03 2016] tty ttyS1: Reconnected due to: Received duplicate percent scaling INFO.
[Thu Dec  8 23:06:06 2016] ttyS1: buffer overrun
[Thu Dec  8 23:06:06 2016] ttyS1: buffer overrun
[Thu Dec  8 23:06:06 2016] ttyS1: buffer overrun
[Thu Dec  8 23:06:06 2016] ttyS1: buffer overrun

dlech, 2016-12-09:
Sounds like we need to re-open #679.

dlech, 2016-12-09:
Are you sure the `value*` attributes are actually disappearing? I would expect to see a no such **file** error if they were. The no such **device** error means that the sensor is not connected but the `value*` attribute is still there.

buess, 2016-12-09:
The files are still there, i only got the message "no such device..."

bmegli, 2016-12-10:
Looks like ENXIO (No such device or address) when reconnecting UART sensor in #690

pepijndevos, 2016-12-22:
I'm occasionally seeing the same behaviour when reading the EV3 Color sensor in a loop from python. It runs for a while and then suddenly crashes with the mentioned `no such device or address` error. I have a logic analyser at hand, but no idea what to look for.

dlech, 2016-12-22:
To those experiencing this issue: Are there any errors in `dmesg`? In particular, I would expect to see "Reconnected due to ...".

buess, 2016-12-22:
yes, i get Received INFO for incorrect mode:

[  325.309309] ttySU1: buffer overrun
[  328.122389] tty ttySU1: Reconnected due to: Received INFO for incorrect mode.
[  332.121605] tty ttySU1: Reconnected due to: Received INFO for incorrect mode.
[  332.288732] ev3_uart_receive_buf: 40 callbacks suppressed
[  332.288787] ttySU1: buffer overrun
[  332.292100] ttySU1: buffer overrun
[  332.295105] ttySU1: buffer overrun
[  332.300407] ttySU1: buffer overrun
[  332.305627] ttySU1: buffer overrun
[  332.310848] ttySU1: buffer overrun
[  332.317257] ttySU1: buffer overrun
[  335.121144] tty ttySU1: Reconnected due to: Received INFO for incorrect mode.
[  335.290490] ttySU1: buffer overrun

pepijndevos, 2016-12-24:
Yes, Reconnection happens:

```
[   39.137709] lego-sensor sensor0: Registered 'lego-ev3-color' on 'in2'.
[   43.828142] using random self ethernet address
[   43.828245] using random host ethernet address
[   45.319603] using random self ethernet address
[   45.319705] using random host ethernet address
[   45.859726] usb0: HOST MAC 32:16:53:35:a8:52
[   45.863829] usb0: MAC 22:16:53:35:a8:52
[   45.902873] usb1: HOST MAC 12:16:53:35:a8:52
[   45.906892] usb1: MAC 02:16:53:35:a8:52
[   46.027777] lego-port port2: Added new device 'in3:ev3-uart-host'
[   46.633826] configfs-gadget gadget: high-speed config #2: c
[   47.497694] lego-sensor sensor1: Registered 'lego-ev3-ir' on 'in3'.
[  170.089240] hrtimer: interrupt took 138345 ns
[  547.456315] ttySU1: buffer overrun
[  547.459087] ttySU1: buffer overrun
[  547.463219] ttySU1: buffer overrun
[  547.465524] ttySU1: buffer overrun
[  547.469072] ttySU1: buffer overrun
[  547.471977] ttySU1: buffer overrun
[  547.475556] ttySU1: buffer overrun
[  547.478771] ttySU1: buffer overrun
[  547.481923] ttySU1: buffer overrun
[  547.485671] ttySU1: buffer overrun
[  926.286448] ev3_uart_receive_buf: 38 callbacks suppressed
[  926.286510] ttySU1: buffer overrun
[  926.289304] ttySU1: buffer overrun
[  926.292745] ttySU1: buffer overrun
[  926.295477] ttySU1: buffer overrun
[  926.299166] ttySU1: buffer overrun
[  926.302437] ttySU1: buffer overrun
[  926.305550] ttySU1: buffer overrun
[  926.308405] ttySU1: buffer overrun
[  926.312352] ttySU1: buffer overrun
[  927.286686] ttySU1: buffer overrun
[  992.880393] ev3_uart_receive_buf: 1 callbacks suppressed
[  992.880453] ttySU1: buffer overrun
[  992.883716] ttySU1: buffer overrun
[  992.886852] ttySU1: buffer overrun
[  995.750153] tty ttySU1: Reconnected due to: No data since last keep-alive.

```

BertLindeman, 2017-01-17:
Similar here:
EV3-brick with:
```
uname -a
Linux ev3dev 4.4.32-17-ev3dev-ev3 #1 PREEMPT Wed Nov 16 11:12:06 CST 2016 armv5tejl GNU/Linux
```

Rebooted without any sensor plugged-in.
(EV3 m+l-motors were plugged-in)

Plugged-in ev3-colorsensor followed by:
```
echo RGB-RAW > /sys/class/lego-sensor/sensor0/mode
```
Waited 10 minutes to be sure that nothing I did interfered.

After that, seen in `dmesg -w`:
```
[Tue Jan 17 13:16:39 2017] lego-port port1: Added new device 'in2:ev3-uart-host'
[Tue Jan 17 13:16:40 2017] Registered EV3 UART sensor line discipline. (29)
[Tue Jan 17 13:16:43 2017] lego-sensor sensor0: Registered 'lego-ev3-color' on 'in2'.
[Tue Jan 17 13:27:30 2017] ttyS0: buffer overrun
[Tue Jan 17 13:27:30 2017] ttyS0: buffer overrun
[Tue Jan 17 13:27:30 2017] ttyS0: buffer overrun
[Tue Jan 17 13:27:30 2017] ttyS0: buffer overrun
[Tue Jan 17 13:27:30 2017] ttyS0: buffer overrun
[Tue Jan 17 13:27:30 2017] ttyS0: buffer overrun
[Tue Jan 17 13:27:30 2017] ttyS0: buffer overrun
[Tue Jan 17 13:27:33 2017] tty ttyS0: Reconnected due to: Received INFO for incorrect mode.
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
[Tue Jan 17 13:31:28 2017] tty ttyS0: Reconnected due to: Received INFO for incorrect mode.
```

The messages in dmesg state **ttyS0**.
Looking at the sensor:
```
ll /sys/class/lego-sensor/
lrwxrwxrwx 1 root root 0 Jan 17 13:16 sensor0 -> ../../devices/platform/serial8250.0/tty/ttyS0/sensor0/

cat  /sys/class/lego-sensor/sensor0/device/uevent
MAJOR=4
MINOR=64
DEVNAME=ttyS0
```

So I assume this proves the EV3-colorsensor is causing the problems.

bmegli, 2017-01-20:
I was getting occasional buffer overruns also in `other-uart` mode when communicating with non-standard devices.

This may be separate issue and/or a way to debug problem without added complexity of EV3 UART Sensor Line Discipline.

bmegli, 2017-01-20:
@dlech from other side - I remember that you were considering once to increase the [EV3_UART_BUFFER_SIZE](https://github.com/ev3dev/lego-linux-drivers/blob/5ccdd2dc17f097d7a766ecad18b54d5372fbf88c/sensors/ev3_uart_sensor_ld.c#L79).

This might help if the overruns are the cause.

FWIW - 256 bytes at 115200 baudrate can be filled in around 20 ms.

dlech, 2017-01-20:
> I was getting occasional buffer overruns also in other-uart mode when communicating with non-standard devices.

Thanks for pointing this out. I hadn't noticed this before, but the error

    [Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun

is coming from the kernel tty driver and not our driver. So, I'm not sure exactly which buffer this is, but it is certainly not our 256 byte buffer. If it did, it would say `Buffer overrun.` (capital B and period at the end).

bmegli, 2017-01-21:
@dlech wrote
> Thanks for pointing this out. I hadn't noticed this before, but the error
``` bash
[Tue Jan 17 13:31:25 2017] ttyS0: buffer overrun
```
> is coming from the kernel tty driver and not our driver. So, I'm not sure exactly which buffer this is, but it is 

From what @buess and @BertLindeman wrote this is on EV3, not BrickPi so may be coming from  [ev3_uart_sensor_ld](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c#L1039) and could be affected by the [EV3_UART_BUFFER_SIZE](https://github.com/ev3dev/lego-linux-drivers/blob/5ccdd2dc17f097d7a766ecad18b54d5372fbf88c/sensors/ev3_uart_sensor_ld.c#L79)

--------------------------------------------------------------

Come to think about that I don't remember exact message I had in `other-uart`, I now found some logs
from 4.4.24-16-ev3dev-ev3 where I had:

``` bash
[ 1659.245249] serial8250: too much work for irq53
[ 1659.447623] serial8250: too much work for irq53
...
[ 1664.277011] serial8250_interrupt: 14 callbacks suppressed
```

which led to CRC failures in my code using other-uart (and lose of synchronization)

I would have to repeat the tests with new kernel to know the messages for sure (but the above are kind of overruns or lose of data anyway).

dlech, 2017-01-21:
> From what @buess and @BertLindeman wrote this is on EV3, not BrickPi so may be coming from ev3_uart_sensor_ld and could be affected by the EV3_UART_BUFFER_SIZE

OK, I must have been imaging things yesterday. [Here is where the error message is coming from](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c#L1039).

Anyone want to try compiling a kernel and making `EV3_UART_BUFFER_SIZE` bigger to see if it fixes the problem?

> [ 1659.245249] serial8250: too much work for irq53

See #623 and #679. This could have been caused by having kernel messages logged to the serial port.

bmegli, 2017-01-21:
> OK, I must have been imaging things yesterday

Yes.

> Anyone want to try compiling a kernel and making EV3_UART_BUFFER_SIZE bigger to see if it fixes the problem?

Yes.

> See #623 and #679. This could have been caused by having kernel messages logged to the serial port.

Yes. 

------------------------------------------------------------------------------------------

Message logging - thank you for the hint. 

I may check `EV3_UART_BUFFER_SIZE` with happiness but my personal task buffers are filled by work (poorly paid but paid) lately. I hope I will not get overrun in them. So it will be scheduled for unknown future unless somebody does it earlier.

bmegli, 2017-01-25:
In all cases I set RGB-RAW first. Sensor plugged to port 2.

## ev3dev-jessie-ev3-generic-2016-12-21 image (download section)

I was able to reproduce the problem.

The worst effect I had was when sensor changed mode to something different from RGB-RAW and stayed in it after reconnect.

## ev3 kernel [4f05fe2](https://github.com/ev3dev/ev3-kernel/commit/4f05fe25f9448fa9dab85bb0785d42bb798480c8) + lego-linux-drivers [03011a5](https://github.com/ev3dev/lego-linux-drivers/commit/03011a571922cfee6a245b310be5930813488eb6)

This is with kernel and lego linux drivers from github and default EV3_UART_BUFFER_SIZE=256 (roughly 18 ms buffer time)

I was able to reproduce the problem.

## ev3 kernel [4f05fe2](https://github.com/ev3dev/ev3-kernel/commit/4f05fe25f9448fa9dab85bb0785d42bb798480c8) + lego-linux-drivers [36a668d](https://github.com/bmegli/lego-linux-drivers/commit/36a668d304ab5f3b990f36091551ff888fff9bf2)

This is with kernel from github and lego linux drivers with modified EV3_UART_BUFFER_SIZE=1024 (roughly 70 ms buffer time)

Apart from single message just right after boot (without anything around):

``` bash
[   37.440078] tty ttyS0: Reconnected due to: No data since last keep-alive.
```

I was not able to reproduce the problem in 1 hour time (stopwatch measured)

I tried to put some load using 5 simultaneous ssh connections with:
- while [ 1 ]; do  cat /sys/class/lego-sensor/sensor0/mode; done
- while [ 1 ]; do  cat /sys/class/lego-sensor/sensor0/value0; done
- top (long term average load was around 4)
- sudo apt-get update (started around 3 times, each time took around 7 minutes)
- dmesg -w

## ev3 kernel [4f05fe2](https://github.com/ev3dev/ev3-kernel/commit/4f05fe25f9448fa9dab85bb0785d42bb798480c8) + lego-linux-drivers [bc96300](https://github.com/bmegli/lego-linux-drivers/commit/bc96300ef95ac16219ec883ea21734346d830257)

This is with kernel from github and lego linux drivers with modified EV3_UART_BUFFER_SIZE=128 (roughly 9 ms buffer time)

With 5 simultaneous ssh connections like above, sometimes I get overruns all the time and "reconnected to..." for various reasons. The sensor is blinking (reconnecting, changings modes etc).

This is especially evident in early phase of `apt-get update`.

I was able to reproduce the effect of permanent mode change (from RGB-RAW to COL-REFLECT)

Without load the sensor generally works stable.

The [log](http://pastebin.com/jyP1G7bM) (exires in 1 month)

dlech, 2017-01-25:
So, you will send a pull request for 36a668d? :wink:

bmegli, 2017-01-25:
PR in ev3dev/lego-linux-drivers/pull/59

bmegli, 2017-01-25:
Side note - 512 bytes could also be a valid (but not tested) option, EV3 probably doesn't need 70 ms buffer time (**rough** calculations at 1024 bytes buffer and 115200 UART), with 256 (roughly 20 ms buffer time) overruns were already rare.

dlech, 2017-01-25:
FWIW, most of the sensors run at 57600.

dlech, 2017-01-27:
Fix has been released in [18-ev3dev kernel](http://www.ev3dev.org/news/2017/01/25/kernel-release-cycle-18/)

closed 2017-01-27 by dlech

BertLindeman, 2017-01-27:
Confirmed fixed.
No more buffer overruns seen using the EV3 color sensor running
```
uname -a
Linux ev3dev 4.4.44-18-ev3dev-ev3 #1 PREEMPT Wed Jan 25 19:40:14 CST 2017 armv5tejl GNU/Linux
```

But if I have the EV3 sensor plugged-in at boot time (port in2) I have seen many
`too much work for irq53` 
But **not each** boot In fact only **one** boot

dlech, 2017-01-27:
Please do not report new issues on closed issues. Open a new issue for each new problem.

bmegli, 2017-01-27:
@BertLindeman but thank you for reporting that concerned issue is fixed! 😄