ev3dev/ev3dev #793: sudo halt does not shut down the EV3

pepijndevos, 2016-12-12:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3dev-jessie-ev3-generic-2016-10-17.img

* My kernel version is: 4.4.24-16-ev3dev-ev3

* My host computer is running: Linux

* I am connecting using: USB

### About my issue:

The other day I tried to shut down via the shell instead of the GUI, so I typed `sudo halt` and then it started shutting down, but did not actually turn off the power and stayed on the linux console.

I'm a bit confused about the meaning of the shutdown/halt/poweroff commands.

I would also greatly appreciate if you could clarify the procedure. I heard that the EV3 copies some memory to/from flash every time, but I can't figure out what and why. What happens if I kill the power like I did yesterday? Are all my changes lost?

laurensvalk, 2016-12-12:
> I'm a bit confused about the meaning of the shutdown/halt/poweroff commands.

I usually use 
```
sudo shutdown -h now
```

> I heard that the EV3 copies some memory to/from flash every time, but I can't figure out what and why. What happens if I kill the power like I did yesterday? Are all my changes lost?

This normally happens with the standard LEGO firmware, unless your use a micro SD card. If I recall correctly, it's because flash can't be accessed during runtime.

This doesn't usually happen with ev3dev, because everything is stored on the micro SD card. However, a non-clean shutdown could corrupt your data, similar to what might happen if you suddenly unplug your PC.

dlech, 2016-12-12:
> I'm a bit confused about the meaning of the shutdown/halt/poweroff commands.

This is the best explanation I could find: http://unix.stackexchange.com/a/196014/91467

TL;DR; `halt` stops the system but does not power it off, `poweroff` stops the system and powers it off. `shutdown` can do either of these depending on which command line option you use.

> I heard that the EV3 copies some memory to/from flash every time

ev3dev does not touch the flash memory at all. Everything is on the SD card and it acts like a normal computer where things are saved immediately to disk.

> I usually use `sudo shutdown -h now`

`sudo poweroff` does the same thing and is less typing.

closed 2016-12-17 by pepijndevos