ev3dev/ev3dev #794: Force the running program to exit if it stays running after sending a SIGTERM

WasabiFan, 2016-12-17:
Currently, it looks like Brickman sends a SIGTERM and then takes over the display and other devices without actually confirming that the child process exited. I think it would make send for it to force it to die if the application didn't exit after some timeout period.

ddemidov, 2016-12-17:
There was a similar issue reported recently by @ndward about the RPyC daemon and brickman. When you launch the rpyc_server.sh script from brickman and then long-press the back button when you are done, it stays running in the background

ddemidov, 2016-12-17:
It looks like the RPyC does respond to the SIGTERM signal though:
```
$ cat rpyc_server.sh
#!/bin/bash
python3 `which rpyc_classic.py`
```

When I send the SIGTERM signal to the process, it does get terminated:
```
$ ./rpyc_server.sh
INFO:SLAVE/18812:server started on [0.0.0.0]:18812
Terminated
```

WasabiFan, 2016-12-17:
It looks like the problem there is that somewhere the script spawns a second process, and only the first gets terminated (and it doesn't properly terminate the second). I'm not familiar enough with the Linux process model to know if there's anything that can be done about this; can you access info about which process started another?

ddemidov, 2016-12-17:
Well, I did send the signal to the parent bash process that spawned the RPyC server, and the server got terminated. I also tried terminating the child python process directly, and that works as well.

dlech, 2016-12-17:
Currently, Brickman only sends a signal to any process attached to the tty. If RPyC is running as a daemon (or is otherwise not attached to the tty somehow), that would explain why it is not receiving a signal from Brickman. Implementing #790 will take care of this issue and many more.

ddemidov, 2016-12-18:
How do you attach a process to the tty? Does not brickman do it already for the processes it starts? RPyC runs in foreground when started from command line, and it does allow to draw on the LCD.

pepijndevos, 2016-12-18:
I can confirm that my program does exit when sending it a SIGTERM from another terminal. (however, not gracefully)

My code is something like this, with the functions literally containing nothing but random conditionals and EV3 API calls. No weird tricks or threads being used. I'm not even using sensors, it just goes through some random poses.

```
if __name__ == "__main__":
    try:
        while True:
            run() # motor commands
            draw() # LCD commands
            sleep(1)
    finally:
        reset() # move motors to starting positions
```

When I ctrl-c the program or send it SIGINT, it gets a keyboard interrupt, runs the reset code and exits. When I send it SIGTERM, it just stops without resetting. When I run and stop it via Brickman, it happily continues running.

dlech, 2016-12-18:
> How do you attach a process to the tty?

"Attached" might not be the best word choice. A process becomes "attached" to a tty simply because the program is started from that tty. To better understand what I mean, run `ps ax`. You will see, for example, that `brickman` is running on `tty1`.

```
  PID TTY      STAT   TIME COMMAND
    1 ?        Ss     0:13 /sbin/init splash
...
  308 ?        Ss     0:00 /usr/sbin/sshd -D
  317 tty1     Ssl+   8:07 /usr/sbin/brickman
  324 ?        Ss     0:00 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
...
```

Whereas most processes have a `?` under `TTY`.

ddemidov, 2016-12-18:
Here is what I see after I start the rpyc server:
```
  378 tty1     Ssl+   0:27 /usr/sbin/brickman
  393 ?        S      0:02 [RTW_CMD_THREAD]
  476 ?        Ss     0:00 /usr/sbin/ntpd -p /var/run/ntpd.pid -g -u 106:110
  496 ?        Ss     0:00 /usr/sbin/nmbd -D
  527 ?        S      0:12 [kworker/0:0]
  528 ?        S      0:00 [kworker/0:2]
  532 tty1     S+     0:00 /bin/openvt --switch --wait -- /bin/bash -c /usr/bin/tty >/tmp/brickman.mdnUdT/tty;tty=$(/bin/cat /tmp/brickman.mdnUdT/tty);/bin/chown robot: $tty;/sbin/runuser --login --shell /bin/bash --command '(cd /home/robot && exec /home/robot/rpyc_server.sh)' robot
  535 tty2     Ss+    0:00 /bin/bash -c /usr/bin/tty >/tmp/brickman.mdnUdT/tty;tty=$(/bin/cat /tmp/brickman.mdnUdT/tty);/bin/chown robot: $tty;/sbin/runuser --login --shell /bin/bash --command '(cd /home/robot && exec /home/robot/rpyc_server.sh)' robot
  554 tty2     S+     0:00 /sbin/runuser --login --shell /bin/bash --command (cd /home/robot && exec /home/robot/rpyc_server.sh) robot
  556 ?        S      0:00 [kworker/u2:1]
  557 ?        Ss     0:00 /lib/systemd/systemd --user
  560 ?        S      0:00 (sd-pam)         
  563 ?        Ss     0:00 -bash -c (cd /home/robot && exec /home/robot/rpyc_server.sh)
  566 ?        S      0:00 /bin/bash /home/robot/rpyc_server.sh
  568 ?        Sl     0:11 python3 /usr/bin/rpyc_classic.py -m oneshot
```

And this is with a random python script (`hello.py`):
```
  616 tty1     S+     0:00 /bin/openvt --switch --wait -- /bin/bash -c /usr/bin/tty >/tmp/brickman.z968HM/tty;tty=$(/bin/cat /tmp/brickman.z968HM/tty);/bin/chown robot: $tty;/sbin/runuser --login --shell /bin/bash --command '(cd /home/robot && exec /home/robot/hello.py)' robot
  619 tty3     Ss+    0:00 /bin/bash -c /usr/bin/tty >/tmp/brickman.z968HM/tty;tty=$(/bin/cat /tmp/brickman.z968HM/tty);/bin/chown robot: $tty;/sbin/runuser --login --shell /bin/bash --command '(cd /home/robot && exec /home/robot/hello.py)' robot
  638 tty3     S+     0:00 /sbin/runuser --login --shell /bin/bash --command (cd /home/robot && exec /home/robot/hello.py) robot
  641 ?        Ss     0:00 -bash -c (cd /home/robot && exec /home/robot/hello.py)
  644 ?        R      0:02 /usr/bin/python3 /home/robot/hello.py
```

Looks like at some point the launched program looses its association with the tty.

dlech, 2016-12-18:
The current implementation of Brickman will only send `SIGTERM` to the processes with `tty2` (or 3, etc.) This usually has the effect of killing any child processes (you can see the process tree by running `ps axf` or `pstree`).  You should, for example, see that `python3 /usr/bin/rpyc_classic.py -m oneshot` is an ancestor of `/sbin/runuser ...`. But, in the case of python programs with threads, this does not seem to kill python. The python process does not receive `SIGTERM` from Brickman.

ddemidov, 2016-12-19:
Could be related:
http://stackoverflow.com/questions/19988726/why-is-signal-sigterm-not-dealt-with-properly-in-my-main-thread

In short, the main thread is the one receiving and handling the signal, so it should be alive when the signal comes. I'll need to check what happens inside the `rpyc_classic.py` script.

dlech, 2016-12-19:
> In short, the main thread is the one receiving and handling the signal, so it should be alive when the signal comes

No. As I already said, the signal does *not* come to the python program.

This is what is going on:
* http://stackoverflow.com/questions/8533377/why-child-process-still-alive-after-parent-process-was-killed-in-linux
* http://unix.stackexchange.com/questions/149741/why-is-sigint-not-propagated-to-child-process-when-sent-to-its-parent-process

So, what I think Brickman should be doing instead of the current behavior is sending `SIGKILL` instead of `SIGTERM` and it should be sent to the entire process group, not just the parent process.

ddemidov, 2016-12-19:
Would something like ddemidov/brickman@3ed1939 work? (not yet tested)

ddemidov, 2016-12-19:
https://github.com/ddemidov/brickman/commit/67ff9b8c6ee7947e28c0949d30f14a431dbb4bc9 should work better (still did not try to compile brickman).

pepijndevos, 2016-12-19:
As far as I know SIGKILL‎ is not a polite way of asking the program to stop. Where does this leave programs to gracefully shut down?From the perspective of Python, SIGINT would be best. That can be handled by a simple try-catch without messing with signal handling and possibly breaking "well behaved" Unix behaviour by, for example not responding to SIGKILL in a timely manner.                                                                                                                                                                                  ‎

dlech, 2016-12-19:
> ddemidov/brickman@67ff9b8 should work better (still did not try to compile brickman).

This is probably OK - except that it needs to be wrapped in `bash -c` so that the shell bits work. But `pkill --parent %s` where `%s` is `Subprocess.get_identifier ()` might be even simpler.

> As far as I know SIGKILL‎ is not a polite way of asking the program to stop.

Which is exactly why we should be using it here. The idea is that your program should be stopping itself somehow and that the long press of the back button is a last resort way to kill a buggy program that cannot be stopped otherwise. It is common for programs to handle `SIGTERM` and `SIGINT` and there is no guarantee that either of these will actually stop the program.

> Where does this leave programs to gracefully shut down?

Brickman takes care of stopping motors, etc. when a program ends whether it ended normally, crashed or was killed. So, technically, it does not matter if a program gracefully shuts down. If you want your program to gracefully shut down on (short) back button press or by other means, this needs to be implemented in the program itself.

WasabiFan, 2016-12-19:
> Which is exactly why we should be using it here. The idea is that your program should be stopping itself somehow and that the long press of the back button is a last resort way to kill a buggy [...]

I don't think we should consider it a "last resort". I expect that many programs would just keep running indefinitely, such as a remote-control app which continuously accepts input until terminated. That doesn't mean that it shouldn't be allowed to exit politely, though. I would rather see a system where Brickman sent a `SIGTERM` and waited until it exited, and sent a `SIGKILL` if the target didn't exit after a set timeout. I also don't think it makes sense for every program to have to implement its own button-based exit logic; there should be a universal and consistent way to exit an app without killing it outright.

> Brickman takes care of stopping motors, etc. when a program ends whether it ended normally, crashed or was killed. So, technically, it does not matter if a program gracefully shuts down.

Things like log files, data storage, remote notifications and mechanism-specific reset require that the app gets a chance to clean up.

pepijndevos, 2016-12-19:
I agree a unixy and consistent way to exit programs has my preference. (again, for Python SIGINT seems the way to go) Saves me a bunch of work every time I write a simple program. Also imagine the possibility of running generic Linux programs or simple shell scripts from brickman. None of those will implement their own exit logic other than ctrl-c.

dwalton76, 2016-12-20:
```I would rather see a system where Brickman sent a SIGTERM and waited until it exited, and sent a SIGKILL if the target didn't exit after a set timeout.```

+1 If brickman sends a ```SIGKILL``` first then yes it can stop the motors afterwards but they may not be stopped where they need to be in order to run the robot again.  Rubiks solvers for example, when you kill those they need to put the motor back in an exact position for the robot to work the next time.  Programs need to be given a chance to exit cleanly.

dlech, 2016-12-20:
But if your Rubik's cube solver is destroying itself (e.g. motors going crazy), I don't think you want to wait for a timeout before killing the program and stopping everything.

> I also don't think it makes sense for every program to have to implement its own button-based exit logic;

This makes as much sense as every program having to implement its own signal handler for `SIGTERM`. And unless you are an experienced Linux users, you won't even know what signals are.

It would be simple enough to add a `ev3.interrupt_on_back_button()` function to ev3dev-lang-python that users could add to the beginning of their program that handles this. It could send `SIGINT` just like a keyboard interrupt. Then the basic program template would look like...

```python
from ev3dev import ev3

if __name__ == "__main__":
    try:
        ev3.interrupt_on_back_button()
        # init other stuff

        while True:
            # do stuff
    except:
        # reset motors, etc.
```

> there should be a universal and consistent way to exit an app without killing it outright.

What do you suggest?

WasabiFan, 2016-12-20:
> But if your Rubik's cube solver is destroying itself (e.g. motors going crazy), I don't think you want to wait for a timeout before killing the program and stopping everything.

Good point. I'm thinking that this won't be an issue for two reasons:

- The timeout would be short (~1 second) so in the worst-case scenario not much damage can be done
- SIGTERM should handle this case unless the programmer explicitly prevents exiting on such a signal

These two factors mean that the intersection where this becomes an issue is the area between "an out-of-control robot", "a mechanism which will be destroyed in seconds" and "a program that eats SIGTERMs without stopping".

> This makes as much sense as every program having to implement its own signal handler for SIGTERM. And unless you are an experienced Linux users, you won't even know what signals are.

That's a faulty comparison. The point of handling it in the proposed manner is that advanced users can utilize the extra wiggle room, while less-experienced users can simply ignore the concept altogether (every platform I am aware of will exit on SIGTERM unless you've done something to prevent that). Even if a beginner accidentally does something that makes their code ignore the initial signal, it definitely won't ignore the second!

> It would be simple enough to add a ev3.interrupt_on_back_button() function to ev3dev-lang-python that users could add to the beginning of their program that handles this.

Sure it's possible and fairly straightforward, but in practice it has multiple flaws. This means that...

- The exit-on-back-button functionality is only available pre-implemented if you are using our Python library
- It will vary in operation depending on the program that is being run, giving the user no consistent means of ending a program politely
- It will only work if the programmer remembers to call that function or implement it themselves, which is inevitably going to be forgotten

> What do you suggest?

The aforementioned timeout-based method.

dlech, 2017-01-14:
Fix is released in `brickman_0.8.1`

WasabiFan, 2017-01-14:
This doesn't resolve the issue. This one is specifically about a uniform timeout-based scheme of ending programs, as opposed to figuring out how to end the whole process tree.

dlech, 2017-01-14:
OK. I didn't see another issue for the more immediate problem of not being to kill everything that was started from Brickman.

WasabiFan, 2017-01-14:
IIRC, that one is on the ev3dev-lang-python repo.