ev3dev/ev3dev #795: EVB skipping medium motor encoder pulses

laurensvalk, 2016-12-21:
The Medium EV3 Motor seems to be skipping pulses from the encoder when connected to EVB. 

### Reproduce:
To reproduce, easy method:

- Attach Medium EV3 motor to EVB. 
- Add something to motor shaft to mark 0 position.
- Open devices/port view on Brickman, or read the position otherwise
- Move the shaft around, like back and forth for a while, then put it back
- Position is very far off. The faster you rotate, the worse it gets. Not a few degrees, easily hundreds of degrees after some time.

To reproduce, harder method:
- [Build this](https://www.youtube.com/watch?v=G0Wb-dzgARM), then figure out why it won't come to stand still at the end.

### System information:

* I am using: Fatcatlab EVB. **But possibly not the release hardware.**

* I installed ev3dev using this image file: ev3dev-jessie-bone-generic-2016-10-17.img

* My kernel version is: 4.4.24-ti-rt-r55-16-ev3dev-bb.org

dlech, 2016-12-21:
Sounds like a duplicate of #710. Does upgrading the kernel fix it?

laurensvalk, 2016-12-21:
After dist-upgrade Brickman hangs at boot (or doesn't respond to buttons), so I'll try a new image later this week and get back with details.

dwalton76, 2016-12-22:
I have seen similar issues on EVB lately but with large motors
- reset the motor
- rotate it by hand 3 full rotations
- rotate it back 3 full rotations to the starting position
- the position is not 0, it can be off by 100s of degrees

dlech, 2016-12-22:
@dwalton76, so is this the same as #710 or a new issue?

dwalton76, 2016-12-22:
A new issue, with #710 you would say "rotate X degrees" but it would rotate far more than X, I could no longer reproduce that.  What I was seeing was after moving some number of degrees and trying to move the motor back to degree 0 always failed.  I then tried rotating by hand and could see the issue where even though I had returned the motor to its original position its "position" was not 0, it would be off by 10s or 100s of degrees.

Sorry I forgot to file it, got busy/distracted.

laurensvalk, 2016-12-24:
Same issue occurs with clean 2016-12-21 image.

However, this does not occur within the LMS2012 app. This means we can rule out a hardware problem, I think.

laurensvalk, 2016-12-30:
This also occurs with EV3, snapshot-ev3dev-jessie-ev3-generic-2016-12-21.

laurensvalk, 2016-12-30:
I have yet to test this with the Large motor. It might have the same issue, but it generally just turns more slowly so the issue is less apparent.

dlech, 2016-12-30:
If you run `watch -n 0.1 cat /sys/class/tacho-motor/motor0/speed` when you do this, about how fast does the EV3/EVB say that you are turning the motor by hand?

laurensvalk, 2016-12-30:
Varying roughly in between -2000 and 2000. I'm currently rotating it back and forth manually near 0 position.

laurensvalk, 2016-12-30:
I see now that 2000 is more than what is currently counted on during normal operation. 

I've only tried normal operation on the EVB so far (similar to what @dwalton76 does), but I'll try it on the EV3 as well.

Things to try when looking for missing pulses

- Constant speed in one direction (and try both forward and backward)
- Fast direction changes (100 to -100 duty cycle)
- Frequent changes (by putting a jitter signal on the duty cycle)
- Slow direction changes (100, decrease slowly to 0, then slowly to -100)
- Keeping the processor busy with other things while doing the above

laurensvalk, 2017-01-03:
A couple thoughts/questions from looking at the [relevant code](https://github.com/ev3dev/lego-linux-drivers/blob/master/ev3/legoev3_motor.c#L141):

- What currently happens when interrupts come in faster than they are handled? Are they queued or discarded? 
If queued and then handled quickly in order, there might be less time between the interrupt function calls than the current 400 usec threshold, thus causing errors. 

- If the interrupt is called on every edge, we know which edge triggered it (except with noise). We could check that against the pin value read during the interrupt function run, which might run at a later time. (Not directly a solution but could be useful for debugging the above.)

- Is it possible to time stamp the interrupt rather than reading the time when the interrupt is handled? (Not directly a solution since we'd still need to read DIR at the right time.)

- The 400 usec threshold is a bit long as the motor can turn faster than the maximum speed currently used in the code.

- Am I correct in assuming we have TACHO_SAMPLES = 128 time stamps available of the most recent edge changes? That's nice as it allows for more advanced error checking. (To begin with, instead of checking the length of an interval to determine what was noise, we could look for impossible accelerations.)

dlech, 2017-01-03:
> What currently happens when interrupts come in faster than they are handled? Are they queued or discarded?

There cannot be a new interrupt until the handler has exited, so if the gpio is changing state faster than the handler can run, then we say that those interrupts are "missed".

> Is it possible to time stamp the interrupt rather than reading the time when the interrupt is handled?

No, it is not possible. The only reason why the interrupt handler would not run exactly when the gpio changes states is because interrupts are disabled in the processor. There is no way to know when an interrupt occurred when interrupts are disabled. 

> Am I correct in assuming we have TACHO_SAMPLES = 128 time stamps available of the most recent edge changes?

Yes, I believe so.

dlech, 2017-01-24:
> Reproduce:
> 
> To reproduce, easy method:
> 
>     Attach Medium EV3 motor to EVB (or EV3).
>     Add something to motor shaft to mark 0 position.
>     Open devices/port view on Brickman, or read the position otherwise, e.g.:
> 
> while true; do cat position; sleep 0.1; done
> 
>     Move the shaft around, like back and forth for a while, then put it back
>     Position is very far off. The faster you rotate, the worse it gets. Not a few degrees, easily hundreds of degrees after some time.

I am not able to reproduce this on an EV3. I wonder if there is some condition I am missing, like having 4 color sensors plugged in on RGB-RAW mode or something like that.

dlech, 2017-01-24:
I am seeing this behavior on the EVB though.

laurensvalk, 2017-01-24:
> I am not able to reproduce this on an EV3. I wonder if there is some condition I am missing, like having 4 color sensors plugged in on RGB-RAW mode or something like that.

I don't know what the problematic conditions might be, but my setup was as follows:

Large motors on port B and D. Medium motor on A. EV3 Touch on 1, EV3 Color on 3, EV3 IR on 4. I didn't change any modes on boot so I suspect they were the default.

How fast are you rotating the motor? I attached a gear for grip (and dial for reference) and rotated it as fast as I could by hand.

dlech, 2017-01-24:
Last night, I was able to confirm that we are not missing IRQs, which is good (it means we can fix the driver without having to use the PRU). I haven't quite pinpointed the problem yet though, hopefully today.

dlech, 2017-01-24:
> How fast are you rotating the motor? I attached a gear for grip (and dial for reference) and rotated it as fast as I could by hand.

on the EVB, 800 deg/sec is enough to cause a problem, but on the EV3, apparently, I can't rotate it fast enough, even when turning it by hand.

dwalton76, 2017-01-24:
> on the EVB, 800 deg/sec is enough to cause a problem, but on the EV3, apparently, I can't rotate it fast enough, even when turning it by hand.

Maybe connect two motors where one is driving the other and gear it up to get the speed way up there and see if it breaks on EV3 then?

dlech, 2017-01-25:
> Last night, I was able to confirm that we are not missing IRQs, which is good (it means we can fix the driver without having to use the PRU).

I may have spoke to soon. I have eliminated the possibility that the "UNDO" code is causing the problem. So, this means that we have timing issues with the interrupt handler. Since the beaglebone uses a -rt kernel, it means that irq handlers can be interrupted and therefore it takes longer before they are actually run. so, what seems to be happening is that it is taking long enough sometimes that the direction pin has changed state between the interrupt firing and reading the state of the pin. So, the driver sees some interrupts as turning in the wrong direction. 

I suppose we could try turning of PREEMPT_RT in the kernel configuration, but really, the ideal solution here would be to implement a quadrature encoder in the PRU.

dlech, 2017-01-25:
Changing to PREEMPT__LL does indeed fix the problem.

dlech, 2017-01-26:
Fix has been release in [18-ev3dev kernel](http://www.ev3dev.org/news/2017/01/25/kernel-release-cycle-18/)

If this is still a problem for very high speeds, please open a new issue for implementing quadrature encoders in the PRU. Or, just open an issue for it anyway. :smile:

closed 2017-01-26 by dlech

dwalton76, 2017-01-26:
Awesome, thank you @dlech !! If this works this means EVB can be used for rubiks cube robots which would be a big win for me (no more offloading the CPU/memory intensive work to a pi3).  I'll give it a try.

WasabiFan, 2017-02-08:
Can this be tagged `EVB` to make it clear that it only affects that platform?

dlech, 2017-02-08:
The OP says that it happens on EV3 too.

WasabiFan, 2017-02-09:
I was taking my cues from the following things:

- You were unable to repro on EV3
- The issue was closed as a result of a fix applied to the EVB tree
- No fix was made for the EV3

If this _is_ a problem on the EV3, shouldn't there be an open issue for it? Or is there one, and I'm just missing it?

WasabiFan, 2017-02-09:
I ask this in part because @ndward emailed me asking about behavior that sounds a lot like this, but on the EV3. Specifically, he was referring to the behavior that he noticed in point 1 in https://github.com/rhempel/ev3dev-lang-python/issues/220. I'm going to test it myself, but first wanted to see if I could find information on it already.

dlech, 2017-02-09:
> If this is a problem on the EV3, shouldn't there be an open issue for it?

If someone says that this is still a problem on the EV3, we will re-open this issue.

> he was referring to the behavior that he noticed in point 1 in rhempel/ev3dev-lang-python#220

I suspect that it is not the same bug. This one seems to only happen at high speeds. At low speeds, we really need to measure motor speed differently. But, that problem would also be best solved by a PRU quadrature encoder.