ev3dev/ev3dev #796: RPI3 + brickPi problems with motors running at boot

MikeDabrowski, 2016-12-27:
### System information:

* I am using: 
  - RaspberryPi  3 Model B v1.2
  - BrickPi (old model but FW upgraded to 2.5)
  - NXT motors and sensors
  - EV3 gyro sensor 

* I installed ev3dev using this image file: *ev3dev-jessie-rpi2-generic-2016-12-21.img*

* My kernel version is: *4.4.32-17-ev3dev-rpi2*

* My host computer is running: *Windows10 pro*

* I am connecting using: *wifi/lan*

* My changes to config.txt (only enabled options):*
```
dtoverlay=brickpi
init_uart_clock=32000000
dtparam=brickpi_battery=okay
dtoverlay=pi3-disable-bt
dtparam=brickpi_led1_trigger=none,brickpi_led2_trigger=none
```
 
* Previous issue: *https://github.com/rhempel/ev3dev-lang-python/issues/279#issuecomment-269021862*

### About my issue:

Boot hangs on *[OK] Started Update UTMP about System Runlevel Changes...*
During boot motors start to run continuously. When boot hangs I can switch to another console and log in or ssh into rpi. 

I tried to run simple script:
```
import ev3dev.brickpi as ev3
m = ev3.LargeMotor('outB')
m.run_timed(time_sp=3000, speed_sp=500)
```
so I powered BrickPi on with battery pack (8x1.2V) and of course motor was already running but I ran `python3 motor.py` anyway. 
Exception : Device is not connected 

As for the changes to config.txt: 
- that line with leds is not working because my leds are burning blue 0.5 s from powering on
- that line with battery is for brickpi+, I rather don't have + model but I tried it anyway and still not working properly
- I dont need bluetooth

dlech, 2016-12-27:
> BrickPi (old model but FW upgraded to 2.5)

The motor controller wiring was changed between BrickPi (HW v1.7.3) and BrickPi+ (HW v2.8). As I understand it, the v2.5 firmware is for **BrickPi+ only** because of these wiring changes.

I strongly suspect that if you revert to the previous firmware, things will work correctly again.

> Boot hangs on [OK] Started Update UTMP about System Runlevel Changes...
> During boot motors start to run continuously. When boot hangs I can switch to another console and log in or ssh into rpi.

Boot does not "hang". I believe that Brickman is trying to run on `tty1` (the default console). It has the console tied up, but apparently does not draw anything on the screen. Anyway, using the HDMI port is not really supported by ev3dev (because I don't have an HDMI display and no one else has stepped up to make it work properly).

You could probably fix it by running `sudo systemctl mask brickman.service` to make sure that Brickman does not run.

MikeDabrowski, 2016-12-27:
Okay, lets say I revert to older FW, which version is supported by ev3dev AND will let me use gyro sensor? Gyro is crucial to my end project. I could probably lower my expectations and change project to something simpler. But it has to work.

MikeDabrowski, 2016-12-27:
During the upgrade process I got some problems and I consulted with JohnC from dexterindustries. At that point I didn't plan using ev3dev so either he didn't know it won't work or just didn't said that .

dlech, 2016-12-27:
The Raspberry Pi model shouldn't make a difference. I think firmware v2.0 will get you fixed up.

MikeDabrowski, 2016-12-27:
OKAY, I reverted to [this](https://github.com/DexterInd/BrickPi/tree/master/Firmware_BrickPi/Updating_the_Firmware/Firmware_Loading_Package_2013.07.05) 
EV3-Firmware_Check.py ran on dexter's raspbian says FW version is 511 (!) 
I can run motors there, gyro not so much.

Switched cards and booted ev3, motors don't run, boot is on UTMP again, when I ran motor.py it still says device not connected, the same traceback message...

dlech, 2016-12-27:
That looks too old. I think [this](https://github.com/DexterInd/BrickPi/tree/master/Firmware_BrickPi/Updating_the_Firmware/Firmware_2.0_Loading_Package) is the firmware that you want.

dlech, 2016-12-27:
> Switched cards and booted ev3, motors don't run, boot is on UTMP again

Did you try masking the service as I suggested? If that didn't work, try `sudo systemctl disable brickman.service`.

MikeDabrowski, 2016-12-27:
And what exactly `sudo systemctl mask brickman.service` and `sudo systemctl disable brickman.service` should do? The first one did some link but still booted up to UTMP, second removed some link but it get to the console ;D and motors are not running wild. 
I still got device not connected exception though.

WasabiFan, 2016-12-27:
> Can you confirm that the script I'm using (mentioned on top) is valid?

It isn't right for BrickPi. There is no port named `outB` on the BrickPi, because they use different names. Instead, you should do this:

```python
m = ev3.LargeMotor(ev3.OUTPUT_B)
```

You can see that I used a special variable that contains the correct port name. The other variables are as follows:

```python
OUTPUT_A = 'ttyAMA0:MA'
OUTPUT_B = 'ttyAMA0:MB'
OUTPUT_C = 'ttyAMA0:MC'
OUTPUT_D = 'ttyAMA0:MD'

INPUT_1 = 'ttyAMA0:S1'
INPUT_2 = 'ttyAMA0:S2'
INPUT_3 = 'ttyAMA0:S3'
INPUT_4 = 'ttyAMA0:S4'
```

MikeDabrowski, 2016-12-27:
@WasabiFan That was exactly what was causing the problem. Thank you very much. 
Could you provide me quick example how to test gyro?

WasabiFan, 2016-12-27:
@ndward has an example on his website [here](https://sites.google.com/site/ev3python/learn_ev3_python/using-sensors). You can look under the "Example 4 (gyro sensor and touch sensor)" heading. That demo includes code for a touch sensor as well, but you should be able to refer to the gyro code anyway.

MikeDabrowski, 2016-12-27:
I changed to ev3dev.brickpi and got this:
AssertionError: Connect a single gyro sensor to any sensor port
My gyro and touch sensors are connected to ports 2 and 1 respectively.

closed 2016-12-27 by MikeDabrowski

dlech, 2016-12-28:
> And what exactly sudo systemctl mask brickman.service and sudo systemctl disable brickman.service should do?

`brickman.service` is a systemd service. Masking it will prevent it from actually running even if something tells it to start. Disabling it will tell it to not start automatically. The file contains a line `Conflicts=getty@tty1.service` which prevents a login (`getty`) from running on the first console (`tty1`). I hoped that masking it would also cause that line to be ignored, but that didn't work, so disabling was the next logical choice.