ev3dev/ev3dev #797: brickpi problem with setting up sensors

MikeDabrowski, 2016-12-27:
### System information:

* I am using: 
  - RaspberryPi  3 Model B v1.2
  - BrickPi (old model but FW upgraded to 2.5)
  - NXT motors and sensors
  - EV3 gyro sensor 

* I installed ev3dev using this image file: *ev3dev-jessie-rpi2-generic-2016-12-21.img*

* My kernel version is: *4.4.32-17-ev3dev-rpi2*

* My host computer is running: *Windows10 pro*

* I am connecting using: *wifi/lan*

* My changes to config.txt (only enabled options):*
```
dtoverlay=brickpi
init_uart_clock=32000000
dtparam=brickpi_battery=okay
dtoverlay=pi3-disable-bt
dtparam=brickpi_led1_trigger=none,brickpi_led2_trigger=none
```
 
* Previous issue: *https://github.com/rhempel/ev3dev-lang-python/issues/279#issuecomment-269021862*

### About my issue:

I have troubles using touch and gyro sensors. So far I tried [this ](https://sites.google.com/site/ev3python/learn_ev3_python/using-sensors) where I changed to ev3dev.brickpi and got this error:
AssertionError: Connect a single gyro sensor to any sensor port

My gyro and touch sensors are connected to ports 2 and 1 respectively.

```
from ev3dev.brickpi import *
from time import sleep

# Connect gyro and touch sensors to any sensor port
# and check they are connected.
gy = GyroSensor('ttyAMA0:S2')
assert gy.connected, "Connect a single gyro sensor to any sensor port"
ts = TouchSensor('ttyAMA0:S1');    assert ts.connected, "Connect a touch sensor to any port"
```
Does not work either

The same situation with touch sensor:
```
import ev3dev.brickpi as ev3
ts = ev3.TouchSensor(ev3.INPUT_1)
m=ev3.LargeMotor(ev3.OUTPUT_B)

while not ts.value():
        m.run_timed(time_sp=100, speed_sp=500)
```
gives : 
```
Traceback (most recent call last):
  File "gyro.py", line 5, in <module>
    while not ts.value():
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 1576, in value
    self._value[n], value = self.get_attr_int(self._value[n], 'value'+str(n))
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 211, in get_attr_int
    attribute, value = self._get_attribute(attribute, name)
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 195, in _get_attribute
    raise Exception('Device is not connected')
Exception: Device is not connected
```
I have found several similar issues but none helped me get this right

WasabiFan, 2016-12-27:
The BrickPi can't auto-detect sensors or motors, so you need to configure them yourself, probably using the `LegoPort` Python class. This is already explained in multiple other issues, e.g. #529.

MikeDabrowski, 2016-12-28:
OK, so I tried to manually set touch sensor:
```
$ echo nxt-analog > /sys/class/lego-port/port0/mode
$ echo lego-nxt-touch > /sys/class/lego-port/port0/set_device
```
as [here](https://github.com/ev3dev/ev3dev/issues/374) but don't know if that is all needed. After those two commands I got sensor1.

Will it work in python as INPUT_1, or after setting up manually can I skip INPUT_1? 
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;*That I can answer after the test: setting `ts = ev3.TouchSensor(ev3.INPUT_1)` works.*
Is there a way I can do it from python? 
In lego-port directory i see ports from 0 to 7. Do they represent physical brickpi ports? S1-5, MA-C? One would be missing. 

I also tried `p = Device('lego-port', name_pattern='*', address='in4') ` but my python doesn't have Device: `NameError: name 'Device' is not defined` 

Where are main libraries located so I could get hints when coding in pycharm?
Where can I learn more about using ev3dev solution with brickpi? I've looked briefly at [this tutorial](https://sites.google.com/site/ev3python/learn_ev3_python/using-sensors) but it was written for ev3 so it wasn't mentioned that brickpi doesn't auto detect, and especially how to manually set sensors.

bmegli, 2016-12-28:
> In lego-port directory i see ports from 0 to 7. Do they represent physical brickpi ports? S1-5, MA-C? One would be missing.

BrickPi input port 5 (not present in BrickPi+) is wired to RaspberryPi I2C and it's controlled by RasbperryPi, not BrickPi. 

For this reason it only works with I2C/NXT and I2C/Other sensors. This is a hardware limitation (the way BrickPi was designed). 

See [brickpi-in-port-5](http://www.ev3dev.org/docs/ports/brickpi-in-port-5/) for more details.

This port (input 5) is currently the only way to get some 3rd party (non-lego) I2C sensors to work with BrickPi (sensors listed in #757).

dlech, 2016-12-28:
The `N` in `portN` and `sensorN` does not mean anything. Read the `name` attribute of ports and sensors to find out which one it is.

As for all of the python questions, it would be better to ask at <https://github.com/rhempel/ev3dev-lang-python>. And in general, you are likely to get better answers if you ask one question per GitHub issue.

dlech, 2016-12-28:
> Where can I learn more about using ev3dev solution with brickpi?

Unfortunately, not many people use BrickPi compared to the EV3, so there is not much BrickPi/ev3dev specific information floating around. However, you have already discovered the important differences (like no sensor auto-detection and names of ports are different). Other than those few things, ev3dev is pretty generic.

MikeDabrowski, 2016-12-28:
> BrickPi input port 5 (not present in BrickPi+) is wired to RaspberryPi I2C and it's controlled by RasbperryPi, not BrickPi.

I think I know what are you talking about. I got 5 physical sensor ports on BrickPi and one is on the other side of the board. On dexter's site they said that port is for their sensors and will not work with lego and other sensors.
Here it might be little different but I don't need that port. I will be happy with 2 touch sensors(NXT) or one gyro(EV3).

MikeDabrowski, 2016-12-28:
> And in general, you are likely to get better answers if you ask one question per GitHub issue.

Sorry I didn't want to be considered spammer. I think I'm generating quite a lot of 'issues' or 'problems' :) 
The reason for that seems to be because I got old, not very popular or well documented devices.

dlech, 2016-12-28:
It's not spam if they are legitimate questions. :wink: Having one question per issue also makes the answers easier for others to find, which is one step closer to having proper documentation.

MikeDabrowski, 2016-12-28:
> Having one question per issue also makes the answers easier for others to find, which is one step closer to having proper documentation.

When I succeed I can leave you guys some readme.md with basic tutorial of what I have learned which you could include somewhere here so the next ones would have easier start with brickpi.

closed 2016-12-28 by MikeDabrowski