ev3dev/ev3dev #798: Relocate website to root ev3dev.org domain

WasabiFan, 2016-12-28:
Currently, the 'site lives at `www.ev3dev.org`. This means that URLs like `ev3dev.org/docs` throw a 404 generated by @dlech's server which hosts `ev3dev.org`. Optimally, we would shift our configuration so that ev3dev.org is directly backed by GitHub Pages, bypassing @dlech's server completely. Unfortunately, we currently have a redirect from `ev3dev.org/archive` to the new location of the ev3dev archives, meaning that we can't put the 'site at the domain root. At some point we may want to give up on said redirect and move the main domain.

_This discussion was started in Gitter chat_

markyi370, 2017-05-16:
Do you mean there are two servers hosting the site?

dlech, 2017-05-16:
More than two now. :smiley: