ev3dev/ev3dev #800: HT SMUX support with EV3

MohammedBadra, 2017-01-01:


dlech, 2017-01-01:
Can you please provide more details about the issue?

We already have support for the [HiTechnic NXT sensor multiplexer](http://www.ev3dev.org/docs/sensors/hitechnic-nxt-sensor-multiplexer/)

MohammedBadra, 2017-01-02:
I'm just trying to work with the HT SMUX on the EV3 brick , which's said to be un-supported on the HT website , downloads 

So as you say the u support a block or drivers for the HT-Smux to work with the EV3 Brick ?? 
If available could u please send me links , install , user guide ? 

Thanks in advance

dlech, 2017-01-02:
It does not work with the official LEGO software. It does work with [ev3dev](http://www.ev3dev.org) though. I don't have any links better than the one I already gave. The notes at the end of the linked page provide some information on how to use it.

MohammedBadra, 2017-01-02:
Oh , excuse me 
I wasn't aware at the very beggining of the identity , use of ev3dev , i thought it was drivers website 
Not a separe os supporting these sensors 

Thanks you very much , i'm going to try the ev3dev with ev3 python , i hope it meets my hopes , requirments , in my project