ev3dev/ev3dev #801: Problem with tone command

jabrena, 2017-01-01:
### System information:

* I am using: LEGO MINDSTORMS EV3
* I installed ev3dev using this image file: *IMAGE NAME GOES HERE*

* My kernel version is: 4.4.24-16-ev3dev-ev3

* My host computer is running: *OSX*

* I am connecting using: Wi-fi

### About my issue:

When I execute this command, I receive Permission denied.
https://github.com/ev3dev/ev3dev/wiki/Using-Sound

robot@ev3dev:/sys/devices/platform/snd-legoev3$ cd /sys/devices/platform/snd-legoev3
robot@ev3dev:/sys/devices/platform/snd-legoev3$ echo 1000 10000 > tone
-bash: tone: Permission denied
robot@ev3dev:/sys/devices/platform/snd-legoev3$ uname -a
Linux ev3dev 4.4.24-16-ev3dev-ev3 #1 PREEMPT Sat Oct 15 11:14:54 CDT 2016 armv5tejl GNU/Linux
robot@ev3dev:/sys/devices/platform/snd-legoev3$ 

Many thanks in advance.

Juan Antonio

tim241, 2017-01-01:
Try ```sudo echo "1000 10000" > tone ```

dlech, 2017-01-01:
We have never really supported the attributes in `/sys/devices/platform/snd-legoev3/`. Use `beep` or the `EV_SND` event with `/dev/input/by-path/platform-snd-legoev3-event` instead.

dlech, 2017-01-01:
If if you really want to use it anyway, you need to do so as root.

    sudo sh -c 'echo 1000 10000 > tone'

jabrena, 2017-01-02:
Thanks @dlech, 

I fixed it.

Cheers

closed 2017-01-02 by jabrena