ev3dev/ev3dev #802: Add `pyserial` to the image

loopspace, 2017-01-05:
The presence of the USB port on the ev3 block makes it easy to add USB devices to communicate with the brick, such as the BBC micro:bit.  Being able to extend the ev3 brick is something that ev3dev can cope with that the standard Lego programming environment can't so could well be a selling point for ev3dev.  As well as the micro:bit, this could include microphones, cameras, and other USB sensors, in addition to providing reliable cabled communication between devices.

I'm therefore suggesting that `pySerial` be included in the ev3dev image so that it can be used "out of the box".

dlech, 2017-01-05:
FYI, the input ports also have a `uart-other` mode, so they can be used for generic serial communications as well (if you want to make your own cabled  and as long as it is 3.3V or 5V).

dlech, 2017-01-17:
The package `python3-serial` is installed by default starting with the 2017-01-17 snapshot.

https://oss.jfrog.org/oss-snapshot-local/org/ev3dev/brickstrap/2017-01-17/

closed 2017-01-17 by dlech