ev3dev/ev3dev #803: install opencv-3

chaturv, 2017-01-08:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3dev-jessie-ev3-generic-2016-10-17

* My kernel version is: 

* My host computer is running: Linux ubuntu 3.19.0-25-generic

* I am connecting using: USB

### About my issue:

Hello,
I have been trying to install opencv-3 for python-3. This is for using the Logitech C-270 webcam with my EV3. I am following a rather elaborate guide at http://www.pyimagesearch.com/2015/10/26/how-to-install-opencv-3-on-raspbian-jessie/

As a test, I was able to follow the guide and install opencv-3 on my host machine which is running Ubuntu. Everything works fine and I am was able to write a simple script to capture camera frames. 

However, when trying to install on my EV3, I hit a very interesting issue - the AA batteries die down before the install is complete! I tried using Anaconda to take care of the install (https://anaconda.org/menpo/opencv3) but had issues installing Anaconda in the first place.

I was wondering if anyone has been able to get opencv-3 installed. Thanks!

moriarty, 2017-01-08:
I haven't installed OpenCV on my EV3, but I have installed OpenCV on my BrickPi, using an old method no longer supported.

Compiling OpenCV requires a lot of ram, and the EV3 doesn't have that, so you're likely running out and thats why the **make** command takes forever. You can try increasing the swap, but this will also take a very long time. (for example, I tried compiling the basic ROS dependencies on the EV3 and that took 24 hours with increased swap.)

I think you'll need to cross compile opencv for the EV3.
This is the current method is: http://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/

I haven't followed the docker instructions, I haven't been actively working with my EV3 in a while and followed the instructions for the older brickstrap method.

dwalton76, 2017-01-08:
I have installed it before but I have the rechargable battery and I just left it running over night. @dlech could we add opencv to the binary install?

dlech, 2017-01-08:
Is opencv 2.4 that already exists in official debian not good enough? It looks like the reason opencv 3.x is not in official debian yet is because it breaks some other packages.

dwalton76, 2017-01-08:
For me opencv 2.4 would be fine

pambrose, 2017-01-08:
3.2 was released a short time ago and it has been solid for me thus far.

One nice thing is it is now installable on OSX via brew.

Another compelling reason to go with 3.2 is a lot of people learn OpenCV from 
http://www.pyimagesearch.com <http://www.pyimagesearch.com/> and Adrian leans toward 3.x in his tutorials and book.

Paul

WasabiFan, 2017-01-08:
I use OpenCV 2.4 in most of my work because that is what's supported on the embedded platform I work with. I haven't found any important features yet which were available in 3.x but not in 2.4; the structure of some of their APIs has changed, but it doesn't seem like there is anything significantly lacking. I have also found that most online resources target 2.4.

pambrose, 2017-01-08:
I should clarify my last statement. I am using 3.2 on a Raspi and feeding the data 
to the EV3 via MQTT and gRPC.

wliu2016, 2017-03-17:
@dlech 
First, sorry to reply an existing issue opened two month ago, but I have googled a lot, tried different methods, and end up with failure to install Opencv in EV3. 

Though I am OK with opencv 2.4 in official debian, the official opencv will be automatically installed in Python 2.  I am mostly working in Python 3 and can not import opencv in Python 3.

Is there a method to install opencv 2.4 in EV3, which works in Python 3?  

Thanks and have a good one.

Wenlong Liu

dlech, 2017-03-17:
Python3 support was only recently added to opencv, so I doubt that it is possible to get it working with 2.4.

http://stackoverflow.com/questions/20953273/install-opencv-for-python-3-3

dlech, 2017-05-02:
There has been some work in debian recently on opencv 3.1. <https://packages.debian.org/source/experimental/opencv>.

If someone is willing to backport this to stretch and be the ev3dev package maintainer for this, we are certainly open to shipping opencv 3.x with ev3dev-stretch.

bidaian, 2017-10-17:
The package is in sid and right now it's installable as-is in stretch! I just added

`deb http://ftp.br.debian.org/debian/ sid main contrib non-free` to /etc/apt/sources.list

and did

`apt-get update`
`apt-get install python3-opencv`

dlech, 2017-10-17:
Did it pull in any other dependencies?

bidaian, 2017-10-19:
Yes. This is the apt-get ouput after adding sid to snapshot-ev3dev-stretch-ev3-generic-2017-09-30.img.xz.
```
# apt-get install python3-opencv
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libbluetooth3 libboost-chrono1.62.0 libboost-python1.62.0
  libboost-system1.62.0 libboost-thread1.62.0 libdbus-glib-1-2 libwebpmux2
  pixy-rules python3-dbus
Use 'apt autoremove' to remove them.
The following additional packages will be installed:
  fontconfig-config gcc-7-base gdal-data libaacs0 libaec0 libarmadillo7
  libarpack2 libatomic1 libavcodec57 libavformat57 libavresample3 libavutil55
  libbdplus0 libblas3 libbluray2 libcharls1 libchromaprint1 libdap25
  libdapclient6v5 libdapserver7v5 libdc1394-22 libdrm-amdgpu1 libdrm-common
  libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libdrm2 libepsilon1
  libexif12 libfontconfig1 libfreexl1 libfyba0 libgd3 libgdal20 libgdcm2.8
  libgeos-3.5.1 libgeos-c1v5 libgeotiff2 libgfortran4 libgif7 libgl1
  libgl1-mesa-dri libgl1-mesa-glx libgl2ps1.4 libglapi-mesa libglu1-mesa
  libglvnd0 libglx-mesa0 libglx0 libgme0 libgphoto2-6 libgphoto2-l10n
  libgphoto2-port12 libgsm1 libhdf4-0-alt libhdf5-100 libhdf5-openmpi-100
  libhwloc-plugins libhwloc5 libibverbs1 libilmbase12 libjson-c3 libjsoncpp1
  libkmlbase1 libkmlconvenience1 libkmldom1 libkmlengine1 libkmlregionator1
  libkmlxsd1 liblapack3 liblept5 libllvm5.0 libltdl7 libmariadbclient18
  libminizip1 libmp3lame0 libmpg123-0 libnetcdf-c++4 libnetcdf11
  libnl-route-3-200 libnss3 libodbc1 libogdi3.2 libopencv-calib3d3.2
  libopencv-contrib3.2 libopencv-core3.2 libopencv-features2d3.2
  libopencv-flann3.2 libopencv-highgui3.2 libopencv-imgcodecs3.2
  libopencv-imgproc3.2 libopencv-ml3.2 libopencv-objdetect3.2
  libopencv-photo3.2 libopencv-shape3.2 libopencv-stitching3.2
  libopencv-superres3.2 libopencv-video3.2 libopencv-videoio3.2
  libopencv-videostab3.2 libopencv-viz3.2 libopenexr22 libopenjp2-7
  libopenmpi2 libopenmpt0 libpciaccess0 libpoppler68 libpq5 libproj12
  libpython3-stdlib libpython3.6-minimal libpython3.6-stdlib libqhull7
  libraw1394-11 libsensors4 libshine3 libsnappy1v5 libsocket++1 libsoxr0
  libspatialite7 libspeex1 libssh-gcrypt-4 libstdc++6 libsuperlu5
  libswresample2 libswscale4 libsz2 libtbb2 libtcl8.6 libtesseract-data
  libtesseract3 libtheora0 libtk8.6 libtwolame0 libtxc-dxtn-s2tc liburiparser1
  libva-drm1 libva-x11-1 libva1 libvdpau-va-gl1 libvdpau1 libvorbisfile3
  libvpx4 libvtk6.3 libwavpack1 libwebpmux3 libx264-148 libx265-130
  libxcb-glx0 libxerces-c3.1 libxft2 libxpm4 libxss1 libxt6 libxvidcore4
  libxxf86vm1 libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers
  mysql-common ocl-icd-libopencl1 odbcinst odbcinst1debian2 openmpi-bin
  openmpi-common poppler-data proj-bin proj-data python3 python3-dbus
  python3-gi python3-minimal python3-numpy python3-olefile python3-pil
  python3.6 python3.6-minimal va-driver-all vdpau-driver-all
Suggested packages:
  libbluray-bdj libgd-tools geotiff-bin gdal-bin libgeotiff-epsg gphoto2
  libhdf4-doc libhdf4-alt-dev hdf4-tools libhwloc-contrib-plugins libmyodbc
  odbc-postgresql tdsodbc unixodbc-bin ogdi-bin pciutils libraw1394-doc
  lm-sensors speex tcl8.6 tk8.6 mpi-default-bin vtk6-doc vtk6-examples
  opencl-icd gfortran poppler-utils ghostscript fonts-japanese-mincho
  | fonts-ipafont-mincho fonts-japanese-gothic | fonts-ipafont-gothic
  fonts-arphic-ukai fonts-arphic-uming fonts-nanum python3-doc python3-tk
  python3-venv python-dbus-doc python3-dbus-dbg gcc python-numpy-doc
  python3-dev python3-nose python3-numpy-dbg python-pil-doc python3-pil-dbg
  python3.6-venv python3.6-doc binutils binfmt-support
The following packages will be REMOVED:
  openrobertalab python3-bluez python3-evdev python3-gattlib
  python3-libpixyusb python3-smbus
The following NEW packages will be installed:
  gcc-7-base gdal-data libaacs0 libaec0 libarmadillo7 libarpack2 libatomic1
  libavcodec57 libavformat57 libavresample3 libavutil55 libbdplus0 libblas3
  libbluray2 libcharls1 libchromaprint1 libdap25 libdapclient6v5
  libdapserver7v5 libdc1394-22 libdrm-amdgpu1 libdrm-common libdrm-freedreno1
  libdrm-nouveau2 libdrm-radeon1 libepsilon1 libexif12 libfreexl1 libfyba0
  libgd3 libgdal20 libgdcm2.8 libgeos-3.5.1 libgeos-c1v5 libgeotiff2
  libgfortran4 libgif7 libgl1 libgl1-mesa-dri libgl1-mesa-glx libgl2ps1.4
  libglapi-mesa libglu1-mesa libglvnd0 libglx-mesa0 libglx0 libgme0
[... 50 lines cut ...]
  libswscale4 libtheora0 libtwolame0 libtxc-dxtn-s2tc libv4l-0 libv4lconvert0 libva-drm1 libva-x11-1 libva1
  libvdpau-va-gl1 libvdpau1 libvorbisfile3 libvpx4 libwavpack1 libx264-148 libx265-95 libxcb-glx0
  libxvidcore4 libxxf86vm1 libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers python-numpy
  va-driver-all vdpau-driver-all
Suggested packages:
  libbluray-bdj gvfs libraw1394-doc lm-sensors speex gcc gfortran python-dev python-nose python-numpy-dbg
  python-numpy-doc
The following NEW packages will be installed:
  libaacs0 libavcodec57 libavformat57 libavutil55 libbdplus0 libblas-common libblas3 libbluray1
  libchromaprint1 libdc1394-22 libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libgail-common libgail18
  libgfortran3 libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libgme0 libgsm1 libgtk2.0-0 libgtk2.0-bin
  libgtk2.0-common libilmbase12 liblapack3 libmp3lame0 libmpg123-0 libopencv-calib3d2.4v5
[... cut ...]
  libswscale4 libtheora0 libtwolame0 libtxc-dxtn-s2tc libv4l-0 libv4lconvert0 libva-drm1 libva-x11-1 libva1
  libvdpau-va-gl1 libvdpau1 libvorbisfile3 libvpx4 libwavpack1 libx264-148 libx265-95 libxcb-glx0
  libxvidcore4 libxxf86vm1 libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers python-numpy
  python-opencv va-driver-all vdpau-driver-all
0 upgraded, 78 newly installed, 0 to remove and 44 not upgraded.
Need to get 28.3 MB of archives.
After this operation, 178 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
```

dlech, 2018-04-19:
OpenCV 3.2, including Python 3 support is now available in ev3dev-stretch.

    sudo apt update
    sudo apt install python3-opencv

closed 2018-04-19 by dlech

speedzshaw, 2018-06-18:
I use sudo apt install python3-opencv. Everything goes well until I reach this step 
"37 upgraded, 289 newly installed, 6 to remove and 336 not upgraded.
Need to get 0 B/175 MB of archives.
After this operation, 724 MB of additional disk space will be used.
Do you want to continue? [Y/n] ". 

Every time I type in Y and enter. After a while, the brick is lost connection. Is there anything that's particular preventing from installing?

Thanks.

bidaian, 2018-06-18:
Could you paste the output after you type "Y" @speedzshaw? If there is no output, does "sudo apt update" work alright?

dlech, 2018-06-18:
@speedzshaw, also best to open a new issue since this issue is closed and explain the exact steps you took. e.g. Are you using ev3dev-jessie or ev3dev-stretch (you should be using stretch)? Did you use the workaround described in this issue (it is no longer needed)?