ev3dev/ev3dev #804: ssh connection via putty not working

pepere, 2017-01-09:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3dev-jessie-ev3-generic-2016-12-21.img

* My kernel version is: unknown...

* My host computer is running: Windows 7

* I am connecting using: Bluetooth

### About my issue:

I am trying to connect to my ev3 brick using Putty. After specifying 'ev3dev' as a destination, I am logging in as 'robot'. But the default password ('maker') is not working. The window returns 'Access denied".

Any help appreciated.

Thank you very much,

Alexis

WasabiFan, 2017-01-09:
Nothing seems obviously wrong to me; can you provide a screenshot of the PuTTY window as well as the settings screen beforehand? I want to make sure I'm imagining the right situation. @dlech am I missing something?

My automatic suggestion would be to try `ev3dev.local` as the hostname, but if you're getting an "access denied" error, you have already gotten past that point, so it wouldn't be a networking issue.

pepere, 2017-01-09:
Hey,

Thanks a lot for your answer!

Putting ".local" as a suffix doesn't right. You're right, it doesn't seem
to be a networking issue since I can reach the login window.
On the ev3 brick, I can see the connection's parameters (DHCP, IP address
and mask).
The gateway is empty though, not sure if it's expected or not.

Screenshots:
Putty:
[image: Inline image 1]

Then terminal opened by putty, after typing user name and password:
[image: Inline image 2]

Not sure why Putty is adding this extension (calumetlubricants.com, one of
my VPN connections, disabled here) on the window pane.
Might be the issue.

Thanks again!

Alexis

WasabiFan, 2017-01-09:
It looks like GitHub doesn't handle images via email well... Can you upload them via their web interface?

pepere, 2017-01-09:
Screenshots re-uploaded as PDFs..

[Putty.pdf](https://github.com/ev3dev/ev3dev/files/692916/Putty.pdf)
[Terminal.pdf](https://github.com/ev3dev/ev3dev/files/692917/Terminal.pdf)

Wouldn't it be just a matter of credentials? The robot/maker credentials would apply to an older image file than the one I flashed (ev3dev-jessie-ev3-generic-2016-12-21.img).

Thanks,

mob41, 2017-01-09:
Try flashing again. If still can't login, download a new image. I believe the data is corrupted.

dlech, 2017-01-09:
> Wouldn't it be just a matter of credentials?

No. If you enter the wrong username or password, you get a different error message.

Try using the IP address from the EV3 instead of the hostname. If that gives the same error message, then I think a bad SD card is the most likely culprit.

dlech, 2017-01-09:
Another possibility: http://serverfault.com/questions/317781/putty-access-denied

markyi370, 2017-01-28:
If it still doesn't work, I have a solution that solved the problem for me.

Remove the SD card from the EV3.

Put it into an ARM-based computer running some decent flavor of Linux (preferably Debian). **Yes, it must be ARM.** If you don't have one, I'm not sure what you can do (You might have to figure out how to use `qemu-arm` or something like that). Here's an example of ARM computers: Beaglebone/Beaglebone Black, Pandaboard (basically any of those dev boards)

Open a terminal on that ARM computer (through UART, SSH, whatever), and execute the following commands, **as root**:
```
mount /dev/`<whatever the sd registered as>` /mnt
chroot /mnt
passwd robot
```

Set the `robot` account's new password, then try to connect again.

dlech, 2017-01-28:
Given the fact that everyone downloads and uses the same image, if you have to chroot into the SD card to reset the default password, there is something wrong with your SD card (or possibly the SD card adapter used to flash it). I would not trust that SD card and would get a new one. If it corrupted the the password, there is no telling what else it corrupted.

Also, if you are using a 32GB card, try a smaller one. It seems that compatibility of 32GB cards is hit and miss.