ev3dev/ev3dev #805: brick does not boot after installing packages

JoaoMarcosVensiBasso, 2017-01-10:
Hi,

I am using EV3.
I installed the 2016 December version of ev3dev.
I can't see the kernel version because I can't boot the EV3.
I was trying to connect Wifi.

I have done this at least ten times, and I am not able to get the brick to be "reliable". Which means, after I boot it for the first time, I ssh in and execute apt-get update, upgrade, and install python3-flask. After that, I get different errors every time that don't let me do anything else. Sometimes I get an exception when trying to connect wifi; sometimes I get errors like this:

sudo: unable to mkdir /var/lib/sudo/lectured: Structure needs cleaning
sudo: unable to mkdir /var/lib/sudo/ts: Read-only file system

or sometimes when booting:

/etc/init exists but couldn't execute it (error -117)

What can I do?

Thank you very much.

WasabiFan, 2017-01-10:
It sounds to me like you have a corrupted SD card. How did you flash it? If it was with Etcher, it should have warned you.

JoaoMarcosVensiBasso, 2017-01-10:
I tried again and after fsck'ing I rebooted and got "unitializaed randim read - 46 bits of entropy available".

I used Etcher and did not get any warnings. I don't get why it is a different error every time.

dlech, 2017-01-10:
What are the specs on the SD card?

JoaoMarcosVensiBasso, 2017-01-10:
32Gb, SanDisk, microSDHC UHS-I.

mob41, 2017-01-10:
>All cards larger than 32GB will not work with the EV3!

Use a SD Card smaller than 32GB. Probably does not work

dlech, 2017-01-10:
Hmm... should be OK. Do you have a different SD card you can try? I have heard of some cases where 32GB cards don't work with SDHC even though they are supposed to. The error messages sound like what happens if you try to use a card > 32GB.

JoaoMarcosVensiBasso, 2017-01-11:
Ok, I am going to try and post the results. Thanks.

JoaoMarcosVensiBasso, 2017-01-12:
It was exactly that. I used a 8Gb card and it worked. Thank you!

dlech, 2017-01-12:
Glad you got it working.

closed 2017-01-12 by dlech