ev3dev/ev3dev #807: Brickpi // Problem detecting motors

jabrena, 2017-01-10:
### System information:

* I am using: *BrickPi+ (HW v2.8)*
* I installed ev3dev using this image file: *IMAGE NAME GOES HERE*
* My kernel version is: *4.4.32-17-ev3dev-rpi2*
* My host computer is running: *OS X*
* I am connecting using: *Wifi*

### About my issue:
I am trying to connect a motor on a BrickPi board, but on /sys/class, I can´t detect. Reading dmesg, I receive this messages.

[    6.853000] Registered BrickPi line discipline. (28)
[    7.868064] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    7.871665] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.528508] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    9.440439] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   16.853709] tty ttyAMA0: Failed to init while setting sensors. (-110)

Any clue?

Juan Antonio

jabrena, 2017-01-10:
Related issue:
https://github.com/ev3dev/ev3dev/issues/585

dlech, 2017-01-10:
What does this have to do with motors? How is this different from #585? I have never been able to reproduce this issue, nor have I seen anyone else report the same problem.

jabrena, 2017-01-10:
Hi @dlech, It is the same problem that previously. Tomorrow, I will test with a Raspi 3.

dlech, 2017-01-11:
OK, so we should re-open #585.

closed 2017-01-11 by dlech