ev3dev/ev3dev #808: beagleboneblack HiTechnic compass

DanAragona, 2017-01-11:
### System information:

Fatcatlab EVB

* I installed ev3dev using this image file: ev3dev-jessie-bone-generic-2016-12-21.img

* My kernel version is: 4.4.31-ti-rt-r67-17-ev3dev-bb.org

Linux

Ethernet wired

### About my issue:

I am having an issue with i2c devices - according to the notes i2c has been implemented on Beaglebone Black for this release?

I am using C++ and have successfully got motors and nxt/analog sensors to work - documentation is good - '''/sys/class/lego-port/port1/device/of_node''' i find my port # by name attribute and then do a set mode nxt-analog and then set-device <device-name>

for I2c: 
I plug in my ht-nxt-compass - port 3  - 
I see under directory '''/sys/class/lego-port/port3 - evb-ports:in3:nxt-i2c-host''' (good so far and dmesg output is perfect) 
then when I look at '''/sys/bus/i2c/drivers/nxt-i2c-sensor''' all I see is bind module uevent  unbind - nothing else

I have read thru the i2c documentation and I tried manually loading the device manually with 
'''echo ht-nxt-compass 0x01 > /sys/bus/i2c/devices/i2c-5/new_device''' - all I get is permission denied.

so how to i get ht-nxt-compass to register?

Thanks for your time

here is my dmesg output:

'''
[   13.610784] evb_pru_i2c rpmsg0: Registered i2c adapter evb-pru-i2c0
[   13.618702] lego-port port1: Registered 'evb-ports:in1' on 'evb-ports:in1'.
[   13.618722] evb-input-port evb-ports:in1: Port does not support 9V on pin 1
[   13.618729] evb-input-port evb-ports:in1: Port cannot discriminate EV3/NXT sensors
[   13.623868] lego-port port2: Registered 'evb-ports:in2' on 'evb-ports:in2'.
[   13.623886] evb-input-port evb-ports:in2: Port does not support 9V on pin 1
[   13.623893] evb-input-port evb-ports:in2: Port cannot discriminate EV3/NXT sensors
[   13.634296] evb_pru_i2c rpmsg1: Registered i2c adapter evb-pru-i2c1
[   13.661275] lego-port port3: Registered 'evb-ports:in3' on 'evb-ports:in3'.
[   13.661299] evb-input-port evb-ports:in3: Port does not support 9V on pin 1
[   13.661305] evb-input-port evb-ports:in3: Port cannot discriminate EV3/NXT sensors
[   13.669640] evb_pru_i2c rpmsg2: Registered i2c adapter evb-pru-i2c2
[   13.670869] lego-port port4: Registered 'evb-ports:in4' on 'evb-ports:in4'.
[   13.670886] evb-input-port evb-ports:in4: Port does not support 9V on pin 1
[   13.670893] evb-input-port evb-ports:in4: Port cannot discriminate EV3/NXT sensors
[   13.694132] evb_pru_i2c rpmsg3: Registered i2c adapter evb-pru-i2c3
[   14.864252] lego-port port5: Registered 'evb-ports:outB' on 'evb-ports:outB'.
[   14.945981] lego-port port6: Registered 'evb-ports:outC' on 'evb-ports:outC'.
[   14.973795] lego-port port7: Registered 'evb-ports:outD' on 'evb-ports:outD'.
[   15.040972] random: nonblocking pool is initialized
[   15.053580] lego-port port3: Added new device 'evb-ports:in3:nxt-i2c-host'
[   15.239618] lego-port port0: Added new device 'evb-ports:outA:lego-ev3-l-motor'
[   15.381711] lego-port port5: Added new device 'evb-ports:outB:lego-ev3-l-motor'
[   15.405386] lego-port port6: Added new device 'evb-ports:outC:lego-ev3-l-motor'
[   15.963212] tacho-motor motor0: Registered 'lego-ev3-l-motor' on 'evb-ports:outA'.
[   16.007608] tacho-motor motor1: Registered 'lego-ev3-l-motor' on 'evb-ports:outB'.
[   16.044341] tacho-motor motor2: Registered 'lego-ev3-l-motor' on 'evb-ports:outC'.
[   18.186174] net eth0: initializing cpsw version 1.12 (0)
[   18.217851] net eth0: initialized cpsw ale version 1.4
[   18.247250] net eth0: ALE Table size 1024
[   18.280310] net eth0: phy found : id is : 0x7c0f1
[   18.347402] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   20.529886] Bluetooth: Core ver 2.21
[   20.536160] NET: Registered protocol family 31
[   20.543597] Bluetooth: HCI device and connection manager initialized
[   20.551705] Bluetooth: HCI socket layer initialized
[   20.557321] Bluetooth: L2CAP socket layer initialized
[   20.564223] Bluetooth: SCO socket layer initialized
[   20.624199] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   20.630873] Bluetooth: BNEP filters: protocol multicast
[   20.637353] Bluetooth: BNEP socket layer initialized
[   21.297654] cpsw 4a100000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   21.306959] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   21.402820] 8021q: 802.1Q VLAN Support v1.8
[   21.407388] 8021q: adding VLAN 0 to HW filter on device eth0
[ 4274.524275] lego-port port2: Added new device 'evb-ports:in2:nxt-analog'
[ 4274.544416] lego-port port2: Removed device 'evb-ports:in2:nxt-analog'
[ 4274.564481] lego-port port2: Added new device 'evb-ports:in2:lego-nxt-light'
[ 4274.606267] lego-port port4: Added new device 'evb-ports:in4:nxt-analog'
[ 4274.623199] lego-port port4: Removed device 'evb-ports:in4:nxt-analog'
[ 4274.646420] lego-port port4: Added new device 'evb-ports:in4:lego-nxt-sound'
[ 4274.683306] lego-port port1: Added new device 'evb-ports:in1:nxt-analog'
[ 4274.700520] lego-port port1: Removed device 'evb-ports:in1:nxt-analog'
[ 4274.725579] lego-sensor sensor1: Registered 'lego-nxt-touch' on 'evb-ports:in1'.
[ 4274.730608] lego-sensor sensor0: Registered 'lego-nxt-light' on 'evb-ports:in2'.
[ 4274.739215] lego-sensor sensor2: Registered 'lego-nxt-sound' on 'evb-ports:in4'.
[ 4274.775927] lego-port port1: Added new device 'evb-ports:in1:lego-nxt-touch'
[ 4360.459317] lego-port port2: Removed device 'evb-ports:in2:lego-nxt-light'
[ 4360.466505] lego-sensor sensor0: Unregistered 'lego-nxt-light' on 'evb-ports:in2'.
[ 4360.490966] lego-sensor sensor3: Registered 'nxt-analog' on 'evb-ports:in2'.
[ 4360.511704] lego-port port2: Added new device 'evb-ports:in2:nxt-analog'
[ 4360.531741] lego-port port2: Removed device 'evb-ports:in2:nxt-analog'
[ 4360.546276] lego-sensor sensor3: Unregistered 'nxt-analog' on 'evb-ports:in2'.
[ 4360.571535] lego-sensor sensor4: Registered 'lego-nxt-light' on 'evb-ports:in2'.
[ 4360.590227] lego-port port2: Added new device 'evb-ports:in2:lego-nxt-light'
[ 4360.631861] lego-port port4: Removed device 'evb-ports:in4:lego-nxt-sound'
[ 4360.646666] lego-sensor sensor2: Unregistered 'lego-nxt-sound' on 'evb-ports:in4'.
[ 4360.674455] lego-sensor sensor5: Registered 'nxt-analog' on 'evb-ports:in4'.
[ 4360.691606] lego-port port4: Added new device 'evb-ports:in4:nxt-analog'
[ 4360.712179] lego-port port4: Removed device 'evb-ports:in4:nxt-analog'
[ 4360.727874] lego-sensor sensor5: Unregistered 'nxt-analog' on 'evb-ports:in4'.
[ 4360.752314] lego-sensor sensor6: Registered 'lego-nxt-sound' on 'evb-ports:in4'.
[ 4360.774529] lego-port port4: Added new device 'evb-ports:in4:lego-nxt-sound'
[ 4360.813197] lego-port port1: Removed device 'evb-ports:in1:lego-nxt-touch'
[ 4360.827682] lego-sensor sensor1: Unregistered 'lego-nxt-touch' on 'evb-ports:in1'.
[ 4360.859382] lego-sensor sensor7: Registered 'nxt-analog' on 'evb-ports:in1'.
[ 4360.875814] lego-port port1: Added new device 'evb-ports:in1:nxt-analog'
[ 4360.892664] lego-port port1: Removed device 'evb-ports:in1:nxt-analog'
[ 4360.913329] lego-sensor sensor7: Unregistered 'nxt-analog' on 'evb-ports:in1'.
[ 4360.937223] lego-sensor sensor8: Registered 'lego-nxt-touch' on 'evb-ports:in1'.
[ 4360.956016] lego-port port1: Added new device 'evb-ports:in1:lego-nxt-touch'
[ 7612.214234] lego-port port3: Removed device 'evb-ports:in3:nxt-i2c-host'
[ 7613.771433] lego-port port3: Added new device 'evb-ports:in3:nxt-i2c-host'
[ 7622.514511] lego-port port3: Removed device 'evb-ports:in3:nxt-i2c-host'
[ 7624.059639] lego-port port3: Added new device 'evb-ports:in3:nxt-i2c-host'
'''

dlech, 2017-01-11:
I don't have a HT Compass sensor to test with, but I have just tried with a HT Angle sensor, which is also an I2C sensor and it is working as expected.

With the sensor plugged into input port 4, please run this and post the output.

    sudo i2cdump -f -y -r 0x00-0x17 6 0x01

If it gives a bunch of `X`s, try this:

    sudo i2cdetect -y 6

P.S. I fixed up your post to use backticks (``` ` ```) for proper formatting. You were using a single quote (`'`) which doesn't do anything special.

dlech, 2017-01-11:
Oh, and make sure the port is in `nxt-i2c` mode. You can see this in the device browser in Brickman or

    cat /sys/class/lego-port/port*/status

DanAragona, 2017-01-11:
So this is what I got from i2ctools: sudo i2cdump -f -y -r 0x00-0x17 6 0x01

0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: fd 56 31 2e 32 33 20 20 48 69 54 65 63 68 6e 63    ?V1.23  HiTechnc
10: 43 6f 6d 70 61 73 73 20                            Compass

also I do get nxt-i2c when I cat the status.

When I check the directory: /sys/bus/i2c/drivers/nxt-i2c-sensor i still
only see: bind module uevent unbind

dlech, 2017-01-12:
It seems that the vendor id is different between hardware/firmware revisions. V2.1 uses `HITECHNC` (see <https://github.com/ev3dev/ev3dev/issues/268#issuecomment-74100854>).

I guess we will have to have duplicate entries for this sensor or make checking the vendor case in-sensitive (if there is a function for that in the kernel).

It also looks like we need to ad a calibrate command for this sensor so that you can calibrate it if needed.

Can you please run `sudo i2cdump -f -y -r 0x41-0x45 6 0x01` also? In the unlikely chance 0x42 is zero, rotate the sensor and run it again.

dlech, 2017-01-12:
Also, `sudo sh -c 'echo ht-nxt-compass 0x01 > /sys/bus/i2c/devices/i2c-6/new_device'`should get you going for the time being.

DanAragona, 2017-01-12:
Here is what I got for  sudo i2cdump -f -y -r 0x41-0x45 6 0x01:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
40:    00 79 01 f3 00                                   .y??.

fyi: i successfully got the compass installed and working with the command
u gave me. Thanks very much

dlech, 2017-01-12:
Excellent, that is what I was hoping to see. We should be able to get automatic detection fixed up now.

dlech, 2017-01-26:
Fix has been release in [18-ev3dev kernel](http://www.ev3dev.org/news/2017/01/25/kernel-release-cycle-18/)

closed 2017-01-26 by dlech