ev3dev/ev3dev #810: How to detect Battery with BrickPi & Raspberry Pi 1 Model B

jabrena, 2017-01-12:
### What did you do?
I am trying to read battery on a Raspberry Pi Model B with Brick Pi but I don´t see the path:

``` bash
robot@ev3dev:/$ cd /sys/class/
robot@ev3dev:/sys/class$ ls
backlight        dc-motor     i2c-adapter       iscsi_host       lego-sensor   pwm          scsi_host    thermal  vc-sm
bcm2708_vcio     devcoredump  i2c-dev           iscsi_iface      mem           raw          servo-motor  tty      vchiq
bcm2835-gpiomem  dma          ieee80211         iscsi_session    misc          rfkill       sound        uio      vtconsole
bdi              gpio         input             iscsi_transport  mmc_host      rtc          spi_master   vc       watchdog
block            graphics     iscsi_connection  leds             net           scsi_device  spidev       vc-cma
bsg              hidraw       iscsi_endpoint    lego-port        power_supply  scsi_disk    tacho-motor  vc-mem
robot@ev3dev:/sys/class$ cd power_supply/
robot@ev3dev:/sys/class/power_supply$ ls
robot@ev3dev:/sys/class/power_supply$ 
```

### What did you expect to happen?

### What actually happended?

### What hardware and software are you using (including verion numbers)?

``` bash
robot@ev3dev:/sys/class/power_supply$ uname -r
4.4.32-17-ev3dev-rpi
``` 

Many thanks in advance

Juan Antonio

dlech, 2017-01-12:
Only BrickPi+ can monitor the battery. Older BrickPis lack the hardware required to do this. Also, for BrickPi+, you have to [enable it in config.txt](https://github.com/ev3dev/docker-library/blob/master/ev3dev-jessie/layers/rpi-common/boot/flash/config.txt#L24)

closed 2017-01-14 by dlech

jabrena, 2017-01-14:
Hi @dlech, It was a knowledge doubt. My board is a BrickPi+ v2.8 so enabled the configuration on config.txt was enough. I fixed :)

Cheers