ev3dev/ev3dev #811: How to detect a Raspberry Pi board by command line?

jabrena, 2017-01-12:
### What did you do?
I would like to know how to detect that ev3dev is running on a raspberry pi board.

Does exist some way to detect it?

One alternative could be the usage of uname-r and parsing later.

``` bash
robot@ev3dev:/sys/class$ uname -r | grep rpi
robot@ev3dev:/sys/class$ uname -r | grep rpi2
robot@ev3dev:/sys/class$ uname -r | grep rpi3
````

Another idea could be use /proc/cpuinfo

``` bash
robot@ev3dev:/sys/class$ cat /proc/cpuinfo | grep BCM2708
Hardware	: BCM2708
``` 

### What did you expect to happen?

### What actually happended?

### What hardware and software are you using (including version numbers)?

Related issue:
https://github.com/ev3dev/ev3dev/issues/523

dlech, 2017-01-12:
#523 explains the available options. `/proc/device-tree/model` is probably currently the best option.

jabrena, 2017-01-12:
For RaspberryPi 1, how to detect BrickPi? Previously, I used the battery path to detect BrickPi

dlech, 2017-01-13:
Did you read #412? It has some suggestions.

jabrena, 2017-01-13:
Tomorrow, I will read in detail again.

Cheers

dlech, 2017-09-10:
In ev3dev-stretch, there is now #523

closed 2017-09-10 by dlech

jabrena, 2017-09-10:
I have to move to `ev3dev-stretch` in next months.

Many thanks @dlech 

Cheers