ev3dev/ev3dev #812: Brake command doesn´t work on BrickPi with RaspberryPi Model B

jabrena, 2017-01-12:
Using latest image for Raspberry Pi 1 and BrickPi:

robot@ev3dev:/sys/class/tacho-motor/motor4$ uname -r
4.4.32-17-ev3dev-rpi

I have found that it is not possible to set brake to stop_action:

``` bash
ev3dev#27|Exception in thread "main" java.lang.RuntimeException: Operation not executed: /sys/class/tacho-motor/motor14/stop_action with value brake
ev3dev#27|	at ev3dev.hardware.EV3DevDevice.setStringAttribute(EV3DevDevice.java:123)
ev3dev#27|	at ev3dev.actuators.motors.BaseRegulatedMotor.brake(BaseRegulatedMotor.java:139)
ev3dev#27|	at examples.raspberryPi1.brickPi.hardware.actuators.motors.LargeMotorDemo.main(LargeMotorDemo.java:15)
Failed command ev3dev#27 with status 1: java -jar /home/robot/ev3dev-lang-java-all-0.4.0-SNAPSHOT.jar
```

In EV3, the same example run without any problem.

Juan Antonio

dlech, 2017-01-12:
That is correct. It is not possible to use `brake` on BrickPi. It is a limitation of the BrickPi firmware.

It does not seem that we have this documented anywhere that is easily accessible. Should add it in a few places, like http://www.ev3dev.org/docs/platform-comparison/.

jabrena, 2017-01-12:
It is a pity, brake is critic to use Motors to build robots using features like "Inverse Kinematics" where motor accuracy is required.

WasabiFan, 2017-01-12:
You can still use the "hold" mode to achieve largely the same thing; "hold" is _more_ precise than "brake".

jabrena, 2017-01-12:
Many thanks @WasabiFan, I will test tomorrow with that mode.

bmegli, 2017-01-13:
@jabrena often you can achieve similiar effect mechanically using [worm gears](http://sariel.pl/2009/06/worm-gear-casings/)

They can be driven from one side and block mechanically from the other so can be used in lifts, cranes, for [holding the camera](https://www.youtube.com/watch?v=dZfVPe_U0yA&t=1m15s) but are useless in some cases (like for robot wheels). 

Worm gears will also not consume power when keeping things in place so may be an option to consider vs BrickPi motor hold mode in some applications.

dlech, 2017-10-03:
This is already documented:

> Not all actions may be supported. Read ``stop_actions`` to get the
> actions available for a particular driver.

closed 2017-10-03 by dlech