ev3dev/ev3dev #813: Detect touch sensor and Use on Raspberry Pi Model B

jabrena, 2017-01-12:
I am testing EV3 Touch Sensor but one, I set the port as uart and with the right device, I can´t see the sensor on the path: /sys/class/lego-sensor

```
robot@ev3dev:~$ ls /sys/class/lego-sensor
robot@ev3dev:~$ echo lego-ev3-touch > /sys/class/lego-port/port0/set_device
robot@ev3dev:~$ echo ev3-uart > /sys/class/lego-port/port0/mode
robot@ev3dev:~$ cd /sys/class/lego-sensor/
```

What is missing?

Juan Antonio

dlech, 2017-01-13:
The EV3 touch sensor is analog, not UART.

Try `echo ev3-analog > /sys/class/lego-port/port0/mode`. You should not need `echo lego-ev3-touch > /sys/class/lego-port/port0/set_device` because the touch sensor is the only known and supported EV3/Analog sensor on BrickPi.

jabrena, 2017-01-14:
Hi,

I have tested but the path always return 1 in the value 0 although the sensor is not pressed:

Logs:

``` 
java -jar /home/robot/ev3dev-lang-java-all-0.4.0-SNAPSHOT.jar
ev3dev#4|2017-01-14 15:52:36 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting motors with port: ttyAMA0:S1
ev3dev#4|2017-01-14 15:52:36 [main] DEBUG ev3dev.utils.Sysfs - ls /sys/class/lego-port
ev3dev#4|2017-01-14 15:52:36 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port0/address
ev3dev#4|2017-01-14 15:52:36 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port0/address
ev3dev#4|2017-01-14 15:52:36 [main] INFO  ev3dev.hardware.EV3DevSensorDevice - detected lego port: /sys/class/lego-port/port0
ev3dev#4|2017-01-14 15:52:36 [main] DEBUG ev3dev.utils.Sysfs - echo ev3-analog > /sys/class/lego-port/port0/mode
ev3dev#4|2017-01-14 15:52:36 [main] DEBUG lejos.utility.Delay - Delay: 1000
ev3dev#4|2017-01-14 15:52:37 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting motors with port: ttyAMA0:S1
ev3dev#4|2017-01-14 15:52:37 [main] DEBUG ev3dev.utils.Sysfs - ls /sys/class/lego-sensor
ev3dev#4|2017-01-14 15:52:37 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor2/address
ev3dev#4|2017-01-14 15:52:37 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-sensor/sensor2/address
ev3dev#4|2017-01-14 15:52:37 [main] INFO  ev3dev.hardware.EV3DevSensorDevice - detected lego sensor: /sys/class/lego-sensor/sensor2
ev3dev#4|2017-01-14 15:52:37 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor2/value0
ev3dev#4|2017-01-14 15:52:37 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Iteration: 0
ev3dev#4|2017-01-14 15:52:37 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Touch: 1
ev3dev#4|2017-01-14 15:52:37 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#4|2017-01-14 15:52:38 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor2/value0
ev3dev#4|2017-01-14 15:52:38 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Iteration: 1
ev3dev#4|2017-01-14 15:52:38 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Touch: 1
ev3dev#4|2017-01-14 15:52:38 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#4|2017-01-14 15:52:38 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor2/value0
ev3dev#4|2017-01-14 15:52:38 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Iteration: 2
ev3dev#4|2017-01-14 15:52:38 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Touch: 1
ev3dev#4|2017-01-14 15:52:38 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#4|2017-01-14 15:52:39 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor2/value0
ev3dev#4|2017-01-14 15:52:39 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Iteration: 3
ev3dev#4|2017-01-14 15:52:39 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Touch: 1
ev3dev#4|2017-01-14 15:52:39 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#4|2017-01-14 15:52:39 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor2/value0
ev3dev#4|2017-01-14 15:52:39 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Iteration: 4
ev3dev#4|2017-01-14 15:52:39 [main] INFO  e.r.b.sensors.ev3.TouchSensorDemo - Touch: 1
ev3dev#4|2017-01-14 15:52:39 [main] DEBUG lejos.utility.Delay - Delay: 500
```

jabrena, 2017-01-14:
But in EV3, it runs:

``` bash
java -jar /home/robot/ev3dev-lang-java-all-0.4.0-SNAPSHOT.jar
ev3dev#6|2017-01-14 16:02:43 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting motors with port: in1
ev3dev#6|2017-01-14 16:02:43 [main] DEBUG ev3dev.utils.Sysfs - ls /sys/class/lego-sensor
ev3dev#6|2017-01-14 16:02:43 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor0/address
ev3dev#6|2017-01-14 16:02:44 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-sensor/sensor0/address
ev3dev#6|2017-01-14 16:02:44 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor0/value0
ev3dev#6|2017-01-14 16:02:44 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Iteration: 0
ev3dev#6|2017-01-14 16:02:44 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Touch: 0
ev3dev#6|2017-01-14 16:02:44 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#6|2017-01-14 16:02:44 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor0/value0
ev3dev#6|2017-01-14 16:02:44 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Iteration: 1
ev3dev#6|2017-01-14 16:02:44 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Touch: 0
ev3dev#6|2017-01-14 16:02:44 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#6|2017-01-14 16:02:45 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor0/value0
ev3dev#6|2017-01-14 16:02:45 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Iteration: 2
ev3dev#6|2017-01-14 16:02:45 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Touch: 0
ev3dev#6|2017-01-14 16:02:45 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#6|2017-01-14 16:02:45 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor0/value0
ev3dev#6|2017-01-14 16:02:45 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Iteration: 3
ev3dev#6|2017-01-14 16:02:45 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Touch: 0
ev3dev#6|2017-01-14 16:02:45 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#6|2017-01-14 16:02:46 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor0/value0
ev3dev#6|2017-01-14 16:02:46 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Iteration: 4
ev3dev#6|2017-01-14 16:02:46 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Touch: 0
ev3dev#6|2017-01-14 16:02:46 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#6|2017-01-14 16:02:47 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor0/value0
ev3dev#6|2017-01-14 16:02:47 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Iteration: 5
ev3dev#6|2017-01-14 16:02:47 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Touch: 0
ev3dev#6|2017-01-14 16:02:47 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#6|2017-01-14 16:02:47 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor0/value0
ev3dev#6|2017-01-14 16:02:47 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Iteration: 6
ev3dev#6|2017-01-14 16:02:47 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Touch: 1
ev3dev#6|2017-01-14 16:02:47 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#6|2017-01-14 16:02:48 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor0/value0
ev3dev#6|2017-01-14 16:02:48 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Iteration: 7
ev3dev#6|2017-01-14 16:02:48 [main] INFO  e.h.sensors.ev3.TouchSensorDemo - Touch: 1
ev3dev#6|2017-01-14 16:02:48 [main] DEBUG lejos.utility.Delay - Delay: 500
```

dlech, 2017-01-14:
Can you please post a bash script that can be used to reproduce the problem? The output of your java program is not entirely clear.

jabrena, 2017-01-14:
Hi,

I send the status in lego-port and lego-sensor. I have the same behaviour for Raspberry Pi 1 & 3

``` bash
robot@ev3dev:~$ ls /sys/class/lego-port
port0  port1  port2  port3  port4  port5  port6  port7
robot@ev3dev:~$ cd port0
-bash: cd: port0: No such file or directory
robot@ev3dev:~$ cd /sys/class/lego-port/
robot@ev3dev:/sys/class/lego-port$ ls
port0  port1  port2  port3  port4  port5  port6  port7
robot@ev3dev:/sys/class/lego-port$ cd port0/
robot@ev3dev:/sys/class/lego-port/port0$ ls
address  device  driver_name  mode  modes  power  set_device  status  subsystem  ttyAMA0:S1:lego-ev3-touch  uevent
robot@ev3dev:/sys/class/lego-port/port0$ cat modes 
none nxt-analog nxt-color nxt-i2c ev3-analog ev3-uart
robot@ev3dev:/sys/class/lego-port/port0$ cat mode  
ev3-analog
robot@ev3dev:/sys/class/lego-port/port0$ cat address 
ttyAMA0:S1
robot@ev3dev:/sys/class/lego-port/port0$ cd ..
robot@ev3dev:/sys/class/lego-port$ cd ..
robot@ev3dev:/sys/class$ cd lego-sensor/
robot@ev3dev:/sys/class/lego-sensor$ ls
sensor3
robot@ev3dev:/sys/class/lego-sensor$ cd sensor3
robot@ev3dev:/sys/class/lego-sensor/sensor3$ ls
address          command   device       fw_version  num_values  subsystem   units   value2  value5
bin_data         commands  direct       mode        poll_ms     text_value  value0  value3  value6
bin_data_format  decimals  driver_name  modes       power       uevent      value1  value4  value7
robot@ev3dev:/sys/class/lego-sensor/sensor3$ cat address 
ttyAMA0:S1
robot@ev3dev:/sys/class/lego-sensor/sensor3$ cat driver_name 
lego-ev3-touch
robot@ev3dev:/sys/class/lego-sensor/sensor3$ cat value0 
1
robot@ev3dev:/sys/class/lego-sensor/sensor3$ 
```

dlech, 2017-01-25:
I cannot reproduce this. Here is what I did:

```
robot@ev3dev:~$ cd /sys/class/lego-port/
robot@ev3dev:/sys/class/lego-port$ ls
port0  port1  port2  port3  port4  port5  port6  port7
robot@ev3dev:/sys/class/lego-port$ cat */address
ttyAMA0:S1
ttyAMA0:S2
ttyAMA0:MA
ttyAMA0:MB
ttyAMA0:S3
ttyAMA0:S4
ttyAMA0:MC
ttyAMA0:MD
robot@ev3dev:/sys/class/lego-port$ cd port0
robot@ev3dev:/sys/class/lego-port/port0$ cat modes
none nxt-analog nxt-color nxt-i2c ev3-analog ev3-uart
robot@ev3dev:/sys/class/lego-port/port0$ echo ev3-analog > mode
robot@ev3dev:/sys/class/lego-port/port0$ cd /sys/class/lego-sensor/sensor0/
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat address 
ttyAMA0:S1
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat driver_name 
lego-ev3-touch
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat value
cat: value: No such file or directory
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat value0
0
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat value0
1
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat value0
0
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat value0
1
robot@ev3dev:/sys/class/lego-sensor/sensor0$ 
```

Perhaps you had a bad connection or were using an incorrect port?

jabrena, 2017-01-25:
I will re test again this week mate.

dlech, 2017-10-03:
Is there anything else to do on this issue?

closed 2018-04-15 by dlech

jabrena, 2018-04-15:
Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/316