ev3dev/ev3dev #816: Sound Pop on start of play back

absentdream, 2017-01-14:
### What did you do?
using python for speech_recognition and play back 
`import speech_recognition as sr
def main():
	r = sr.Recognizer()
	with sr.Microphone() as source:
		print("Say something!")
		audio = r.listen(source) <<---- this is where it pops`

just before it starts it pops (assume initalising the sound card

i have read that seting ramp_ms to a higher value might help but this causes it to error

/etc/modprobe.d/ev3.conf
options snd_legoev3 ramp_ms=100

any ideas?

Image file:         ev3dev-jessie-ev3-generic-2016-10-17
Kernel version:     4.4.32-17-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2017-01-14:
The EV3 does not have a sound card. It uses a PWM signal and a low pass filter to modulate the speaker. The pop happens as a result of the sound driver making a sudden large change in the signal we are sending to the speaker.

As far as I can tell, this is what the ramping does:

[image]

I think this is wrong and what is causing the pop. I imagine that the error was a result of assuming that 0 was the proper "no sound" state, but we are using unsigned values for PCM data, so 50% duty cycle on the PWM is actually "no sound" and 0 is "maxed out". So the pop is either caused by this "maxed out" bit or the fact that the rate of change of the sound signal changes sharply when transitioning to or from the ramp. If you know calculus, then you will know that that the rate of change at the point between two straight lines with different slope is infinite (or approaches infinity).

I've cleaned up the [sound driver on the FatcatLab EVB](https://github.com/ev3dev/lego-linux-drivers/blob/master/evb/evb_sound.c) to remove the ramping altogether. However, I had to remove the FIQ bits that make the sound quality much better on the EV3 because the BeagleBone does not allow the use of FIQ. 

So, that driver needs to be fixed up to have a pluggable back end where we can use an hrtimer, FIQ or PRU. Then we can use it on the EV3 and see if it fixes the popping sound problem.

absentdream, 2017-01-14:
ok.. so thats really low down in the system. i will run ev3dev_config and update it .. lets hope we can get rid of it 
Many thanks

dlech, 2017-01-18:
I've managed to improve things a bit but I still don't have it quite right. The work I have been doing is with a 4.9.y kernel, so it will be included in the planned ev3dev-stretch release. But the current ev3dev-jessie release is running a 4.4.y kernel and I am not planning on backporting the changes.

The changes can be found [here](https://github.com/ev3dev/ev3-kernel/commits/v4.9.4-ev3dev-bpo-stretch-r4-ev3-lms2012_1) and [here](https://github.com/ev3dev/lego-linux-drivers/commits/v4.9.4-ev3dev-bpo-stretch-r4-ev3-lms2012_1).

dlech, 2017-08-12:
I think I have this fixed in the 4.9.x kernel (ev3dev-stretch). There is still a pop the first time a sound plays, but after that, I am not hearing a pop.

closed 2017-08-12 by dlech