ev3dev/ev3dev #817: What is the recommend voltage to shutdown a robot using BrickPi+?

jabrena, 2017-01-14:
Hi,

If a robot the reach this voltage: 7019100.0, what is the recommendation to shutdown the Raspberry Pi + Brickpi + EV3Dev to replace the batteries?

``` bash
java -jar /home/robot/ev3dev-lang-java-all-0.4.0-SNAPSHOT.jar
ev3dev#16|2017-01-14 16:15:32 [main] TRACE ev3dev.hardware.EV3DevDevice - Detected platform: BRICKPI
ev3dev#16|2017-01-14 16:15:32 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/power_supply/brickpi-battery/voltage_now
ev3dev#16|2017-01-14 16:15:32 [main] INFO  e.r.brickPi.sensors.BatteryDemo - Battery Voltage: 7019100.0
```

dlech, 2017-01-14:
Duplicate of #580.

Please search for existing issues before starting new ones.

closed 2017-01-14 by dlech

WasabiFan, 2017-01-14:
@jabrena opened the dupe too :laughing: