ev3dev/ev3dev #818: Possible problem with Motors on PiStorms V1 + Raspberry Pi 3 + Latest image

jabrena, 2017-01-15:
### What did you do?

I tested a large motor with PiStorms v1 mounted on Raspberry Pi 3 using latest image and the EV3Dev doesn´t have any problem, but the motor doesn work.

### What did you expect to happen?

```
robot@ev3dev:/sys/class/tacho-motor/motor8$ echo reset > /sys/class/tacho-motor/motor9/command
robot@ev3dev:/sys/class/tacho-motor/motor8$ echo 500 > /sys/class/tacho-motor/motor9/speed_sp
robot@ev3dev:/sys/class/tacho-motor/motor8$ echo hold > /sys/class/tacho-motor/motor9/stop_action
robot@ev3dev:/sys/class/tacho-motor/motor8$ echo normal > /sys/class/tacho-motor/motor9/polarity
robot@ev3dev:/sys/class/tacho-motor/motor8$ echo 500 > /sys/class/tacho-motor/motor9/speed_sp
robot@ev3dev:/sys/class/tacho-motor/motor8$ echo run-forever > /sys/class/tacho-motor/motor9/command
robot@ev3dev:/sys/class/tacho-motor/motor8$ cat /sys/class/tacho-motor/motor9/state
running
robot@ev3dev:/sys/class/tacho-motor/motor8$ cat /sys/class/tacho-motor/motor9/speed_sp
500
robot@ev3dev:/sys/class/tacho-motor/motor8$ echo stop > /sys/class/tacho-motor/motor9/command
```
 
### What actually happened?

If you observe traces, EV3Dev says that the motor is running but the motor doesn`t move.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-rpi2-generic-2016-12-21
Kernel version:     4.4.32-17-ev3dev-rpi2
Board:              Raspberry Pi 3 Model B Rev 1.2
Revision:           a02082
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-01-15:
> Besides, it is not possible with a Regulated motor to swith to an unregulated mode.

I don't think that PiStorms has an unregulated mode. The user manual says that it does (i.e. it says that you can use RCX motors), but I have not figured out how to make it work. If someone would like to play around with it and try to figure it out, be my guest. I would suggest opening an issue specifically for this feature if you want this to be looked into.

> The method rotate or rotateTo never finish

Can you please post a bash script that can be used to reproduce the problem? It is too much work for me to try to figure out what your java program is doing.

jabrena, 2017-01-16:
Yes, tomorrow, I will send all data

jabrena, 2017-01-16:
Hi @dlech, this is the summary from logs:

```
echo tacho-motor > /sys/class/lego-port/port2/mode
echo reset > /sys/class/tacho-motor/motor10/command
echo 100 > /sys/class/tacho-motor/motor10/speed_sp
echo hold > /sys/class/tacho-motor/motor10/stop_action
cat /sys/class/tacho-motor/motor10/position
echo 90 > /sys/class/tacho-motor/motor10/position_sp
echo run-to-abs-pos > /sys/class/tacho-motor/motor10/command
cat /sys/class/tacho-motor/motor10/state
```

I observed the same behaviour with Pi 1 & 3

This week, I will test PiStorms V2.

Cheers.

Juan Antonio

dlech, 2017-01-25:
I performed the following and the motor worked as expected. The motor rotated 90 degrees and then stopped (well, it chatters a bit, but that is a mindsensors.com firmware bug - see #696).

```
root@ev3dev-rpi3:~# cd /sys/class/tacho-motor/motor2
root@ev3dev-rpi3:/sys/class/tacho-motor/motor2# cat address
pistorms:BAM1
root@ev3dev-rpi3:/sys/class/tacho-motor/motor2# echo 400 > speed_sp
root@ev3dev-rpi3:/sys/class/tacho-motor/motor2# echo hold > stop_action
root@ev3dev-rpi3:/sys/class/tacho-motor/motor2# cat position
-39
root@ev3dev-rpi3:/sys/class/tacho-motor/motor2# echo 90 > position_sp 
root@ev3dev-rpi3:/sys/class/tacho-motor/motor2# echo run-to-abs-pos > command
root@ev3dev-rpi3:/sys/class/tacho-motor/motor2# cat state
running
root@ev3dev-rpi3:/sys/class/tacho-motor/motor2#
```

There is no need to set the mode of output the port. It is already tacho-motor by default.

jabrena, 2017-01-25:
Mmmm, so for Raspi 1, it is necessary to detect the version of Raspberry Pi to not execute the operation. Anyway to see the firmware installed on PiStorms V1 & V2?

Maybe if I have some Java Hasmap with all version could enough to detect the generation:

http://elinux.org/RPi_HardwareHistory#Raspberry_Pi_2_Model_B

```
/proc/cpuinfo | grep Revision
``` 

I will test this week.

In the test with all boards, I detected other problems. I will create a summary along this week.

Many Thanks @dlech 

Cheers

Juan Antonio

dlech, 2017-07-28:
Is this still a problem?

jabrena, 2017-07-28:
Hi @dlech, 

Many months ago, I only put the focus on Raspi3, but if you like, tomorrow, I will execute a test iteration with raspi 1. With Raspi3 and PiStorms V2, I have received some sensors issues. 

Take a look the following slides about my conclusions with previous image:

```
robot@ev3dev:~$ uname -r
4.4.68-20-ev3dev-ev3

robot@ev3dev:~$ uname -r
4.4.68-20-ev3dev-rpi2
```

http://ev3dev-lang-java.github.io/#/3/1
http://ev3dev-lang-java.github.io/#/3/2
http://ev3dev-lang-java.github.io/#/3/3

This weekend, I will inform about the status @dlech 

Juan Antonio

jabrena, 2017-07-28:
**Note:** I continue using `Jessie` if you need to test `Stretch`, I could help in the testing process, but at the moment I only use Jessie. :)

closed 2017-10-03 by dlech