ev3dev/ev3dev #819: Question about PID loop in Lego motors

nicesai, 2017-01-15:
Could someone please explain how the PID loop works in the lego motor driver. In particular ... 

1. Since there are PID parameters, its clear that there is an internal PID loop. Is this loop in the driver SW? or motor firmware (if there is one).

2. When is this PID loop used? I see the following values for the command param, which ones uses the internal pid loop and which ones don't? Are the following statements correct?
        run-forever:  no-pid
        run-to-abs-pos: pid with position as feedback
        run-to-rel-pos: pid with position as feedback
        run-timed: no-pid
        run-direct: no-pid

3. When internal PID is used, how is the speed setting works? is it just the slew rate of the position setpoints that are generated internally?

4. Could someone please point to the SW which implements these pid loops? I am curious and want to learn more.

Thank you in advance.

dlech, 2017-01-15:
> Since there are PID parameters, its clear that there is an internal PID loop. Is this loop in the driver SW? or motor firmware (if there is one).

On the EV3 (and FatcatLab EVB and BrickPi), there is a PID in the Linux kernel driver. On PiStorms, the PID is in firmware on the device.

> When is this PID loop used?

There are two cases, when we are regulating speed during run commands (except `run-direct`) and when we are regulating position during the stop command when the stop action is `hold`.

> I see the following values for the command param, which ones uses the internal pid loop and which ones don't?

All run commands use a PID to regulate speed - except `run-direct`.

> Are the following statements correct?

Only the last one.

> When internal PID is used, how is the speed setting works? is it just the slew rate of the position setpoints that are generated internally?

I'm not sure I understand these questions. Could you clarify what you are asking?

> Could someone please point to the SW which implements these pid loops?

There is a [generic PID algorithm in motors/tacho_motor_helper.c](https://github.com/ev3dev/lego-linux-drivers/blob/master/motors/tacho_motor_helper.c#L90). It is called [here](https://github.com/ev3dev/lego-linux-drivers/blob/6ad31886d442530ebd4ed812fb451ee0f19b1731/ev3/legoev3_motor.c#L740) for speed regulation and [here](https://github.com/ev3dev/lego-linux-drivers/blob/6ad31886d442530ebd4ed812fb451ee0f19b1731/ev3/legoev3_motor.c#L743) for position regulation. The PID constants depend on the motor type and are found in [motors/ev3_motor_defs.c](https://github.com/ev3dev/lego-linux-drivers/blob/master/motors/ev3_motor_defs.c).

nicesai, 2017-02-05:
Hi dlech, thank you for the prompt answers. Sorry couldn't reply quickly as I was busy with new member in our family :-) 
  
>> I'm not sure I understand these questions. Could you clarify what you are asking?

I am trying to understand the architecture of the SW. Specifically the internal PID implementation. I am myself a control systems engineer, so I am a bit more curious about the implementation details. Thanks for those links, they helped. I have few more questions but I want to read the existing docs on the website first before bothering you further.

dlech, 2017-02-08:
I look forward to your feedback (when you have time between the :baby_bottle: and such).

closed 2020-08-17 by dlech