ev3dev/ev3dev #820: Brickman: Back key calls shutdown window

JorgePe, 2017-01-17:
Snapshot image, 17Jan2017
snapshot-ev3dev-jessie-ev3-generic-2017-01-16.img.xz  

No updates.

Each time I go to a sub-menu (like 'Wireless and Networks' or 'Battery') and then return by pressing the 'Back' key I get a "Shutodown..." window.
If press "back" again i immediately returns to the "Shutdown...".

At first I thought I had the key stuck or something like that but no.
If I restart Brickman from ssh

`sudo service brickman restart`

everything works fine again until I press "back" again.

dlech, 2017-01-17:
This bug was introduced by <https://github.com/ev3dev/ev3devKit/commit/6ddc4a4c0431f38e8866bed255c2fbd34be3b3f7>. Feel free to double-check me, but as far as I can tell, this fix was the right thing to do and we probably have a problem in the board setup file in the EV3 kernel.

dlech, 2017-01-17:
This looks like an easy fix for any aspiring kernel hackers out there. In the [EV3 board configuration file](https://github.com/ev3dev/ev3-kernel/blob/ev3dev-jessie/arch/arm/mach-davinci/board-legoev3.c#L226), we have the buttons defined as "active low" (see [struct gpio_keys_button definition](https://github.com/ev3dev/ev3-kernel/blob/5bae80fb82b77919712e16970dd4d04725e69eaa/include/linux/gpio_keys.h#L8). But according to the LEGO schematics, these keys are "active high".

[image]

Essentially, we just need to change the `1`s to `0`s in the first link that I gave (board configuration file) and we should be fixed up.

dlech, 2017-01-17:
This can be confirmed using `sudo evtest` on an EV3. Pressing a key should result in `value 1` and releasing a key should result in `value 0`.

```
Event: time 1484675280.764426, type 1 (EV_KEY), code 14 (KEY_BACKSPACE), value 1
Event: time 1484675280.764426, -------------- EV_SYN ------------
Event: time 1484675282.194419, type 1 (EV_KEY), code 14 (KEY_BACKSPACE), value 0
Event: time 1484675282.194419, -------------- EV_SYN ------------
```

JorgePe, 2017-01-17:
Yes, it's inverted, I get "0" when pressing and "1" when releasing:
```
Event: time 1484686004.156402, type 1 (EV_KEY), code 28 (KEY_ENTER), value 0
Event: time 1484686004.156402, -------------- EV_SYN ------------
Event: time 1484686004.376406, type 1 (EV_KEY), code 28 (KEY_ENTER), value 1
Event: time 1484686004.376406, -------------- EV_SYN ------------
Event: time 1484686030.236387, type 1 (EV_KEY), code 14 (KEY_BACKSPACE), value 0
Event: time 1484686030.236387, -------------- EV_SYN ------------
Event: time 1484686030.426413, type 1 (EV_KEY), code 14 (KEY_BACKSPACE), value 1
Event: time 1484686030.426413, -------------- EV_SYN ------------
```

bmegli, 2017-01-25:
As a side effect of looking into other issue I have built [4f05fe2](https://github.com/ev3dev/ev3-kernel/commit/4f05fe25f9448fa9dab85bb0785d42bb798480c8) kernel.

This was what I got (so correct events):

``` bash
Event: time 1484675280.764426, type 1 (EV_KEY), code 14 (KEY_BACKSPACE), value 1
Event: time 1484675280.764426, -------------- EV_SYN ------------
Event: time 1484675282.194419, type 1 (EV_KEY), code 14 (KEY_BACKSPACE), value 0
Event: time 1484675282.194419, -------------- EV_SYN ------------
```

but the problem in Brickman remained. Now I am confused 😕.

dlech, 2017-01-25:
It depends on what version ev3devKit is. ` ev3dev-sysinfo -k`.

dlech, 2017-01-26:
Fix has been release in [18-ev3dev kernel](http://www.ev3dev.org/news/2017/01/25/kernel-release-cycle-18/)

closed 2017-01-26 by dlech