ev3dev/ev3dev #821: ev3dev at 2017 Embedded Linux Conference in Portland, OR

dlech, 2017-01-18:
### What did you do?

I made plans to attend [ELC 2017](http://events.linuxfoundation.org/events/embedded-linux-conference) (my wife has always wanted to visit Portland) and I applied for a spot in the [technical showcase](http://elinux.org/ELC_2017_Technical_Showcase).

### What did you expect to happen?

For the application to be accepted.

### What actually happened?

The application was accepted.

---

So, it is official. Ev3dev will be making an appearance at a major Linux conference. If anyone is going to be in the Portland, OR area on the evening of Feb. 22, 2017 and would like to talk to people about ev3dev, please get in touch with me. Also, if anyone has any suggestions about what are the most interesting things about ev3dev that I should share or what sort of robots I should bring, feel free to leave a comment here.

dwalton76, 2017-01-19:
Very cool, congrats!!  To me BALANC3R and MINDCUB3R would be cool robots to demo.

WasabiFan, 2017-01-19:
I like it how you formatted this post as a problem report :laughing: I'd say @dwalton76 has the right idea. Showing off something that depends on Linux-specific features (that wouldn't work on the normal OS) would be cool -- maybe one of the live web IDEs or something like that.

I won't be in Portland (one state off :wink:) but I'd be happy to help out with any planning or poster design stuff if I have some free time.

bmegli, 2017-01-19:
Awesome!

> if anyone has any suggestions about what are the most interesting things about ev3dev

I guess you know all that... ;-)

## Ideas for live presentation

### Live Control (from shell)

It's super cool to just plug/unplug motors/sensors and control them live from shell knowing that it is generic mechanism (not tailored for particular robot). 

Interested people could even ssh themselves and control the hardware. 

This also exposes the sysfs interface.

### Live Control (from python)

Through RPyC?  

On one live presentation when I was writing/using R bindings I was doing something similiar from RServe executing code live/gathering statistics/plotting distributions while talking. It wouldn't be possible without ev3dev (without lots of work, and here it was for free).

### Distributed Control 

Like crowd control of hardware.

I once had a presentation on [ZooKeeper](https://zookeeper.apache.org/) and I paired it with ev3dev for fun. There was distributed work queue on the robot. The robot just took 3 commands:
- R### (rotate by +- degrees)
- G### (go by +- cm)
- any other **string** (speak this **string**)

Anybody who joined the ZooKeeper cloud could control the robot through ZooKeeper CLI. Really funny, especially when people want the robot to go to different places.

### Speech Control

This is rather for RPI + ev3dev + microphone. Just robot which executes spoken commands (limited subset for simplicity). 

## Ideas for poster

### ev3dev sysfs device tree

Probably obvious idea.

-------------------------------------------------------

I have no idea how @dlech finds time for everything he does 😄

bmegli, 2017-01-19:
Huh, there is general pattern: tool + ev3dev = cool  😄

- ssh + ev3dev (cool control from sheel)
- R + ev3dev (cool live statistics with live control)
- tts + ev3dev (cool talking robots)
- stt + ev3dev (cool robots executing spoken commends)
- zookeeper + ev3dev (cool distributed robots)
- gstreamer + ev3dev (cool video/audio streaming from robot)

Part of it is just the power of linux. Ev3dev unique is sysfs + platform abstraction that makes it programming language and platform indpendent all mixed with linux utilities and simplicity/availability of prototyping robots from Lego.

JorgePe, 2017-01-19:
One interesting time (well, at least for me) is the symbiosis between linux, microcomputer and mechanical framework. Arduino and other embedded systems (Linux or not) are great but usually lack "real world" interfacing,  the EV3 allows us to use world wide available standard parts to build almost everything we can think of without caring for bolts, nuts, drilling, soldering...

Perhaps you can show some kind of collaborative robots or a robot that uses "common" linux libraries from the scientific/academic community (like OpenCV or neural networks).

Best luck!

dwalton76, 2017-01-19:
"Showing off something that depends on Linux-specific features (that wouldn't work on the normal OS)" I am going to shamelessly plug some of my own robots here...since I know in detail how they can't be done on the normal lego OS :)

This one uses a web UI for driving the robot, the web server is written in python:
https://www.youtube.com/watch?v=x5VauXr7W4A

This one uses OpenCV (and a lot of python and various other bits of rubiks cube solving software from the web) to solve various size rubiks cubes.  It can solve 2x2x2, 3x3x3 and 4x4x4 but I plan on getting it to work for 5x5x5, 6x6x6 and 7x7x7
https://www.youtube.com/watch?v=7_drnwXl1Yk&t=25s

BertLindeman, 2017-01-19:
Great "issue" this ;-) 💐 

Please do not forget the demo-effect using an EV3-color-sensor on ev3dev see #792

rhempel, 2017-01-19:
Portland is awesome - this is great news David and I am looking forward to hearing about how it goes. Having EV3 support in the mainline kernel is also a fantastic accomplishment. Let us know when the livestream of the talk is ready😀

dlech, 2017-01-20:
I won't actually be giving a talk, but I will have a table for a few hours for demonstrations. One thing I hope to be showing off is an EV3 booting from a mainline v4.11-rc kernel with no extra patches. This should happen as long as the v4.10 release is done by then. The device tree just got [picked up today](https://git.kernel.org/cgit/linux/kernel/git/nsekhar/linux-davinci.git/log/?h=v4.11/dt), so things are looking good for EV3 in mainline Linux.

laurensvalk, 2017-01-24:
If you have the space, bring a monitor to run a loop of cool ev3dev robot videos.

dlech, 2017-01-27:
Here is my poster so far. I would like to add one or two images that exemplify each of the statements on the left.

[ELC2017_Poster.pdf](https://github.com/ev3dev/ev3dev/files/734246/ELC2017_Poster.pdf)

WasabiFan, 2017-01-27:
> I would like to add one or two images that exemplify each of the statements on the left. Any suggestions?

This seems like legos. http://www.ev3dev.org/projects/2016/08/23/Congkak-Robot/

As for the "open-source software" one, what about something OpenCV-related?

dlech, 2017-01-27:
I kind of like this with all of the LEGO parts in the background: https://www.youtube.com/watch?v=0J5F9TPrqr0

dlech, 2017-01-27:
Here is the video: https://youtu.be/0d0ktACCwT4

dlech, 2017-01-27:
I think I am pretty happy with this: [ELC2017_Poster.pdf](https://github.com/ev3dev/ev3dev/files/736161/ELC2017_Poster.pdf)

bmegli, 2017-01-27:
Awesome!

I am non-native in English so maybe it's just that:

Shouldn't it be "detailed technical information" instead of "detail technical information"?

You may also want to change poster meta information (スライド in author/title, this shows in browser tab).

dlech, 2017-01-27:
> Shouldn't it be "detailed technical information" instead of "detail technical information"?

That was part of the template. I'm not sure I can change it. Apparently it has been that way since at least [2012](http://elinux.org/images/1/1c/Elce-2012-showcase-posters.pdf).

dlech, 2017-01-27:
スライド 1 translates to "Slide 1". I suppose that explains why the green text is a bit funny. Apparently the person that made the template is Japanese. :smiley:

rhempel, 2017-01-27:
Great job in the video!

laurensvalk, 2017-01-30:
For the untrained eye, it might be a bit hard to distinguish the robot from the background in that picture?

dlech, 2017-02-25:
The conference is over and I will be heading home tomorrow. Since this is the first time I have done something like this, I packed kind of light (in terms of robots), but the ev3dev table was still quite popular at the technical showcase. It was also really fun to meet many of the people I have corresponded with.

closed 2017-02-25 by dlech

WasabiFan, 2017-02-25:
That's really great to hear! I've been so preoccupied that I completely forgot that it was even happening 😞 It'd be cool if more opportunities like this arise, now that we know that it can go well! I can't say I am of much use on the Linux side of things, but I'd otherwise be happy to help out with prep for these events in the future.