ev3dev/ev3dev #822: docker image and ros: apt can not resolve all dependencies

osmado, 2017-01-22:
Based on the "ev3dev/debian-jessie-cross" docker image, I am trying to make an "ev3dev + ros" image.  I followed the steps of Moriarty's project and all was very easy, but after compile a ros example the result was that it didn't work.  It was because the program was linked to amd64 (my host computer).  After that, I modified the dockfile in order to add armel dependencies but it does not work now. The problem is that "apt" command with armel dependencies can't resolve all dependencies (but it worked in the brickstrap image).  I sure that I make some mistake but I do not know what. Any ideas?

The wrong Dockerfile is:

...

FROM ev3dev/debian-jessie-cross

RUN DEBIAN_FRONTEND=noninteractive sudo apt-get install -y \
    apt-utils:armel \
    build-essential:armel \
    cmake:armel \
    initramfs-tools:armel \
    libboost-all-dev:armel \
    libboost-dev:armel \
    libbz2-dev:armel \
    libc6-dev:armel \
    libconsole-bridge-dev:armel \
    libgtest-dev:armel \
    liblog4cxx10:armel \
    liblog4cxx10-dev:armel \
    liblz4-dev:armel \
    libtinyxml-dev:armel \
    libpython2.7-stdlib:armel \
    libyaml-cpp-dev:armel \
    libyaml-dev:armel \
    python-coverage \ 
    python-empy \
    python-imaging \
    python-mock \
    python-netifaces \
    python-nose \
    python-numpy \
    python-paramiko \
    python-pip \
    python-yaml
...

dlech, 2017-01-22:
`ev3dev/debian-jessie-cross` won't work unless every dependency is multi-arch capable. So really, it is only good for simple projects.

Try using `ev3dev/debian-jessie-armel-cross` instead. It will be more like the old brickstrap where everything is armel already.

dlech, 2017-01-22:
If you intend to create a bootable image and you are not just cross-compiling, you should base your image on `ev3dev/ev3dev-jessie-ev3-base` or `ev3dev/ev3dev-jessie-generic-ev3`.

osmado, 2017-01-23:
Thanks a lot for your fast answer.   I has made a new docker file with "ev3dev/ev3dev-jessie-ev3-generic" and it works fine.  Tomorrow I will use it to compile a ros module to ensure it links to armel libraries.

osmado, 2017-01-24:
I has compiled a ros module. I has copied it to the ev3 and all works fine.
Thanks again.  I would like to organize the dockfile and the example ros module and, after that upload this little work in the ev3 project page.

moriarty, 2017-01-26:
Awesome!

dlech, 2017-02-08:
http://www.ev3dev.org/projects/2017/02/05/ROS-and-wiimote/

It looks like you got it working. Please close the issue if the problem is resolved.

closed 2017-02-08 by osmado