ev3dev/ev3dev #827: UART sensors worst case "latency"

bmegli, 2017-01-27:
UART sensors lag sometimes. 

## Important Notes
- this is rare (e.g. like once in 10 minutes)
- it seems related to load in some way 
- I would assess lag up to 50 ms (rationale below, may be wrong!)

## Issue Because

- now that buffer overrun issue (e.g. #792) is fixed we will have no way to know about this lag

## Issue Objective

- to assess if anything can be done about that (in the long term)

--------------------------------------------------------------------------------

By lag for sensor I mean that they show value from some time before.

The rationale for lag time assesement is following:
1.  With UART sensor 256 bytes circular buffer we note occassional overruns
2.  [EV3 hardware developper kit](https://education.lego.com/en-us/support/mindstorms-ev3/developer-kits) states that most EV3 UART sensors can report value at 1000 HZ (every 1 ms),
3. For EV3 gyro in GYRO-G&A mode I got overrun, data message has 2 + 2 + 2 = 6 bytes
4. For 256 bytes in buffer this means 256/6 ~= 42 messages
5. 42 messages out of 1000 possible per second, give 42 ms lag (messages waiting to be processed for some reason). If it's less than 1000 messages the lag is bigger. 1000 messages agrees more-or-less with 57600 bitrate and 6 byte messages.

I was unable to reproduce problem in GYRO-ANG mode (it's 4 bytes message then so more buffer time). For the color sensor it was noticed in RGB-RAW mode (10 bytes data messages, less buffer time).

One way to gather stats about the lag is going down with EV3_UART_BUFFER_SIZE. A combination of data message size, buffer size and overrun frequency gives information about lag frequency and time.

dlech, 2017-01-27:
> we will have no way to know about this lag

You can make a good guess as to when it is happing by observing that the value does not change. Like this: https://github.com/ev3dev/ev3dev/issues/324#issuecomment-101779745

> to assess if anything can be done about that (in the long term)

Part of the problem could be that we are processing the data in a workqueue. Workqueues are only scheduled every 100Hz and we are using the main system workqueue, so there could be other long running tasks scheduled before us that are causing delays.

The easier fix would be to convert [this function](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c#L475) to a tasklet provided that it does not call any functions that sleep. A harder fix would involve modifying line disciplines to use threaded interrupts (this would need to be submitted to the linux kernel mailing list).

bmegli, 2017-01-27:
Note - the overrun problem is fixed in Kernel Release Cycle 18, here it was resurrected artifically to study the lag.

EV3_UART_BUFFER_SIZE=256

I have modified message for overrun to know how many bytes were delivered to line discipline and how many are free in circular buffer [here](https://github.com/bmegli/lego-linux-drivers/blob/ev3-uart-buffer/sensors/ev3_uart_sensor_ld.c#L1037)

This is what I got for color sensor in RGB-RAW (over long time with high load):

scheme:
device: buffer ovverun delivered_bytes/free_CB_bytes

``` bash
[  330.410664] ttyS0: buffer overrun 256/129
[  330.410886] ttyS0: buffer overrun 256/249
[  333.290091] tty ttyS0: Reconnected due to: Received INFO for incorrect mode.
[  350.170695] ttyS0: buffer overrun 8/5
[  350.172414] ttyS0: buffer overrun 8/5
[  350.190027] ttyS0: buffer overrun 56/5
[  350.191982] ttyS0: buffer overrun 8/5
[  350.192786] ttyS0: buffer overrun 8/5
[  350.195983] ttyS0: buffer overrun 8/3
[  350.197475] ttyS0: buffer overrun 8/3
[  353.080063] tty ttyS0: Reconnected due to: Invalid name INFO.
[  355.311778] ttyS0: buffer overrun 8/1
[  355.313276] ttyS0: buffer overrun 8/1
[  355.315077] ttyS0: buffer overrun 8/1
[  355.315978] ttyS0: buffer overrun 2/1
[  355.318028] ttyS0: buffer overrun 8/1
[  355.319570] ttyS0: buffer overrun 8/1
[  355.327033] ttyS0: buffer overrun 34/1
[  355.328968] ttyS0: buffer overrun 8/1
[  355.330840] ttyS0: buffer overrun 8/1
[  355.332944] ttyS0: buffer overrun 8/1
[  358.280073] tty ttyS0: Reconnected due to: Received INFO for incorrect mode.
[  360.378760] ev3_uart_receive_buf: 39 callbacks suppressed
[  360.378820] ttyS0: buffer overrun 8/7
[  363.270085] tty ttyS0: Reconnected due to: Received INFO for incorrect mode.
[  380.178883] ttyS0: buffer overrun 8/1
[  380.185566] ttyS0: buffer overrun 32/1
[  380.188513] ttyS0: buffer overrun 16/1
[  380.191100] ttyS0: buffer overrun 8/1
[  380.192965] ttyS0: buffer overrun 8/1
[  380.194160] ttyS0: buffer overrun 4/1
[  380.197037] ttyS0: buffer overrun 8/1
[  380.198681] ttyS0: buffer overrun 8/1
[  380.200338] ttyS0: buffer overrun 8/1
[  383.060164] tty ttyS0: Reconnected due to: Received INFO for incorrect mode.
[  395.358790] ev3_uart_receive_buf: 13 callbacks suppressed
[  395.358849] ttyS0: buffer overrun 8/5
[  395.360353] ttyS0: buffer overrun 8/5
[  395.380089] ttyS0: buffer overrun 71/5
[  395.380150] ttyS0: buffer overrun 23/5
[  395.382268] ttyS0: buffer overrun 8/5
[  395.383498] ttyS0: buffer overrun 8/5
[  398.250059] tty ttyS0: Reconnected due to: Received INFO for incorrect mode.
[  400.318828] ttyS0: buffer overrun 8/5
[  400.325298] ttyS0: buffer overrun 32/5
[  400.327045] ttyS0: buffer overrun 8/5
[  400.328546] ttyS0: buffer overrun 8/5
[  403.140510] tty ttyS0: Reconnected due to: Received INFO for incorrect mode.
[  405.388669] ttyS0: buffer overrun 8/1
[  405.397469] ttyS0: buffer overrun 34/1
[  405.398936] ttyS0: buffer overrun 8/1
[  405.400637] ttyS0: buffer overrun 8/1
[  405.402422] ttyS0: buffer overrun 8/1
[  405.405357] ttyS0: buffer overrun 8/1
[  405.407281] ttyS0: buffer overrun 8/1
[  405.408046] ttyS0: buffer overrun 8/1
[  405.410060] ttyS0: buffer overrun 6/1
[  405.411664] ttyS0: buffer overrun 8/1
[... 34 lines cut ...]
[  560.314530] ttyS0: buffer overrun 8/3
[  625.470296] ttyS0: buffer overrun 8/3
[  625.473123] ttyS0: buffer overrun 8/3
[  625.474627] ttyS0: buffer overrun 7/3
[  625.476274] ttyS0: buffer overrun 8/2
[  625.478219] ttyS0: buffer overrun 8/2
[  625.486773] ttyS0: buffer overrun 32/2
[  625.487006] ttyS0: buffer overrun 8/2
[  625.488433] ttyS0: buffer overrun 8/2
[  625.499395] ttyS0: buffer overrun 44/2
[  628.320201] tty ttyS0: Reconnected due to: Received INFO for incorrect mode.
```

Conclussions:
- it's overrun that triggers reconnections
- recconection triggers more overruns when CB is not emptied
- of each overrun series only the first in series is informative (rest is artifact)

Higher level conclussions:
- there are big overruns like 256 delivered/249 free
- there are small overruns (the CB buffer almost full)

Big overruns are rare. Little can done about big overruns because this is what we get from lower kernel layer tty receive_buf. We can't process it if we don't have it. If we receive it in a big chunk the lag is not in line discipline.  Maybe something can be done but this is kernel tty code, not ev3dev code.

Something can be done about small overruns (e.g. what @dlech mentions about workqueues).

bmegli, 2017-01-27:
For EV3 Gyro in GYRO-G&A mode, 6 byte messages (this is hard to reproduce)

``` bash
[  556.250760] ttyS0: buffer overrun 3/2
[  556.252186] ttyS0: buffer overrun 8/2
[  911.384985] ttyS0: buffer overrun 8/3
[  911.386403] ttyS0: buffer overrun 8/3
[  911.388052] ttyS0: buffer overrun 8/3
[  911.389652] ttyS0: buffer overrun 6/3
[  911.391050] ttyS0: buffer overrun 8/3
[  911.392485] ttyS0: buffer overrun 8/3
[  911.393802] ttyS0: buffer overrun 8/3
[  911.396151] ttyS0: buffer overrun 8/3
[  911.397551] ttyS0: buffer overrun 8/3
[  911.404807] ttyS0: buffer overrun 32/3
[  914.380120] tty ttyS0: Reconnected due to: Bad message size
[ 2513.311136] ev3_uart_receive_buf: 1 callbacks suppressed
[ 2513.311239] ttyS0: buffer overrun 8/1
[ 2513.312237] ttyS0: buffer overrun 2/1
[ 2516.370071] tty ttyS0: Reconnected due to: Unexpected mode.
[ 2548.282578] ttyS0: buffer overrun 8/1
```

bmegli, 2017-01-27:
> You can make a good guess as to when it is happing by observing that the value does not change. Like this: #324 (comment)

Right! Like rotating constantly around with gyro sensor and watching when it's value doesn't change even though we know it does. It will work correctly if user space program sampling gyro runs during the lag (and not if it doesn't).